espressif/esp-matter #74: [Feature request] ESP Matter work with IPv6 over BLE (RFC7668)

mrninhvn, 2022-10-28:
Matter is an application layer that runs on IPv6 networks, so it would be nice if it supports IPv6 BLE networks.
Most ESP32 chips have BLE and already have an IPv6 stack implemented, although it hasn't been updated for a while:
[https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/netif/lowpan6_ble.c](url)

jonsmirl, 2022-10-28:
The Matter group has pushed Matter over BLE to the next release (or maybe further). It is in the Matter spec, they just aren't going to support it until later.

BLE Mesh support is available right now. 
https://github.com/espressif/esp-matter/tree/main/examples/blemesh_bridge

dhrishi, 2022-11-02:
@mrninhvn IPv6 for BLE is not a part of Matter v1.0 specification and hence the implementation. The supported transports for Matter in v1.0 over IP are: Wi-Fi, Ethernet and Thread. 

Please close the issue if your question is answered.

closed 2022-11-03 by mrninhvn