espressif/esp-matter #75: chip-tool pairing failed

sidhoo-root, 2022-10-28:
i meet this [issue](https://github.com/espressif/esp-matter/issues/2) too. can u help me
details:

ubuntu 20.04 on vitual meachine
device is esp32c3,

1. flash the manufacturing binary https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool#readme,
2. run device, log:
ESP-ROM:esp32c3-api1-20210207
Build:Feb 7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
I (30) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (30) boot: compile time 14:24:49
I (30) boot: chip revision: 3I (37) boot.esp32c3: SPI Mode : DIO
I (41) boot.esp32c3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label Usage Type ST Offset Length
I (62) boot: 0 sec_cert unknown 3f 06 0000d000 00003000
I (70) boot: 1 nvs WiFi data 01 02 00010000 00006000
I (77) boot: 2 otadata OTA data 01 00 00016000 00002000
I (85) boot: 3 phy_init RF data 01 01 00018000 00001000
I (92) boot: 4 ota_0 OTA app 00 10 00020000 001e0000
I (100) boot: 5 ota_1 OTA app 00 11 00200000 001e0000
I (107) boot: 6 fctry WiFi data 01 02 003e0000 00006000
I (115) boot: End of partition table
I (119) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=2f3f8h (193528) map
I (158) esp_image: segment 1: paddr=0004f420 vaddr=3fc90400 size=00bf8h ( 3064) load
I (159) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=fb500h (1029376) map
I (324) esp_image: segment 3: paddr=0014b528 vaddr=3fc90ff8 size=02ad0h ( 10960) load
I (327) esp_image: segment 4: paddr=0014e000 vaddr=40380000 size=1033ch ( 66364) load
I (342) esp_image: segment 5: paddr=0015e344 vaddr=50000010 size=00010h ( 16) load
I (348) boot: Loaded app from partition at offset 0x20000
I (348) boot: Disabling RNG early entropy source...
I (363) cpu_start: Pro cpu up.
I (371) cpu_start: Pro cpu start user code
I (371) cpu_start: cpu freq: 160000000
I (371) cpu_start: Application information:
I (374) cpu_start: Project name: light
I (379) cpu_start: App version: v1.0
I (383) cpu_start: Compile time: Oct 28 2022 16:11:32
I (389) cpu_start: ELF file SHA256: e7c0ddc45aab754c...
I (395) cpu_start: ESP-IDF: v4.4.2
I (400) heap_init: Initializing. RAM available for dynamic allocation:
I (407) heap_init: At 3FCA46E0 len 0001B920 (110 KiB): DRAM
I (414) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (420) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM0;32mI (427) spi_flash: detected chip: generic
I (432) spi_flash: flash io: dio
I (437) sleep: Configure to isolate all GPIO pins in sleep state
I (442) sleep: Enable automatic switching of GPIO sleep configuration
I (450) coexist: coexist rom version 9387209
I (454) cpu_start: Starting scheduler.
I (467) led_driver_ws2812: Initializing light driver
I (467) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (487) app_main: Light created with endpoint_id 1
E (487) esp_matter_core: Command 0x0047 on cluster 0x0300 already exists. Not creating again.
I (497) pp: pp rom version: 9387209
I (497) net80211: net80211 rom version: 9387209
I (507) wifi:wifi driver task: 3fcb0920, prio:23, stack:6656, core=0
I (507) system_api: Base MAC address is not set
I (517) system_api: read default base MAC address from EFUSE
I (517) wifi:wifi firmware version: eeaa27d
I (527) wifi:wifi certification version: v7.0
I (527) wifi:config NVS flash: enabled
[... 128 lines cut ...]
I (1797) phy: chan:11,max_power:80
I (1807) phy: chan:12,max_power:80
I (1807) phy: chan:13,max_power:80
I (1807) phy: chan:14,max_power:80
W (1817) wifi:Haven't to connect to a suitable AP now!
I (1817) chip[DL]: Done driving station state, nothing else to do...
I (1827) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MATTER-2481)
I (1837) chip[DL]: Device already advertising, stop active advertisement and restart
I (1847) NimBLE: GAP procedure initiated: stop advertising.
I (1857) NimBLE: GAP procedure initiated: advertise;
I (1857) NimBLE: disc_mode=2
I (1867) NimBLE: adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1877) NimBLE:
I (1877) app_main: Commissioning window opened
I (1877) led_driver_ws2812: led set r:0, g:0, b:0
I (1887) led_driver_ws2812: led set r:0, g:0, b:0
I (1887) led_driver_ws2812: led set r:25, g:20, b:16
I (31447) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable, device name MATTER-2481)
I (31447) chip[DL]: Device alrea32mI (31457) NimBLE: GAP procedure initiated: stop advertising.
I (31467) NimBLE: GAP procedure initiated: advertise;
I (31467) NimBLE: disc_mode=2
I (31467) NimBLE: adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (31477) NimBLE:

3. run chip-tool cmd :./chip-tool pairing ble-wifi 1234 Redmi_huan 87654321 38802273 2481

jonsmirl, 2022-10-28:
Your device is failing to attach to your wifi router.
```
I (1807) phy: chan:14,max_power:80
W (1817) wifi:Haven't to connect to a suitable AP now!
```

sidhoo-root, 2022-10-30:
@jonsmirl yes, but chip-tool report the same error on a ubuntu meachine,here it is
[image]

jonsmirl, 2022-10-30:
I can not get chip pairing to work on my Ubuntu machine unless I add a second USB Bluetooth adapter and tell chip-tool to use it. This is because the Ubuntu Bluetooth code has control of the first adapter. Some people can get it working with just one, but I am unable to get it working solo.

BTW, the screen shot cuts off before any error is visible.

dhrishi, 2022-10-31:
@sidhoo-root Can you upload/attach the complete chip-tool logs. The screenshot does not seem to show the required errors

sidhoo-root, 2022-10-31:
@dhrishi ok. is this issue related to my desktop ubuntu meachine ?

logs:
efsz@lipeiquan-pcub:~/esp/esp-matter/connectedhomeip/connectedhomeip/chip_tool$ ./chip-tool pairing ble-wifi 1234 Redmi_huan 87654321 38802273 2481
[1667114262.217646][58592:58592] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1667114262.217754][58592:58592] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1667114262.217773][58592:58592] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1667114262.217785][58592:58592] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1667114262.217867][58592:58592] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-yS6iCG)
[1667114262.217946][58592:58592] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1667114262.217954][58592:58592] CHIP:DL: NVS set: chip-counters/reboot-count = 13 (0xD)
[1667114262.218155][58592:58592] CHIP:DL: Got Ethernet interface: enp3s0
[1667114262.218387][58592:58592] CHIP:DL: Found the primary Ethernet interface:enp3s0
[1667114262.218550][58592:58592] CHIP:DL: Failed to get WiFi interface
[1667114262.218553][58592:58592] CHIP:DL: Failed to reset WiFi statistic counts
[1667114262.218566][58592:58592] CHIP:IN: UDP::Init bind&listen port=0
[1667114262.218597][58592:58592] CHIP:IN: UDP::Init bound to port=44644
[1667114262.218599][58592:58592] CHIP:IN: UDP::Init bind&listen port=0
[1667114262.218608][58592:58592] CHIP:IN: UDP::Init bound to port=33567
[1667114262.218609][58592:58592] CHIP:IN: BLEBase::Init - setting/overriding transport
[1667114262.218610][58592:58592] CHIP:IN: TransportMgr initialized
[1667114262.218616][58592:58592] CHIP:FP: Initializing FabricTable from persistent storage
[1667114262.218623][58592:58592] CHIP:TS: Last Known Good Time: [unknown]
[1667114262.218625][58592:58592] CHIP:TS: Setting Last Known Good Time to firmware build time 2022-10-30T14:59:44
[1667114262.223003][58592:58592] CHIP:ZCL: Using ZAP configuration...
[1667114262.223290][58592:58592] CHIP:CTL: System State Initialized...
[1667114262.223300][58592:58592] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1667114262.223308][58592:58592] CHIP:CTL: Setting attestation nonce to random value
[1667114262.223337][58592:58592] CHIP:CTL: Setting CSR nonce to random value
[1667114262.223345][58592:58592] CHIP:IN: UDP::Init bind&listen port=5550
[1667114262.223352][58592:58592] CHIP:IN: UDP::Init bound to port=5550
[1667114262.223354][58592:58592] CHIP:IN: UDP::Init bind&listen port=5550
[1667114262.223361][58592:58592] CHIP:IN: UDP::Init bound to port=5550
[1667114262.223363][58592:58592] CHIP:IN: TransportMgr initialized
[1667114262.223445][58592:58597] CHIP:DL: CHIP task running
[1667114262.223555][58592:58597] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1667114262.223562][58592:58597] CHIP:CTL: Setting attestation nonce to random value
[1667114262.223570][58592:58597] CHIP:CTL: Setting CSR nonce to random value
[1667114262.223625][58592:58597] CHIP:CTL: Couldn't get ExampleOpCredsCAKey from storage: ../examples/chip-tool/config/PersistentStorage.cpp:85: CHIP Error 0x000000A0: Value not found in the persisted storage
[1667114262.224337][58592:58597] CHIP:CTL: Couldn't get ExampleOpCredsICAKey from storage: ../examples/chip-tool/config/PersistentStorage.cpp:85: CHIP Error 0x000000A0: Value not found in the persisted storage
[1667114262.224506][58592:58597] CHIP:CTL: Generating RCAC
[1667114262.224632][58592:58597] CHIP:CTL: Generating ICAC
[1667114262.224739][58592:58597] CHIP:CTL: Generating NOC
[1667114262.224861][58592:58597] CHIP:FP: Validating NOC chain
[1667114262.225162][58592:58597] CHIP:FP: NOC chain validation successful
[1667114262.225196][58592:58597] CHIP:FP: Added new fabric at index: 0x1
[1667114262.225199][58592:58597] CHIP:FP: Assigned compressed fabric ID: 0x8ED7A56E6C4F421C, node ID: 0x000000000001B669
[1667114262.225201][58592:58597] CHIP:TS: Last Known Good Time: 2022-10-30T14:59:44
[1667114262.225203][58592:58597] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1667114262.225205][58592:58597] CHIP:TS: Retaining current Last Known Good Time
[1667114262.225328][58592:58597] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1667114262.225536][58592:58597] CHIP:TS: Committing Last Known Good Time to storage: 2022-10-30T14:59:44
[1667114262.225725][58592:58597] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1667114262.225755][58592:58597] CHIP:IN: UDP::Init bind&listen port=5550
[1667114262.225763][58592:58597] CHIP:IN: UDP::Init bound to port=5550
[1667114262.225764][58592:58597] CHIP:IN: UDP::Init bind&listen port=5550
[1667114262.225769][58592:58597] CHIP:IN: UDP::Init bound to port=5550
[1667114262.225770][58592:58597] CHIP:IN: TransportMgr initialized
[1667114262.227663][58592:58597] CHIP:IN: SecureSession[0x7fdcac01dbb0]: Allocated Type:1 LSID:53257
[1667114262.227670][58592:58597] CHIP:SC: Assigned local session key ID 53257
[1667114262.227679][58592:58597] CHIP:SC: Including MRP parameters in PBKDF param request
[1667114262.227705][58592:58597] CHIP:EM: <<< [E:41872i M:237146974] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
[1667114262.227711][58592:58597] CHIP:IN: (U) Sending msg 237146974 to IP address 'BLE'
[1667114262.227714][58592:58597] CHIP:IN: Message appended to BLE send queue
[... cut ...]
[1667114292.233374][58592:58592] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1667114292.233410][58592:58592] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1667114292.233423][58592:58592] CHIP:DL: Inet Layer shutdown
[1667114292.233433][58592:58592] CHIP:DL: BLE shutdown
[1667114292.233444][58592:58592] CHIP:DL: System Layer shutdown
[1667114292.233482][58592:58592] CHIP:TOO: Run command failure: ../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:245: CHIP Error 0x00000032: Timeout

dhrishi, 2022-10-31:
@sidhoo-root Yes, it looks like the issue related to BLE on your host machine. See the below in your logs:

>[1667114287.233598][58592:58597] CHIP:IN: BleConnection Error: ../examples/chip-tool/third_party/connectedhomeip/src/platform/Linux/BLEManagerImpl.cpp:655: CHIP Error 0x00000003: Incorrect state
[1667114287.233610][58592:58597] CHIP:BLE: No adapter available for new connection establishment

sidhoo-root, 2022-11-01:
@dhrishi Thanks , i am trying to pair the device with another laptop computer

jonsmirl, 2022-11-01:
I just stick a $2 USB Bluetooth adapter into the computer and then use '--ble-adapter 1' on chip-tool.  

This is kind of hit or miss. Some of my computers work ok without the extra adapter, some need it. This is not anything to do with Matter, I think it is a function of the Bluetooth hardware in the computers. Some computers have very minimal BT hardware which is not capable of being shared.

sidhoo-root, 2022-11-03:
@jonsmirl @dhrishi thank you . it works within the laptop computer, the device can be pairing now. but it comes anther problem, the device can not be control by chiptool. i will open another issue to push the log

closed 2022-11-03 by sidhoo-root