espressif/esp-matter #76: How do I enable ap to configure network information instead of configuring network information through ble

shun-dgut, 2022-10-28:


jonsmirl, 2022-10-28:
You can bring the ESP32 up on wifi first using whatever method you choose. You can use SmartAP if you want. This has nothing to do with Matter.  

After the ESP32 is up and on wifi, start Matter. Matter will make wifi MDNS entries which will allow anyone on the network to find the device and commission it. There is no requirement to initially configure via BLE. Most people are going to do it that way and there are standard tools for doing that, but it is not required.  

This should work, but I have not tried it. I don't know how to configure ESP-matter to ignore BLE.

jadhavrohit924, 2022-11-02:
Steps:-

- Flash light examle of esp-matter
- Connect wifi through matter CLI

> `matter wifi connecct <SSID ><pass>`

- Commission light using controller 

> ./out/debug/chip-tool pairing onnetwork <node-id> <setup-pin-code>

dhrishi, 2022-11-02:
@shun-dgut Can you please try out the steps mentioned above? Let us know if it helps.

shun-dgut, 2022-11-03:
>  a error occurred executing the command
[image]

dhrishi, 2022-11-03:
@shun-dgut You missed "connect" in the command?

dhrishi, 2022-11-03:
@shun-dgut This should work fine. Please close the issue if it does

dhrishi, 2022-11-04:
Closing this now. Feel free to re-open with the details in case you still see the issue

closed 2022-11-04 by dhrishi

wch188, 2023-07-07:
@dhrishi I have a similar problem
I use the ap to connect the matter device to WiFi and pairing it with the command./chip-tool pairing onnetwork 0x11 20202021, but I have had no success. What should the success log show?How do I control the device?
dev:esp32c3 project:light env:WSL 2 Ubuntu 22.04

this device log 
```
I (101) boot:  5 ota_0            OTA apESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1704
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f68
entry 0x403cc710
I (30) boot: ESP-IDF v5.1 2nd stage bootloader
I (30) boot: compile time Jul  7 2023 10:51:15
I (30) boot: chip revision: v0.3
I (33) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (71) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (86) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (93) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (101) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (108) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (116) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (124) boot: End of partition table
I (128) boot: No factory image, trying OTA 0
I (133) esp_image: segment 0: paddr=00020020 vaddr=3c110020 size=38ec8h (233160) map
I (178) esp_image: segment 1: paddr=00058ef0 vaddr=3fc94a00 size=03494h ( 13460) load
I (181) esp_image: segment 2: paddr=0005c38c vaddr=40380000 size=03c8ch ( 15500) load
I (187) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=107d90h (1080720) map
I (365) esp_image: segment 4: paddr=00167db8 vaddr=40383c8c size=10be0h ( 68576) load
I (385) boot: Loaded app from partition at offset 0x20000
I (441) boot: Set actual ota_seq=1 in otadata[0]
I (441) boot: Disabling RNG early entropy source...
I (452) cpu_start: Unicore app
I (452) cpu_start: Pro cpu up.
I (461) cpu_start: Pro cpu start user code
I (461) cpu_start: cpu freq: 160000000 Hz
I (461) cpu_start: Application information:
I (464) cpu_start: Project name:     light
I (469) cpu_start: App version:      1.0
I (473) cpu_start: Compile time:     Jul  7 2023 10:51:06
I (479) cpu_start: ELF file SHA256:  ddeb78d8eadc7d4e...
I (485) cpu_start: ESP-IDF:          v5.1
I (490) cpu_start: Min chip rev:     v0.3
I (495) cpu_start: Max chip rev:     v0.99
I (499) cpu_start: Chip rev:         v0.3
I (504) heap_init: Initializing. RAM available for dynamic allocation:
I (511) heap_init: At 3FCA8DC0 len 00033950 (206 KiB): DRAM
I (518) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (524) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (531) spi_flash: detected chip: generic
I (535) spi_flash: flash io: dio
W (540) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (550) sleep: Configure to isolate all GPIO pins in sleep state
I (557) sleep: Enable automatic switching of GPIO sleep configuration
I (564) coexist: coex firmware version: ebddf30
I (569) coexist: coexist rom version 9387209
I (574) app_start: Starting scheduler on CPU0
I (579) main_task: Started on CPU0
I (579) main_task: Calling app_main()
I (589) led_driver_ws2812: Initializing light driver
[... 203 lines cut ...]
[1688696954.529154][5952:5954] CHIP:CTL: Setting attestation nonce to random value
[1688696954.529192][5952:5954] CHIP:CTL: Setting CSR nonce to random value
[1688696954.529433][5952:5954] CHIP:CTL: Generating NOC
[1688696954.529736][5952:5954] CHIP:FP: Validating NOC chain
[1688696954.530081][5952:5954] CHIP:FP: NOC chain validation successful
[1688696954.530119][5952:5954] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1688696954.530123][5952:5954] CHIP:TS: Last Known Good Time: 2023-07-05T12:57:04
[1688696954.530134][5952:5954] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1688696954.530136][5952:5954] CHIP:TS: Retaining current Last Known Good Time
[1688696954.530417][5952:5954] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1688696954.530672][5952:5954] CHIP:TS: Committing Last Known Good Time to storage: 2023-07-05T12:57:04
[1688696954.530926][5952:5954] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 4CAE7458FBFA89EC)
[1688696985.564281][5952:5954] CHIP:DIS: Timeout waiting for mDNS resolution.

onoff toggle 0x11 1
[1688697121.759857][5952:5952] CHIP:TOO: Command: onoff toggle 0x11 1
[1688697121.760055][5952:5954] CHIP:TOO: Sending command to node 0x11
[1688697121.760126][5952:5954] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000011]
[1688697121.760140][5952:5954] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1688697121.760149][5952:5954] CHIP:DIS: OperationalSessionSetup[1:0000000000000011]: State change 1 --> 2
[1688697121.960688][5952:5954] CHIP:DIS: Checking node lookup status after 200 ms
[1688697152.791312][5952:5954] CHIP:DIS: Timeout waiting for mDNS resolution.
[1688697166.774943][5952:5954] CHIP:DIS: Checking node lookup status after 45015 ms
[1688697166.775009][5952:5954] CHIP:DIS: OperationalSessionSetup[1:0000000000000011]: operational discovery failed: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout
```

wch188, 2023-07-07:
```
[1688696939.098228][5952:5952] CHIP:DL: Failed to get WiFi interface
[1688696939.098244][5952:5952] CHIP:DL: Failed to reset WiFi statistic counts
```
Does this affect it? I look at the other issues. You're not focusing on this。
In addition,
```
 pairing onnetwork 0x11 20202021 
```
What is the onnetwork in this directive? Can this alone give chip-tool access to the network and pair matter?

dhrishi, 2023-07-08:
>I use the ap to connect the matter device to WiFi 

Which command did you use to connect the Matter device to the Wi-Fi network? If not the command/CLI, how did you connect the device to the AP?

wch188, 2023-07-10:
Use this command "matter esp wifi connect xxx xxx",xxx is my account and password

```
I (93) boot:  4 phy_init         RF data         ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1704
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f68
entry 0x403cc710
I (30) boot: ESP-IDF v5.1 2nd stage bootloader
I (30) boot: compile time Jul  7 2023 10:51:15
I (30) boot: chip revision: v0.3
I (33) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (71) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (86) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (93) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (101) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (108) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (116) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (124) boot: End of partition table
I (128) boot: No factory image, trying OTA 0
I (133) esp_image: segment 0: paddr=00020020 vaddr=3c110020 size=38ec8h (233160) map
I (178) esp_image: segment 1: paddr=00058ef0 vaddr=3fc94a00 size=03494h ( 13460) load
I (181) esp_image: segment 2: paddr=0005c38c vaddr=40380000 size=03c8ch ( 15500) load
I (187) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=107d90h (1080720) map
I (365) esp_image: segment 4: paddr=00167db8 vaddr=40383c8c size=10be0h ( 68576) load
I (385) boot: Loaded app from partition at offset 0x20000
I (448) boot: Set actual ota_seq=1 in otadata[0]
I (448) boot: Disabling RNG early entropy source...
I (459) cpu_start: Unicore app
I (459) cpu_start: Pro cpu up.
I (468) cpu_start: Pro cpu start user code
I (468) cpu_start: cpu freq: 160000000 Hz
I (468) cpu_start: Application information:
I (471) cpu_start: Project name:     light
I (476) cpu_start: App version:      1.0
I (480) cpu_start: Compile time:     Jul  7 2023 10:51:06
I (486) cpu_start: ELF file SHA256:  2f1125a851616a04...
I (492) cpu_start: ESP-IDF:          v5.1
I (497) cpu_start: Min chip rev:     v0.3
I (502) cpu_start: Max chip rev:     v0.99
I (507) cpu_start: Chip rev:         v0.3
I (511) heap_init: Initializing. RAM available for dynamic allocation:
I (519) heap_init: At 3FCA8DC0 len 00033950 (206 KiB): DRAM
I (525) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (532) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (539) spi_flash: detected chip: generic
I (542) spi_flash: flash io: dio
W (547) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (558) sleep: Configure to isolate all GPIO pins in sleep state
I (564) sleep: Enable automatic switching of GPIO sleep configuration
I (571) coexist: coex firmware version: ebddf30
I (576) coexist: coexist rom version 9387209
I (581) app_start: Starting scheduler on CPU0
I (586) main_task: Started on CPU0
I (586) main_task: Calling app_main()
I (596) led_driver_ws2812: Initializing light driver
I (596) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
W (616) esp_matter_core: Command 0x00000047 on cluster 0x00000300 already exists. Not creating again.
I (616) app_main: Light created with endpoint_id 1
[... 184 lines cut ...]
I (27956) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.3.30/255.255.255.0 gateway 192.168.3.1
I (27956) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (27966) app_main: Interface IP Address changed
I (27966) chip[DIS]: Updating services using commissioning mode 1
I (27976) chip[DIS]: CHIP minimal mDNS started advertising.
I (27986) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (27996) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 026784F030E2C3B8.
I (28006) chip[DIS]: mDNS service published: _matterc._udp
I (31476) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (31476) chip[DL]: Device already advertising, stop active advertisement and restart
I (31496) NimBLE: GAP procedure initiated: stop advertising.

I (31496) NimBLE: GAP procedure initiated: advertise;
I (31506) NimBLE: disc_mode=2
I (31506) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (31516) NimBLE:

I (35656) wifi:<ba-add>idx:1 (ifx:0, b0:e5:ed:fc:90:e8), tid:7, ssn:2, winSize:64
```
At this time, I use the above command in chip-tool, but there is no output from the device log, and the device does not turn off the light when the command to control the light is sent,and it took a long time for the device log to output the following.
```
I (375606) ROUTE_HOOK: Ignore invalid ICMP packet
I (385946) ROUTE_HOOK: Ignore invalid ICMP packet

```

wch188, 2023-07-10:
Output after  longer(maybe ten min?) @dhrishi 
```
E (901476) chip[BLE]: CancelBleIncompleteConnection() failed, err = 3
I (901476) chip[SVR]: Closing pairing window
I (901476) chip[DIS]: Updating services using commissioning mode 0
I (901486) chip[DIS]: CHIP minimal mDNS started advertising.
I (901496) NimBLE: GAP procedure initiated: stop advertising.

I (901496) chip[DL]: CHIPoBLE advertising stopped
I (901516) app_main: Commissioning window closed
```

dhrishi, 2023-07-14:
@wch188 I tried the on-network commissioning and it works fine at my end. I have attached the files for both device as well as the chip-tool side. Note, the host machine running the chip-tool is connected to the same AP to which the device is connected.

I directly used the light example firmware from [ESP Launchpad](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml). Can you try that out so that as well?
[esp32-c3.txt](https://github.com/espressif/esp-matter/files/12047216/esp32-c3.txt)
[chip-tool.txt](https://github.com/espressif/esp-matter/files/12047218/chip-tool.txt)

wch188, 2023-07-14:
Ok, I'll try. In addition, I bought a Bluetooth adapter, using Bluetooth control successfully.

Do you know of any Android products whose software can connect to matter? I want to control the light routine through the phone app connection. I don't have an iPhone right now

dhrishi, 2023-07-14:
>Do you know of any Android products whose software can connect to matter? I want to control the light routine through the phone app connection. I don't have an iPhone right now

There's an Android CHIPTool app in the connectedhomeip repo which [you can build](https://github.com/espressif/connectedhomeip/blob/master/docs/guides/android_building.md#building-android-chiptool-from-scripts) and try. Although, I am not sure about how well is it maintained. If you have a Google hub then you can use the Android phone with the Home app and test commissioning.

wch188, 2023-07-14:
Android CHIPTool I tried this with IP and it kept popping back so it didn't work. I've just tried the above method using on-network and it doesn't work. I suspect that the virtual machine's network is not really accessing the network that the matter device is connected to, and I will continue to find out why when I have some time later. Do you know how to make the matter device log display the QR code? I see some people can display QR code on their log when they use light.

dhrishi, 2023-07-14:
> I've just tried the above method using on-network and it doesn't work. I suspect that the virtual machine's network is not really accessing the network that the matter device is connected to

You can validate that by running the command `avahi-browse _matterc._udp -r` on the host machine running the chip-tool. Ideally, you should see the Matter device advertising over the IP network

>Do you know how to make the matter device log display the QR code? I see some people can display QR code on their log when they use light.

Why do you need that? It was previously present in the device log, but that was later removed as the QR code (and the relevant contents) should not be present on the device. You can generate the QR code on the host using `chip-tool payload` command. Also, the default QR code for the ESP examples is here (at the bottom): https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning

wch188, 2023-07-14:
Because I want to try the way through the two-dimensional code, Android CHIPTool will not blink back. avahi-browse _matterc._udp -r this command has to wait, I use nonetwork mode to test found not, re-use ble-wifi mode, and then it shows the following problem. What could be the cause? The whole process is the same as what I did successfully before.
[chip-tool_issues.txt](https://github.com/espressif/esp-matter/files/12048017/chip-tool_issues.txt)

wch188, 2023-07-14:
[image]

This is the configuration on my current network, and below is the previous successful configuration. Could that be the reason? I am currently unable to connect to the network using the following configuration, so I am not sure if it is the cause

[image]

jonsmirl, 2023-07-14:
Google sample Android app:
https://github.com/google-home/sample-app-for-matter-android

wch188, 2023-07-17:
OK,thank you.