espressif/esp-matter #77: PartsList of EP0 and Bridges

jonsmirl, 2022-10-28:
Section 9.13.6 of the CHIP spec describes how a controller is able to monitor devices being added or removed from a bridge.
```
When an update to the set of Bridged Devices (which are exposed according to the Section 9.13.11,
“Best practices for Bridge Manufacturers”) occurs, the Bridge SHALL
• on the Descriptor cluster of endpoint 0: update the PartsList attribute (add/remove entries from
this list)
• update the exposed endpoints and their descriptors according to the new set of Bridged Devices

Nodes that wish to be notified of added/removed devices SHOULD monitor changes of the PartsList
attribute in the Descriptor cluster on endpoint 0.
```

Is the esp-matter endpoint creation code modfying the PartLists on EP0 as required?  These needs to dynamically happen, not just at initialization time. If I have a commissioned bridge running, and then add another BLE/Zigbee device to it, the PartList on EP0 should dynamically change.

chiragatal, 2022-10-31:
This is not explicitly happening in esp-matter. I could see some code in `attribute-storage.cpp` which reports the PartsList when a dynamic endpoint is enabled/disabled. For enabling, i.e. adding new endpoints, I think this should work.

For disabling, we call the `emberAfClearDynamicEndpoint()` API from esp-matter, which then reports the PartsList. But I am not sure if it will work as expected. We'll have to do some tests.

wqx6, 2022-12-20:
PartsList is always read by this [function](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/descriptor/descriptor.cpp#L58-L110). This function in descriptor will return an encoded value according to the data model of the matter node. So we do not need to modify the parts list of Endpoint 0 or other parent endpoints when we enable or disable the endpoints.

dhrishi, 2023-01-27:
@jonsmirl Is this addressed and can be closed?

dhrishi, 2023-02-15:
Closing this. Please re-open in case of any further questions

closed 2023-02-15 by dhrishi