espressif/esp-matter #78: ESP Matter app connect to WiFi failed

linhr3, 2022-10-31:
Hi Espressif Experts,
I tried to run the light DEMO but the ESP Matter app showed that the transaction timed out.

The detail:
board chip: ESP32
ubuntu 20.04 on virtual machine
idf.4.4.2
matter code commited by 2020.9.13
After scanning the QR code, I chose the WiFi item. Then the device should have succeeded in  connecting to the AP (I initiated the AP from the iphone). But after a while, the app showed that the transaction timed out. And the log in terminal is showned as below:

I (19053) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:489898 on exchange 41313r
I (19063) chip[NP]: ESP NetworkCommissioningDelegate: SSID: young
I (19063) chip[DL]: Disconnecting WiFi station interface
I (19073) wifi:state: run -> init (0)
I (19073) wifi:pm stop, total sleep time: 13168002 us / 16545894 us

W (19073) wifi:<ba-del>idx
I (19093) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (19123) chip[DL]: WiFi station mode change: Enabled -> Disabled
I (19143) chip[DL]: Changing ESP WiFi mode: STA -> STA+AP
I (19153) wifi:mode : sta (b8:d6:1a:33:d0:e4) + softAP (b8:d6:1a:33:d0:e5)
I (19153) wifi:Total power save buffer number: 16
I (19153) wifi:Init max length of beacon: 752/752
I (19163) wifi:Init max length of beacon: 752/752
I (19183) chip[DL]: WiFi station mode change: Disabled -> Enabled
E (19183) chip[DL]: Long dispatch time: 131 ms, for event type 7
I (19183) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
W (19193) wifi:Haven't to connect to a suitable AP now!
I (19193) chip[DL]: WiFi station state change: Connected -> NotConnected
I (19213) chip[DL]: WiFi station interface disconnected
I (19213) chip[ZCL]: WiFiDiagnosticsDelegate: OnDisconnectionDetected
I (19223) chip[ZCL]: WiFiDiagnosticsDelegate: OnConnectionStatusChanged
I (19223) chip[DL]: IPv4 Internet connectivity LOST
I (19223) chip[DL]: IPv6 Internet connectivity LOST
I (19243) chip[DL]: Attempting to connect WiFi station interface
I (19243) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (19253) wifi:I (19253) chip[DL]: Done driving station state, nothing else to do...
ap channel adjust o:1,1 n:6,2
I (19253) wifi:new:<6,0>, old:<1,1>, ap:<6,2>, sta:<6,0>, prof:1
I (19273) wifi:state: init -> auth (b0)
I (19273) chip[DL]: Done driving station state, nothing else to do...
I (19273) chip[DL]: Done driving station state, nothing else to do...
I (19283) wifi:state: auth -> assoc (0)
I (19293) chip[DL]: Done driving station state, nothing else to do...
I (19293) chip[DL]: WIFI_EVENT_AP_START
I (19303) wifi:I (19303) chip[DL]: WiFi AP state change: NotActive -> Active
state: assoc -> run (10)
I (19303) chip[DL]: Changing ESP WiFi mode: STA+AP -> STA
I (19323) wifi:mode : sta (b8:d6:1a:33:d0:e4)
I (19333) chip[DL]: WiFi AP state change: Active -> Deactivating
I (19333) chip[DL]: Updating advertising data
I (19333) chip[DL]: WIFI_EVENT_AP_STOP
I (19343) chip[DL]: WiFi AP state change: Deactivating -> NotActive
I (19363) wifi:connected with young, aid = 1, channel 6, BW20, bssid = 0e:ed:4c:f3:55:ff
I (19363) wifi:security: WPA2-PSK, phy: bgn, rssi: -52
I (19383) wifi:pm start, type: 1

I (19383) chip[DL]: WIFI_EVENT_STA_CONNECTED
I (19393) chip[DL]: WiFi station state change: Connecting -> Connecting_Succeeded
I (19393) chip[DL]: WiFi station state change: Connecting_Succeeded -> Connected
I (19403) chip[DL]: WiFi station interface connected
I (19403) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0030's Attribute 0x0000 is 16 **********
I (19423) chip[IN]: Prepared secure message 0x3ffe65b4 to 0xFFFFFFFB00000000 (1)  of type 0x9 and protocolId (0, 1) on exchange 41313r with MessageCounter:215812904.
I (19433) chip[IN]: Sending encrypted msg 0x3ffe65b4 with MessageCounter:215812904 to 0xFFFFFFFB00000000 (1) at monotonic time: 00000000000048BC msec
W (19433) wifi:<ba-add>idx:0 (ifx:0, 0e:ed:4c:f3:55:ff), tid:0, ssn:2, winSize:64I (19453) NimBLE: GATT procedure initiated: indicate; 

I (19463) NimBLE: att_handle=14

I (19463) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (19483) chip[ZCL]: WiFiDiagnosticsDelegate: OnConnectionStatusChanged
I (19483) chip[DL]: Done driving station state, nothing else to do...
I (19483) chip[DIS]: Advertise operational node 2160C831DCCAD7D4-0000000000000041
I (19503) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 2160C831DCCAD7D4-0000000000000041.
I (19513) chip[DIS]: mDNS service published: _matter._tcp
I (19513) chip[SVR]: Operational advertising enabled
I (19523) chip[DL]: Updating advertising data
I (19563) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (19633) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (20263) esp_netif_handlers: sta ip: 172.20.10.2, mask: 255.255.255.240, gw: 172.20.10.1
I (20263) chip[DL]: IP_EVENT_STA_GOT_IP
I (20263) chip[DL]: IPv4 address ready on WiFi station interface: 172.20.10.2/255.255.255.240 gateway 172.20.10.1
I (20283) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (20293) app_main: Interface IP Address changed
I (20293) chip[DIS]: Updating services using commissioning mode 0
I (20303) chip[DIS]: CHIP minimal mDNS started advertising.
I (20313) chip[DIS]: Advertise operational node 2160C831DCCAD7D4-0000000000000041
I (20313) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 2160C831DCCAD7D4-0000000000000041.
I (20333) chip[DIS]: mDNS service published: _matter._tcp
I (20333) ROUTE_HOOK: Hook already installed on netif, skip...
I (21263) chip[DL]: IP_EVENT_GOT_IP6
I (21263) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fe80:0000:0000:0000:bad6:1aff:fe33:d0e4
I (21273) app_main: Interface IP Address changed
I (21273) chip[DIS]: Updating services using commissioning mode 0
I (21273) chip[DIS]: CHIP minimal mDNS started advertising.
I (21293) chip[DIS]: Advertise operational node 2160C831DCCAD7D4-0000000000000041
I (21303) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 2160C831DCCAD7D4-0000000000000041.
I (21313) chip[DIS]: mDNS service published: _matter._tcp
I (21323) ROUTE_HOOK: Hook already installed on netif, skip...
I (22063) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (24263) chip[DL]: IP_EVENT_GOT_IP6
I (24263) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: 2408:8459:5060:166e:bad6:1aff:fe33:d0e4
I (24273) chip[DL]: IPv6 Internet connectivity ESTABLISHED
I (24273) app_main: Interface IP Address changed
I (24273) chip[DIS]: Updating services using commissioning mode 0
I (24293) chip[DIS]: CHIP minimal mDNS started advertising.
I (24303) chip[DIS]: Advertise operational node 2160C831DCCAD7D4-0000000000000041
I (24303) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 2160C831DCCAD7D4-0000000000000041.
I (24323) chip[DIS]: mDNS service published: _matter._tcp
I (24323) ROUTE_HOOK: Hook already installed on netif, skip...
I (24573) NimBLE: GATT procedure initiated: indicate; 
I (24573) NimBLE: att_handle=14

I (24573) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (24643) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (27163) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (29663) NimBLE: GATT procedure initiated: indicate; 
I (29673) NimBLE: att_handle=14

I (29673) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (29743) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (32203) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (34703) NimBLE: GATT procedure initiated: indicate; 
I (34713) NimBLE: att_handle=14

I (34713) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (34753) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (37243) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (39743) NimBLE: GATT procedure initiated: indicate; 
I (39753) NimBLE: att_handle=14

I (39753) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (39793) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
I (42343) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (44843) NimBLE: GATT procedure initiated: indicate; 
I (44853) NimBLE: att_handle=14

I (44853) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (44893) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14 
E (59863) chip[BLE]: ack recv timeout, closing ep 0x3ffcbf20
I (59863) chip[BLE]: Releasing end point's BLE connection back to application.
I (74563) chip[FS]: Fail-safe timer expired

thanks

dhrishi, 2022-10-31:
@linhr3 Can you please try out with the latest codebase? Also, please share the software version of the iPhone and the HomePod/Apple TV, so that we can try that out. Note that, you would require the hub to perform successful commissioning.

linhr3, 2022-11-02:
@dhrishi Hello
We tried with the lastest codebase but still failed.
Detail:
iphone12promax
system: ios 16.1
We do not have the HomePod/Apple TV and did not try with them. We tried with Espressif-Matter app (V1.0.2) from App Store. And the app even could not pair with the esp32 device this time with the lastest codebase.
Shoud we need the hub (HomePod or Apple TV) to perform the commission? We will buy a HomePod soon.

linhr3, 2022-11-03:
@dhrishi Hi
We added the board to Apple Home with a HomePod successfully. Thanks!
There is still a question that why we need a HomePod or a Apple TV as a hub to perform successful commissioning? What role does a hub play during commissioning? Thank you.

closed 2022-11-03 by linhr3