espressif/esp-matter #79: examples/light   chip tool Configuration failed

shun-dgut, 2022-11-03:
chip： esp32c3  
sdk :    commitid 669070a3d2f9d8cf3dd08cafe8a82ee316a4cf90

LOG：
[2022-11-03 12:21:13.335]# RECV ASCII>
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16bc
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
[0;32mI (30) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 18:46:54[0m
[0;32mI (30) boot: chip revision: 3[0m
[0;32mI (32) boot.esp32c3: SPI Speed      : 80MHz[0m
[0;32mI (37) boot.esp32c3: SPI Mode       : DIO[0m
[0;32mI (

[2022-11-03 12:21:13.336]# RECV ASCII>
42) boot.esp32c3: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Enabling RNG early entropy source...[0m
[0;32mI (52) boot: Partition Table:[0m
[0;32mI (56) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (63) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000[0m
[0;32mI (70) boot:  1 nvs              WiFi data        01 02 00010000 00006000[0m
[0;32mI (78) boot:  2 otadata          OTA data         01 00 00016000 00002000[0m
[0;32mI (85) boot:  3

[2022-11-03 12:21:13.342]# RECV ASCII>
 phy_init         RF data          01 01 00018000 00001000[0m
[0;32mI (93) boot:  4 ota_0            OTA app          00 10 00020000 001e0000[0m
[0;32mI (100) boot:  5 ota_1            OTA app          00 11 00200000 001e0000[0m
[0;32mI (108) boot:  6 fctry            WiFi data        01 02 003e0000 00006000[0m
[0;32mI (115) boot: End of partition table[0m
[0;32mI (120) boot: No factory image, trying OTA 0[0m
[0;32mI (125) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=2eee0h (19

[2022-11-03 12:21:13.343]# RECV ASCII>
2224) map[0m
[0;32mI (163) esp_image: segment 1: paddr=0004ef08 vaddr=3fc90400 size=01110h (  4368) load[0m
[0;32mI (164) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=f4de4h (1002980) map[0m

[2022-11-03 12:21:13.377]# RECV ASCII>
[0;32mI (326) esp_image: segment 3: paddr=00144e0c vaddr=3fc91510 size=02590h (  9616) load[0m
[0;32mI (328) esp_image: segment 4: paddr=001473a4 vaddr=40380000 size=1033ch ( 66364) load[0m
[0;32mI (343) esp_image: segment 5: paddr=001576e8 vaddr=50000010 size=00010h (    16) load[0m
[0;32mI (349) boot: Loaded app from partition at offset 0x20000[0m

[2022-11-03 12:21:13.433]# RECV ASCII>
[0;32mI (410) boot: Set actual ota_seq=1 in otadata[0][0m
[0;32mI (410) boot: Disabling RNG early entropy source...[0m
[0;32mI (421) cpu_start: Pro cpu up.[0m
[0;32mI (429) cpu_start: Pro cpu start user code[0m
[0;32mI (429) cpu_start: cpu freq: 160000000[0m
[0;32mI (429) cpu_start: Application information:[0m
[0;32mI (432) cpu_start: Project name:     light[0m
[0;32mI (437) cpu_start: App version:      v1.0[0m
[0;32mI (442) cpu_start: Compile time:     Oct 28 2022 18:46:47[0m
[0

[2022-11-03 12:21:13.478]# RECV ASCII>
;32mI (448) cpu_start: ELF file SHA256:  a19d7d05878f65b8...[0m
[0;32mI (454) cpu_start: ESP-IDF:          v4.4.2-dirty[0m
[0;32mI (459) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (466) heap_init: At 3FCA4650 len 0001B9B0 (110 KiB): DRAM[0m
[0;32mI (472) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM[0m
[0;32mI (479) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM[0m
[... 771 lines cut ...]
[0;32mI (17942) app_main: Commissioning complete[0m
[0;32mI (17952) NimBLE: GAP procedure initiated: stop advertising.
[0m
[0;32mI (17952) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19
[0m
[0;32mI (17872) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 0[0m
[0;32mI (17972) chip[DL]: CHIPoBLE unsubscribe received[0m
[0;32mI (17972) chip[D

[2022-11-03 12:21:31.004]# RECV ASCII>
L]: BLE GAP connection terminated (con 1 reason 0x213)[0m
[0;33mW (17982) Timer: Timer not stopped[0m
[0;33mW (17982) Timer: Timer not stopped[0m
[0;33mW (18002) BTDM_INIT: esp_bt_mem_release not implemented, return OK[0m
[0;32mI (18002) esp_matter_core: BLE deinit successful and memory reclaimed[0m
[0;31mE (18012) chip[DL]: Long dispatch time: 137 ms, for event type 32781[0m
[0;32mI (18012) chip[EM]: >>> [E:58290r M:120997418 (Ack:44921704)] (S) Msg RX from 1:000000000001B669 [158C] --- T

[2022-11-03 12:21:31.065]# RECV ASCII>
ype 0000:10 (SecureChannel:StandaloneAck)[0m
[0;32mI (18022) app_main: Commissioning window closed[0m
[0;31mE (18032) chip[BLE]: no endpoint for unsub recvd[0m

[2022-11-03 12:21:51.173]# RECV ASCII>
W (38132) wifi:<ba-add>idx:0 (ifx:0, 86:1b:77:95:f6:32), tid:0, ssn:10, winSize:64

shubhamdp, 2022-11-03:
@shun-dgut Logs says that commissioning was successful.
> �[0;32mI (17872) chip[SVR]: Commissioning completed successfully�[0m

Can you elaborate a bit more on "chip tool Configuration failed"?

A minor suggestion: May be for next issues, you can put the logs into file or put them under [<detail>](https://gist.github.com/scmx/eca72d44afee0113ceb0349dd54a84a2).

shun-dgut, 2022-11-03:
> @shun-dgut Logs says that commissioning was successful.
> 
> > �[0;32mI (17872) chip[SVR]: Commissioning completed successfully�[0m
> 
> Can you elaborate a bit more on "chip tool Configuration failed"?
> 
> A minor suggestion: May be for next issues, you can put the logs into file or put them under .

Thanks for your reply, but it seems out of control. I tried adding with home in ios 16.1, ca
[matter send cmd.txt](https://github.com/espressif/esp-matter/files/9926719/matter.send.cmd.txt)
n connect to the same network, but can't control it, home display doesn't respond.
This is the control log of the chip tool.

shun-dgut, 2022-11-03:
> @shun-dgut Logs says that commissioning was successful.
> 
> > �[0;32mI (17872) chip[SVR]: Commissioning completed successfully�[0m
> 
> Can you elaborate a bit more on "chip tool Configuration failed"?
> 
> A minor suggestion: May be for next issues, you can put the logs into file or put them under .

This is the control log of the chip tool
[matter send cmd.txt](https://github.com/espressif/esp-matter/files/9926730/matter.send.cmd.txt)

shun-dgut, 2022-11-03:
I also tried adding devices using ios16.1 home, devices can be added but not controlled

shubhamdp, 2022-11-03:
@shun-dgut Looks like you have attached an empty log file. Can you please re-attach the one with logs?

> I also tried adding devices using ios16.1 home, devices can be added but not controlled

Did you use this along with Apple HomePod or Apple TV? Because, it would be required for controlling the devices.

shun-dgut, 2022-11-03:
[matter send cmd.txt](https://github.com/espressif/esp-matter/files/9926823/matter.send.cmd.txt)

Sorry, this is the log. 
Matter over wifi can't communicate directly with home?

shun-dgut, 2022-11-03:
> @shun-dgut Looks like you have attached an empty log file. Can you please re-attach the one with logs?
>
>
> Did you use this along with Apple HomePod or Apple TV? Because, it would be required for [...]
[matter send cmd.txt](https://github.com/espressif/esp-matter/files/9926845/matter.send.cmd.txt)
s.

Sorry, this is the log.
Matter over wifi can't communicate directly with home?

shubhamdp, 2022-11-03:
> Matter over wifi can't communicate directly with home?

Please take a look at this thread https://github.com/espressif/esp-matter/issues/78#issuecomment-1296698698

dhrishi, 2022-11-04:
@shun-dgut Please close this if your issue is resolved

dhrishi, 2022-11-21:
@shun-dgut Closing this as the original issue is addressed. Please create a new one in case you see any problems

closed 2022-11-21 by dhrishi