espressif/esp-matter #80: chip-tool onoff toggle failed

sidhoo-root, 2022-11-03:
steps:
1../chip-tool pairing ble-wifi 1234 Redmi_huan 87654321 38802273 2481
2../chip-tool onoff toggle 1234 1

efsz@huxu-ubnb:~/esp/esp-matter/connectedhomeip/connectedhomeip/SOME-PATH$ ./chip-tool pairing ble-wifi 1234 Redmi_huan 87654321 38802273 2481
[1667469144.149246][197684:197684] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1667469144.149327][197684:197684] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1667469144.149345][197684:197684] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1667469144.149355][197684:197684] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1667469144.149421][197684:197684] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-DzTng9)
[1667469144.149551][197684:197684] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1667469144.149565][197684:197684] CHIP:DL: NVS set: chip-counters/reboot-count = 4 (0x4)
[1667469144.149904][197684:197684] CHIP:DL: Got Ethernet interface: eno1
[1667469144.150174][197684:197684] CHIP:DL: Found the primary Ethernet interface:eno1
[1667469144.150313][197684:197684] CHIP:DL: Got WiFi interface: wlp4s0
[1667469144.150607][197684:197684] CHIP:DL: Found the primary WiFi interface:wlp4s0
[1667469144.150625][197684:197684] CHIP:IN: UDP::Init bind&listen port=0
[1667469144.150639][197684:197684] CHIP:IN: UDP::Init bound to port=45963
[1667469144.150641][197684:197684] CHIP:IN: UDP::Init bind&listen port=0
[1667469144.150648][197684:197684] CHIP:IN: UDP::Init bound to port=36752
[1667469144.150650][197684:197684] CHIP:IN: BLEBase::Init - setting/overriding transport
[1667469144.150651][197684:197684] CHIP:IN: TransportMgr initialized
[1667469144.150656][197684:197684] CHIP:FP: Initializing FabricTable from persistent storage
[1667469144.150681][197684:197684] CHIP:TS: Last Known Good Time: 2022-11-03T16:57:51
[1667469144.150775][197684:197684] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x2C4C9DB5C55EB610, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1667469144.153551][197684:197684] CHIP:ZCL: Using ZAP configuration...
[1667469144.154278][197684:197684] CHIP:DL: MDNS failed to join multicast group on eno1 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found
[1667469144.155204][197684:197684] CHIP:CTL: System State Initialized...
[1667469144.155216][197684:197684] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1667469144.155226][197684:197684] CHIP:CTL: Setting attestation nonce to random value
[1667469144.155231][197684:197684] CHIP:CTL: Setting CSR nonce to random value
[1667469144.155241][197684:197684] CHIP:IN: UDP::Init bind&listen port=5550
[1667469144.155248][197684:197684] CHIP:IN: UDP::Init bound to port=5550
[1667469144.155250][197684:197684] CHIP:IN: UDP::Init bind&listen port=5550
[1667469144.155254][197684:197684] CHIP:IN: UDP::Init bound to port=5550
[1667469144.155256][197684:197684] CHIP:IN: TransportMgr initialized
[1667469144.155294][197684:197689] CHIP:DL: CHIP task running
[1667469144.155376][197684:197689] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1667469144.155384][197684:197689] CHIP:CTL: Setting attestation nonce to random value
[1667469144.155395][197684:197689] CHIP:CTL: Setting CSR nonce to random value
[1667469144.155933][197684:197689] CHIP:CTL: Generating NOC
[1667469144.156142][197684:197689] CHIP:FP: Validating NOC chain
[1667469144.156503][197684:197689] CHIP:FP: NOC chain validation successful
[1667469144.156536][197684:197689] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1667469144.156541][197684:197689] CHIP:TS: Last Known Good Time: 2022-11-03T16:57:51
[1667469144.156545][197684:197689] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1667469144.156547][197684:197689] CHIP:TS: Retaining current Last Known Good Time
[1667469144.156841][197684:197689] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1667469144.157123][197684:197689] CHIP:TS: Committing Last Known Good Time to storage: 2022-11-03T16:57:51
[1667469144.157475][197684:197689] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1667469144.157488][197684:197689] CHIP:IN: UDP::Init bind&listen port=5550
[1667469144.157506][197684:197689] CHIP:IN: UDP::Init bound to port=5550
[1667469144.157511][197684:197689] CHIP:IN: UDP::Init bind&listen port=5550
[1667469144.157523][197684:197689] CHIP:IN: UDP::Init bound to port=5550
[1667469144.157527][197684:197689] CHIP:IN: TransportMgr initialized
[1667469144.159755][197684:197689] CHIP:IN: SecureSession[0x7fc4bc021900]: Allocated Type:1 LSID:59877
[1667469144.159769][197684:197689] CHIP:SC: Assigned local session key ID 59877
[1667469144.159789][197684:197689] CHIP:SC: Including MRP parameters in PBKDF param request
[1667469144.159808][197684:197689] CHIP:EM: <<< [E:44399i M:182748376] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
[1667469144.159820][197684:197689] CHIP:IN: (U) Sending msg 182748376 to IP address 'BLE'
[1667469144.159828][197684:197689] CHIP:IN: Message appended to BLE send queue
[1667469144.159832][197684:197689] CHIP:SC: Sent PBKDF param request
[1667469144.159838][197684:197689] CHIP:CTL: Setting wifi credentials from parameters
[1667469144.159842][197684:197689] CHIP:CTL: Setting attestation nonce to random value
[1667469144.159848][197684:197689] CHIP:CTL: Setting CSR nonce to random value
[1667469144.159855][197684:197689] CHIP:CTL: Commission called for node ID 0x00000000000004D2
[1667469144.159931][197684:197690] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1667469144.159983][197684:197687] CHIP:DL: TRACE: Bus acquired for name C-0434
[1667469144.170018][197684:197689] CHIP:DL: PlatformBlueZInit init success
[1667469144.176577][197684:197687] CHIP:BLE: BLE removing known devices.
[... 557 lines cut ...]
[1667469149.431579][197684:197684] CHIP:CTL: Shutting down the commissioner
[1667469149.431591][197684:197684] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1667469149.431633][197684:197684] CHIP:CTL: Shutting down the controller
[1667469149.431645][197684:197684] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1667469149.431654][197684:197684] CHIP:FP: Forgetting fabric 0x1
[1667469149.431666][197684:197684] CHIP:TS: Pending Last Known Good Time: 2022-11-03T16:57:51
[1667469149.431734][197684:197684] CHIP:TS: Previous Last Known Good Time: 2022-11-03T16:57:51
[1667469149.431743][197684:197684] CHIP:TS: Reverted Last Known Good Time to previous value
[1667469149.431758][197684:197684] CHIP:CTL: Shutting down the commissioner
[1667469149.431764][197684:197684] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1667469149.431783][197684:197684] CHIP:CTL: Shutting down the controller
[1667469149.431788][197684:197684] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1667469149.432086][197684:197684] CHIP:DMG: IM WH moving to [Uninitialized]
[1667469149.432094][197684:197684] CHIP:DMG: IM WH moving to [Uninitialized]
[1667469149.432099][197684:197684] CHIP:DMG: IM WH moving to [Uninitialized]
[1667469149.432102][197684:197684] CHIP:DMG: IM WH moving to [Uninitialized]
[1667469149.432109][197684:197684] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1667469149.432129][197684:197684] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1667469149.432271][197684:197684] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-5kBjq8)
[1667469149.432470][197684:197684] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1667469149.432487][197684:197684] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1667469149.432492][197684:197684] CHIP:DL: Inet Layer shutdown
[1667469149.432495][197684:197684] CHIP:DL: BLE shutdown
[1667469149.433225][197684:197684] CHIP:DL: System Layer shutdown
[1667469149.433251][197684:197684] CHIP:TOO: Run command failure: ../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1031: CHIP Error 0x000000AC: Internal error

sidhoo-root, 2022-11-03:
efsz@huxu-ubnb:~/esp/esp-matter/connectedhomeip/connectedhomeip/SOME-PATH$ ./chip-tool onoff toggle 1234 1
[1667469153.974623][198816:198816] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1667469153.974712][198816:198816] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1667469153.974732][198816:198816] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1667469153.974745][198816:198816] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1667469153.974832][198816:198816] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-kzbzzK)
[1667469153.974958][198816:198816] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1667469153.974970][198816:198816] CHIP:DL: NVS set: chip-counters/reboot-count = 5 (0x5)
[1667469153.975331][198816:198816] CHIP:DL: Got Ethernet interface: eno1
[1667469153.975610][198816:198816] CHIP:DL: Found the primary Ethernet interface:eno1
[1667469153.975891][198816:198816] CHIP:DL: Got WiFi interface: wlp4s0
[1667469153.976180][198816:198816] CHIP:DL: Found the primary WiFi interface:wlp4s0
[1667469153.976200][198816:198816] CHIP:IN: UDP::Init bind&listen port=0
[1667469153.976215][198816:198816] CHIP:IN: UDP::Init bound to port=46732
[1667469153.976218][198816:198816] CHIP:IN: UDP::Init bind&listen port=0
[1667469153.976226][198816:198816] CHIP:IN: UDP::Init bound to port=32776
[1667469153.976228][198816:198816] CHIP:IN: BLEBase::Init - setting/overriding transport
[1667469153.976230][198816:198816] CHIP:IN: TransportMgr initialized
[1667469153.976236][198816:198816] CHIP:FP: Initializing FabricTable from persistent storage
[1667469153.976267][198816:198816] CHIP:TS: Last Known Good Time: 2022-11-03T16:57:51
[1667469153.976394][198816:198816] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x2C4C9DB5C55EB610, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1667469153.978885][198816:198816] CHIP:ZCL: Using ZAP configuration...
[1667469153.979587][198816:198816] CHIP:DL: MDNS failed to join multicast group on eno1 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found
[1667469153.980538][198816:198816] CHIP:CTL: System State Initialized...
[1667469153.980557][198816:198816] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1667469153.980575][198816:198816] CHIP:CTL: Setting attestation nonce to random value
[1667469153.980585][198816:198816] CHIP:CTL: Setting CSR nonce to random value
[1667469153.980602][198816:198816] CHIP:IN: UDP::Init bind&listen port=5550
[1667469153.980615][198816:198816] CHIP:IN: UDP::Init bound to port=5550
[1667469153.980620][198816:198816] CHIP:IN: UDP::Init bind&listen port=5550
[1667469153.980629][198816:198816] CHIP:IN: UDP::Init bound to port=5550
[1667469153.980632][198816:198816] CHIP:IN: TransportMgr initialized
[1667469153.980703][198816:198821] CHIP:DL: CHIP task running
[1667469153.980841][198816:198821] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1667469153.980856][198816:198821] CHIP:CTL: Setting attestation nonce to random value
[1667469153.980875][198816:198821] CHIP:CTL: Setting CSR nonce to random value
[1667469153.981533][198816:198821] CHIP:CTL: Generating NOC
[1667469153.981754][198816:198821] CHIP:FP: Validating NOC chain
[1667469153.982129][198816:198821] CHIP:FP: NOC chain validation successful
[1667469153.982164][198816:198821] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1667469153.982169][198816:198821] CHIP:TS: Last Known Good Time: 2022-11-03T16:57:51
[1667469153.982171][198816:198821] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1667469153.982173][198816:198821] CHIP:TS: Retaining current Last Known Good Time
[1667469153.982477][198816:198821] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1667469153.982731][198816:198821] CHIP:TS: Committing Last Known Good Time to storage: 2022-11-03T16:57:51
[1667469153.982959][198816:198821] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1667469153.982967][198816:198821] CHIP:IN: UDP::Init bind&listen port=5550
[1667469153.982979][198816:198821] CHIP:IN: UDP::Init bound to port=5550
[1667469153.982982][198816:198821] CHIP:IN: UDP::Init bind&listen port=5550
[1667469153.982992][198816:198821] CHIP:IN: UDP::Init bound to port=5550
[1667469153.982996][198816:198821] CHIP:IN: TransportMgr initialized
[1667469153.985226][198816:198821] CHIP:TOO: Sending command to node 0x4d2
[1667469153.985292][198816:198821] CHIP:CSM: FindOrEstablishSession: PeerId = [1:00000000000004D2]
[1667469153.985297][198816:198821] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1667469153.985303][198816:198821] CHIP:CTL: OperationalSessionSetup[1:00000000000004D2]: State change 1 --> 2
[1667469153.985358][198816:198821] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on eno1:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:409: OS Error 0x02000065: Network is unreachable
[1667469153.985459][198816:198821] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
[1667469154.185011][198816:198821] CHIP:DIS: Checking node lookup status after 200 ms
[... cut ...]
[1667469156.986293][198816:198821] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on eno1:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:409: OS Error 0x02000065: Network is unreachable
[1667469156.986568][198816:198821] CHIP:DIS: mDNS broadcast had only partial success: 4 successes and 1 failures.
[1667469160.986345][198816:198821] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on eno1:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:409: OS Error 0x02000065: Network is unreachable
[1667469160.986631][198816:198821] CHIP:DIS: mDNS broadcast had only partial success: 4 successes and 1 failures.

dhrishi, 2022-11-03:
@sidhoo-root Please provide more details:
1. Commit id used - esp-matter as well as the submodule connectedhomeip. Please make sure that it is the latest
2. Is chip-tool compiled on the same commit as the esp-matter example you are trying? If not, please recompile as the error that is seen hints at the inconsistency.

sidhoo-root, 2022-11-03:
@dhrishi 
idf branch : v4.4.2
esp-matter commit id :  3401f142664c76172365a49503c83c47ca568051 (grafted, HEAD -> main, origin/main, origin/HEAD)
connectedhomeip commit id : commit 87bee4de25886ba00f1a177cd0f8791a48528a1f (grafted, HEAD)
                                                            Author: Shubham Patil <shubham.patil@espressif.com>
                                                            Date:   Mon Oct 17 21:07:08 2022 +0530

jonsmirl, 2022-11-04:
Your chip-tool and device are on different Matter releases.
```
[1667469149.292796][197684:197689] CHIP:CTL: Failed in verifying 'Attestation Information' command received from the device: err 601. Look at AttestationVerificationResult enum to understand the errors
[1667469149.292826][197684:197689] CHIP:CTL: Error on commissioning step 'AttestationVerification': '../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1031: CHIP Error 0x000000AC: Internal error'
[1667469149.292835][197684:197689] CHIP:CTL: Failed to perform commissioning step 8
[1667469149.292844][197684:197689] CHIP:CTL: Going from commissioning step 'AttestationVerification' with lastErr = '../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1031: CHIP Error 0x000000AC: Internal error' -> 'Cleanup'
[1667469149.292859][197684:197689] CHIP:CTL: Performing next commissioning step 'Cleanup' with completion status = '../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1031: CHIP Error 0x000000AC: Internal error'
[1667469149.292867][197684:197689] CHIP:CTL: Expiring failsafe on proxy 0x7fc4bc024130
[1667469149.292915][197684:197689] CHIP:DMG: ICR moving to [AddingComm]
```

You are in this situation.  https://github.com/google-home/sample-app-for-matter-android/issues/29

sidhoo-root, 2022-11-04:
@jonsmirl Should i flash the manufacturing binary again ?
and there is a new Failed report : [1667550728.773151][1631208:1631211] CHIP:BLE: Device 1C:52:16:B9:A1:10 does not look like a CHIP device.

dhrishi, 2022-11-04:
@sidhoo22 Clean and re-compile chip-tool with the latest esp-matter code base and try with the light firmware on the same codebase, so that both are in sync.

dhrishi, 2022-11-09:
@sidhoo22 Did it work? Please close the issue if it did

sidhoo22, 2022-11-10:
@dhrishi no, it not work  i will post the log later, thanks for your patience

dhrishi, 2022-11-10:
Please follow https://github.com/espressif/esp-matter/issues/80#issuecomment-1303126211 and if it still doesn't work then share the logs on both chip-tool and ESP32 side along with the commit ids on which both of these have been compiled.

sidhoo22, 2022-11-16:
1. chip-tool commit id ,
> ~/workspace/esp-matter/connectedhomeip/connectedhomeip/BUILD_PATH$ git log -1
> commit 87bee4de25886ba00f1a177cd0f8791a48528a1f (HEAD)
> Author: Shubham Patil <shubham.patil@espressif.com>
> Date:   Mon Oct 17 21:07:08 2022 +0530
2. esp-matter commit id,

dhrishi, 2022-11-16:
@sidhoo22 Your chip-tool should be connected to the same Wi-Fi network to which you are connecting the ESP32 to. Can you confirm if you are following this? From the device log, it is seen that after connecting to Wi-Fi it starts advertisement over mDNS but the other side (chip-tool) does not discover it and send Sigma1 CASE message.

The chip-tool log you have shared does not seem to be the correct one. In it, it is not able to find the device.

sidhoo22, 2022-11-16:
@dhrishi yes, let chiptool connet to the wifi router(ipv6) which the matter dev conneted. and run the onoff cmd, log as follow

> ~/workspace/esp-matter/connectedhomeip/connectedhomeip/BUILD_PATH$ ./chip-tool onoff toggle 1234 0x1
> [1668593207.744985][1789038:1789038] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
> [1668593207.745354][1789038:1789038] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
> [1668593207.745417][1789038:1789038] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
> [1668593207.745470][1789038:1789038] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
> [1668593207.745639][1789038:1789038] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-k2BMzT)
> [1668593207.746008][1789038:1789038] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1668593207.746085][1789038:1789038] CHIP:DL: NVS set: chip-counters/reboot-count = 27 (0x1B)
> [1668593207.746611][1789038:1789038] CHIP:DL: Got Ethernet interface: eno1
> [1668593207.746981][1789038:1789038] CHIP:DL: Found the primary Ethernet interface:eno1
> [1668593207.747391][1789038:1789038] CHIP:DL: Got WiFi interface: wlp4s0
> [1668593207.747742][1789038:1789038] CHIP:DL: Found the primary WiFi interface:wlp4s0
> [1668593207.747814][1789038:1789038] CHIP:IN: UDP::Init bind&listen port=0
> [1668593207.747850][1789038:1789038] CHIP:IN: UDP::Init bound to port=35115
> [1668593207.747856][1789038:1789038] CHIP:IN: UDP::Init bind&listen port=0
> [1668593207.747874][1789038:1789038] CHIP:IN: UDP::Init bound to port=47037
> [1668593207.747879][1789038:1789038] CHIP:IN: BLEBase::Init - setting/overriding transport
> [1668593207.747883][1789038:1789038] CHIP:IN: TransportMgr initialized
> [1668593207.747894][1789038:1789038] CHIP:FP: Initializing FabricTable from persistent storage
> [1668593207.747963][1789038:1789038] CHIP:TS: Last Known Good Time: 2022-11-16T17:15:34
> [1668593207.748089][1789038:1789038] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xCC35586E4C89C1E3, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
> [1668593207.749936][1789038:1789038] CHIP:ZCL: Using ZAP configuration...
> [1668593207.751338][1789038:1789038] CHIP:DL: MDNS failed to join multicast group on eno1 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found
> [1668593207.752424][1789038:1789038] CHIP:CTL: System State Initialized...
> [1668593207.752464][1789038:1789038] CHIP:CTL: Stopping commissioning discovery over DNS-SD
> [1668593207.752511][1789038:1789038] CHIP:CTL: Setting attestation nonce to random value
> [1668593207.752531][1789038:1789038] CHIP:CTL: Setting CSR nonce to random value
> [1668593207.752560][1789038:1789038] CHIP:IN: UDP::Init bind&listen port=5550
> [1668593207.752594][1789038:1789038] CHIP:IN: UDP::Init bound to port=5550
> [1668593207.752602][1789038:1789038] CHIP:IN: UDP::Init bind&listen port=5550
> [1668593207.752617][1789038:1789038] CHIP:IN: UDP::Init bound to port=5550
> [1668593207.752621][1789038:1789038] CHIP:IN: TransportMgr initialized
> [1668593207.752834][1789038:1789043] CHIP:DL: CHIP task running
> [1668593207.753144][1789038:1789043] CHIP:CTL: Stopping commissioning discovery over DNS-SD
> [1668593207.753161][1789038:1789043] CHIP:CTL: Setting attestation nonce to random value
> [1668593207.753181][1789038:1789043] CHIP:CTL: Setting CSR nonce to random value
> [1668593207.753857][1789038:1789043] CHIP:CTL: Generating NOC
> [1668593207.754221][1789038:1789043] CHIP:FP: Validating NOC chain
> [1668593207.754631][1789038:1789043] CHIP:FP: NOC chain validation successful
> [1668593207.754694][1789038:1789043] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
> [1668593207.754705][1789038:1789043] CHIP:TS: Last Known Good Time: 2022-11-16T17:15:34
> [1668593207.754709][1789038:1789043] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
> [1668593207.754713][1789038:1789043] CHIP:TS: Retaining current Last Known Good Time
> [1668593207.755431][1789038:1789043] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
> [1668593207.755750][1789038:1789043] CHIP:TS: Committing Last Known Good Time to storage: 2022-11-16T17:15:34
> [1668593207.756029][1789038:1789043] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
> [1668593207.756046][1789038:1789043] CHIP:IN: UDP::Init bind&listen port=5550
> [1668593207.756076][1789038:1789043] CHIP:IN: UDP::Init bound to port=5550
> [1668593207.756079][1789038:1789043] CHIP:IN: UDP::Init bind&listen port=5550
> [1668593207.756092][1789038:1789043] CHIP:IN: UDP::Init bound to port=5550
> [1668593207.756095][1789038:1789043] CHIP:IN: TransportMgr initialized
> [1668593207.757430][1789038:1789043] CHIP:TOO: Sending command to node 0x4d2
> [1668593207.757554][1789038:1789043] CHIP:CSM: FindOrEstablishSession: PeerId = [1:00000000000004D2]
> [1668593207.757561][1789038:1789043] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
> [1668593207.757574][1789038:1789043] CHIP:CTL: OperationalSessionSetup[1:00000000000004D2]: State change 1 --> 2
[... cut ...]
> [1668593222.768734][1789038:1789038] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
> [1668593222.768747][1789038:1789038] CHIP:DL: Inet Layer shutdown
> [1668593222.768754][1789038:1789038] CHIP:DL: BLE shutdown
> [1668593222.768764][1789038:1789038] CHIP:DL: System Layer shutdown
> [1668593222.768833][1789038:1789038] CHIP:TOO: Run command failure: ../examples/chip-tool/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout

sidhoo22, 2022-11-17:
@jonsmirl i hv no idea could you help me :) thanks

sidhoo22, 2022-11-18:
@dhrishi problem fixed. cuase the chiptool meachine netif eno1 is not connect with the wifi router, i down (ifconfig eno1 down) the netif , and repairing the device , and it works ! thanks for your help

closed 2022-11-18 by sidhoo22