espressif/esp-matter #81: set-target esp32c3 error

Firesuiry, 2022-11-04:
https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html
I follow the document and fail when set-target esp32c3
the error is :
`
:~/esp/light$ idf.py set-target esp32

Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/yinsy/esp/light/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/yinsy/esp/light/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 /home/yinsy/esp/light"...
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/yinsy/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/yinsy/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/yinsy/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Solving dependencies requirements
Updating lock file at /home/yinsy/esp/light/dependencies.lock
Processing 3 dependencies:
[1/3] espressif/button (2.0.1)
[2/3] espressif/led_strip (1.1.0-alpha)
[3/3] idf (4.4.1)
-- Found Python3: /home/yinsy/.espressif/python_env/idf4.4_py3.10_env/bin/python3.10 (found version "3.10.6") found components: Interpreter 
CMake Error at /home/yinsy/esp/esp-idf/tools/cmake/build.cmake:200 (message):
  Failed to resolve component 'app_reset'.
Call Stack (most recent call first):
  /home/yinsy/esp/esp-idf/tools/cmake/build.cmake:231 (__build_resolve_and_add_req)
  /home/yinsy/esp/esp-idf/tools/cmake/build.cmake:512 (__build_expand_requirements)
  /home/yinsy/esp/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:41 (project)
`

how can i solve this error?

jonsmirl, 2022-11-04:
When you copied out of tree you probably did not copy everything, it is here:
https://github.com/espressif/esp-matter/tree/main/examples/common

Firesuiry, 2022-11-04:
> 

Thank you a lot!
I copy all the example dir and it works!

closed 2022-11-04 by Firesuiry