espressif/esp-matter #82: How to use chip-tool?

Firesuiry, 2022-11-05:
I have built a example light with esp32c3
I want to connect it with chip-tool
But it need setup_passcode and I dont know where to get it.
I try it but it seems connecting fail.

The log from esp32c3 is as follows:
'''
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
I (30) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (30) boot: compile time 22:00:44
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (41) boot.esp32c3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (70) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (77) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (85) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (92) boot:  4 ota_0            OTA app          00 10 00020000 001e0000
I (100) boot:  5 ota_1            OTA app          00 11 00200000 001e0000
I (107) boot:  6 fctry            WiFi data        01 02 003e0000 00006000
I (115) boot: End of partition table
I (119) boot: No factory image, trying OTA 0
I (124) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=2efd0h (192464) map
I (163) esp_image: segment 1: paddr=0004eff8 vaddr=3fc90400 size=01020h (  4128) load
I (164) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=f4e50h (1003088) map
I (325) esp_image: segment 3: paddr=00144e78 vaddr=3fc91420 size=02680h (  9856) load
I (327) esp_image: segment 4: paddr=00147500 vaddr=40380000 size=1033ch ( 66364) load
I (343) esp_image: segment 5: paddr=00157844 vaddr=50000010 size=00010h (    16) load
I (348) boot: Loaded app from partition at offset 0x20000
I (393) boot: Set actual ota_seq=1 in otadata[0]
I (393) boot: Disabling RNG early entropy source...
I (404) cpu_start: Pro cpu up.
I (413) cpu_start: Pro cpu start user code
I (413) cpu_start: cpu freq: 160000000
I (413) cpu_start: Application information:
I (416) cpu_start: Project name:     light
I (420) cpu_start: App version:      v1.0
I (425) cpu_start: Compile time:     Nov  5 2022 13:50:42
I (431) cpu_start: ELF file SHA256:  1ca4d5ca67212762...
I (437) cpu_start: ESP-IDF:          v4.4.2
I (442) heap_init: Initializing. RAM available for dynamic allocation:
I (449) heap_init: At 3FCA4650 len 0001B9B0 (110 KiB): DRAM
I (455) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (462) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (469) spi_flash: detected chip: generic
I (473) spi_flash: flash io: dio
I (478) sleep: Configure to isolate all GPIO pins in sleep state
I (484) sleep: Enable automatic switching of GPIO sleep configuration
I (491) coexist: coexist rom version 9387209
I (496) cpu_start: Starting scheduler.
I (523) led_driver_ws2812: Initializing light driver
I (523) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (533) app_main: Light created with endpoint_id 1
E (533) esp_matter_core: Command 0x0047 on cluster 0x0300 already exists. Not creating again.
I (543) pp: pp rom version: 9387209
I (543) net80211: net80211 rom version: 9387209
[... 190 lines cut ...]
[1667627929.187336][18807:18812] CHIP:FP: Validating NOC chain
[1667627929.188357][18807:18812] CHIP:FP: NOC chain validation successful
[1667627929.188454][18807:18812] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1667627929.188466][18807:18812] CHIP:TS: Last Known Good Time: 2022-11-04T16:11:17
[1667627929.188475][18807:18812] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1667627929.188482][18807:18812] CHIP:TS: Retaining current Last Known Good Time
[1667627929.189216][18807:18812] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1667627929.189852][18807:18812] CHIP:TS: Committing Last Known Good Time to storage: 2022-11-04T16:11:17
[1667627929.190445][18807:18812] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1667627929.190465][18807:18812] CHIP:IN: UDP::Init bind&listen port=5550
[1667627929.190501][18807:18812] CHIP:IN: UDP::Init bound to port=5550
[1667627929.190510][18807:18812] CHIP:IN: UDP::Init bind&listen port=5550
[1667627929.190530][18807:18812] CHIP:IN: UDP::Init bound to port=5550
[1667627929.190537][18807:18812] CHIP:IN: TransportMgr initialized
[1667627960.227283][18807:18812] CHIP:DIS: Timeout waiting for mDNS resolution.
[1667627981.110249][18807:18811] CHIP:DL: No wifi interface name. Ignoring IP update event.
[1667628049.186089][18807:18807] CHIP:TOO: Run command failure: ../commands/common/CHIPCommand.cpp:513: CHIP Error 0x00000032: Timeout
[1667628068.031544][18807:18811] CHIP:DL: No wifi interface name. Ignoring IP update event.
[1667628150.053108][18807:18811] CHIP:DL: No wifi interface name. Ignoring IP update event.
[1667628232.065105][18807:18811] CHIP:DL: No wifi interface name. Ignoring IP update event.
[1667628316.106758][18807:18811] CHIP:DL: No wifi interface name. Ignoring IP update event.
[1667628379.658315][18807:18811] CHIP:DL: No wifi interface name. Ignoring IP update event.
[1667628388.678755][18807:18811] CHIP:DL: No wifi interface name. Ignoring IP update event.
[1667628470.700702][18807:18811] CHIP:DL: No wifi interface name. Ignoring IP update event.
'''

dhrishi, 2022-11-05:
@Firesuiry If it is the default example from the repository, you can find the documentation for commissioning using chip-tool [here](https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#commissioning)

Firesuiry, 2022-11-05:
I follow this but the connect seems failing, the log is at the bottom of the issue~

dhrishi, 2022-11-05:
@Firesuiry Looks like some issue with the chip-tool Wi-Fi adapter:

>[1667628573.022205][18807:18812] CHIP:BLE: No adapter available for new connection establishment

If the Wi-Fi interface is available and working fine, then please start afresh:

Reset to factory the device by entering the following command on the device (ESP32) console:
`matter esp factoryreset`

Chip-tool command:
```
$ chip-tool interactive start
$ pairing ble-wifi 0x7283 <ssid> <passphrase> 20202021 3840
```

Firesuiry, 2022-11-05:
> @Firesuiry Looks like some issue with the chip-tool BLE adapter:
> 
> > [1667628573.022205][18807:18812] CHIP:BLE: No adapter available for new connection establishment
> 
> If the Bluetooth interface is available and working fine, then please start afresh:
> 
> Reset to factory the device by entering the following command on the device (ESP32) console: `matter esp factoryreset`
> 
> Chip-tool command:
> 
> ```
> $ chip-tool interactive start
> $ pairing ble-wifi 0x7283 <ssid> <passphrase> 20202021 3840
> ```
Thank you for your reply~
There are still 3 problems:
1. My PC don`t have a BLE adapter, should I buy one to connect it?
2. the ESP32C3 has joined the wifi, can I connect it with "pairing onnetwork 0x11 20202021"?
3. I tried reset but got an error, the log is as follows:
```
I (7507) chip[DL]: IPv6 Internet connectivity ESTABLISHED
I (7507) app_main: Interface IP Address changed
I (7517) chip[DIS]: Updating services using commissioning mode 0
I (7517) chip[DIS]: CHIP minimal mDNS started advertising.
I (7537) chip[DIS]: Advertise operational node 992322CA0AB8BB0A-0000000000000001
I (7537) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 992322CA0AB8BB0A-0000000000000001.
I (7557) chip[DIS]: mDNS service published: _matter._tcp
I (7557) ROUTE_HOOK: Hook already installed on netif, skip...
> matter esp factoryreset
Error: 83886338
```

dhrishi, 2022-11-05:
@Firesuiry If you do not have BLE adapter then you can try on-network commissioning. But the above case won't help as it is in some inconsistent state of being commissioned ie. BLE based commissioning incomplete.

Erase flash and re-flash (as reset to factory is not working on your side) and follow [this](https://github.com/espressif/esp-matter/issues/76#issuecomment-1299587379)

If you want to try out BLE based commissioning then you will require a BLE adapter

Firesuiry, 2022-11-06:
> @Firesuiry If you do not have BLE adapter then you can try on-network commissioning. But the above case won't help as it is in some inconsistent state of being commissioned ie. BLE based [...]

I tried 
```
chip-tool pairing onnetwork 0x11 20202021
```
the log is as follows, dose it success?
```
yinsy@yinsy-B450M-GAMING:~/esp$ chip-tool interactive start
[1667735632.871747][6075:6075] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1667735632.873267][6075:6075] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1667735632.873305][6075:6075] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1667735632.873325][6075:6075] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1667735632.873419][6075:6075] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Lck0HE)
[1667735632.873556][6075:6075] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1667735632.873567][6075:6075] CHIP:DL: NVS set: chip-counters/reboot-count = 3 (0x3)
[1667735632.873735][6075:6075] CHIP:DL: Got Ethernet interface: enp9s0
[1667735632.873820][6075:6075] CHIP:DL: Found the primary Ethernet interface:enp9s0
[1667735632.873935][6075:6075] CHIP:DL: Failed to get WiFi interface
[1667735632.873940][6075:6075] CHIP:DL: Failed to reset WiFi statistic counts
[1667735632.873962][6075:6075] CHIP:IN: UDP::Init bind&listen port=0
[1667735632.873979][6075:6075] CHIP:IN: UDP::Init bound to port=42336
[1667735632.873982][6075:6075] CHIP:IN: UDP::Init bind&listen port=0
[1667735632.873995][6075:6075] CHIP:IN: UDP::Init bound to port=55028
[1667735632.873998][6075:6075] CHIP:IN: BLEBase::Init - setting/overriding transport
[1667735632.874001][6075:6075] CHIP:IN: TransportMgr initialized
[1667735632.874011][6075:6075] CHIP:FP: Initializing FabricTable from persistent storage
[1667735632.874047][6075:6075] CHIP:TS: Last Known Good Time: 2022-11-04T16:11:17
[1667735632.874440][6075:6075] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x33FCBF0B0691884B, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1667735632.874760][6075:6075] CHIP:ZCL: Using ZAP configuration...
[1667735632.874958][6075:6075] CHIP:CTL: System State Initialized...
[1667735632.874973][6075:6075] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1667735632.874988][6075:6075] CHIP:CTL: Setting attestation nonce to random value
[1667735632.874996][6075:6075] CHIP:CTL: Setting CSR nonce to random value
[1667735632.875009][6075:6075] CHIP:IN: UDP::Init bind&listen port=5550
[1667735632.875022][6075:6075] CHIP:IN: UDP::Init bound to port=5550
[1667735632.875025][6075:6075] CHIP:IN: UDP::Init bind&listen port=5550
[1667735632.875041][6075:6075] CHIP:IN: UDP::Init bound to port=5550
[1667735632.875044][6075:6075] CHIP:IN: TransportMgr initialized
[1667735632.875134][6075:6080] CHIP:DL: CHIP task running
[1667735633.356055][6075:6079] CHIP:DL: No wifi interface name. Ignoring IP update event.
>>> pairing onnetwork 0x11 20202021             
[1667735655.156283][6075:6080] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1667735655.156316][6075:6080] CHIP:CTL: Setting attestation nonce to random value
[1667735655.156389][6075:6080] CHIP:CTL: Setting CSR nonce to random value
[1667735655.156936][6075:6080] CHIP:CTL: Generating NOC
[1667735655.157486][6075:6080] CHIP:FP: Validating NOC chain
[1667735655.158621][6075:6080] CHIP:FP: NOC chain validation successful
[1667735655.158722][6075:6080] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1667735655.158735][6075:6080] CHIP:TS: Last Known Good Time: 2022-11-04T16:11:17
[1667735655.158743][6075:6080] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1667735655.158749][6075:6080] CHIP:TS: Retaining current Last Known Good Time
[1667735655.159550][6075:6080] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1667735655.160204][6075:6080] CHIP:TS: Committing Last Known Good Time to storage: 2022-11-04T16:11:17
[1667735655.160806][6075:6080] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1667735655.160829][6075:6080] CHIP:IN: UDP::Init bind&listen port=5550
[1667735655.160867][6075:6080] CHIP:IN: UDP::Init bound to port=5550
[1667735655.160874][6075:6080] CHIP:IN: UDP::Init bind&listen port=5550
[1667735655.160896][6075:6080] CHIP:IN: UDP::Init bound to port=5550
[1667735655.160903][6075:6080] CHIP:IN: TransportMgr initialized
[1667735686.194536][6075:6080] CHIP:DIS: Timeout waiting for mDNS resolution.
[1667735711.380320][6075:6079] CHIP:DL: No wifi interface name. Ignoring IP update event.
[1667735775.156561][6075:6075] CHIP:TOO: Run command failure: ../commands/common/CHIPCommand.cpp:513: CHIP Error 0x00000032: Timeout
[1667735796.424806][6075:6079] CHIP:DL: No wifi interface name. Ignoring IP update event.

```

Firesuiry, 2022-11-06:
> @Firesuiry If you do not have BLE adapter then you can try on-network commissioning. But the above case won't help as it is in some inconsistent state of being commissioned ie. BLE based [...]

what`s more, my device is esp32c3-devkitm,
how can I close the led on the board?
I tried 
```
>>> onoff off 0x11 0x0  
[1667736008.418236][6075:6080] CHIP:TOO: Sending command to node 0x11
[1667736008.418447][6075:6080] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000011]
[1667736008.418481][6075:6080] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1667736008.418499][6075:6080] CHIP:CTL: OperationalSessionSetup[1:0000000000000011]: State change 1 --> 2
[1667736008.619154][6075:6080] CHIP:DIS: Checking node lookup status after 200 ms
[1667736023.426908][6075:6080] CHIP:DIS: Checking node lookup status after 15008 ms
[1667736023.426948][6075:6080] CHIP:DIS: OperationalSessionSetup[1:0000000000000011]: operational discovery failed: ../third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout
[1667736023.426960][6075:6080] CHIP:-: ../third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout at ../commands/clusters/ModelCommand.cpp:62
[1667736023.427028][6075:6075] CHIP:TOO: Run command failure: ../third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout
[1667736039.442516][6075:6080] CHIP:DIS: Timeout waiting for mDNS resolution.
[1667736039.493752][6075:6079] CHIP:DL: No wifi interface name. Ignoring IP update event.
[1667736116.506368][6075:6079] CHIP:DL: No wifi interface name. Ignoring IP update event.
[1667736186.518540][6075:6079] CHIP:DL: No wifi interface name. Ignoring IP update event.
```
 but it has no effect..

Firesuiry, 2022-11-06:
[image]

Firesuiry, 2022-11-09:
I success! thank you~

closed 2022-11-09 by Firesuiry

sidhoo22, 2022-11-17:
> what`s more, my device is esp32c3-devkitm, how can I close the led on the board? I tried
>
> ```
> [1667736008.418236][6075:6080] CHIP:TOO: Sending command to node 0x11
> [1667736008.418447][6075:6080] [...]

@Firesuiry hi, i hv met the same issue, ./chip-tool onoff toggle 1234 0x1 failed, can u share how to solve this issue? thanks

Firesuiry, 2022-11-17:
I reflash the device and retry it.  it successes.  
I don not know why but recomand you use 0x1234 instead of 1234.

   

  
 &gt;&gt;&gt; onoff off 0x11 0x0   [1667736008.418236][6075:6080] CHIP:TOO: Sending command to node 0x11 [1667736008.418447][6075:6080] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000011] [1667736008.418481][6075:6080] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found [1667736008.418499][6075:6080] CHIP:CTL: OperationalSessionSetup[1:0000000000000011]: State change 1 --&gt; 2 [1667736008.619154][6075:6080] CHIP:DIS: Checking node lookup status after 200 ms [1667736023.426908][6075:6080] CHIP:DIS: Checking node lookup status after 15008 ms [1667736023.426948][6075:6080] CHIP:DIS: OperationalSessionSetup[1:0000000000000011]: operational discovery failed: ../third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout [1667736023.426960][6075:6080] CHIP:-: ../third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout at ../commands/clusters/ModelCommand.cpp:62 [1667736023.427028][6075:6075] CHIP:TOO: Run command failure: ../third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout [1667736039.442516][6075:6080] CHIP:DIS: Timeout waiting for mDNS resolution. [1667736039.493752][6075:6079] CHIP:DL: No wifi interface name. Ignoring IP update event. [1667736116.506368][6075:6079] CHIP:DL: No wifi interface name. Ignoring IP update event. [1667736186.518540][6075:6079] CHIP:DL: No wifi interface name. Ignoring IP update event.

sidhoo22, 2022-11-17:
@Firesuiry 0x1234 is not work too .

sidhoo22, 2022-11-18:
@Firesuiry erase flash and reflash , repairing. it suceses.  thanks