espressif/esp-matter #85: matter console over /dev/ttyACM0

jonsmirl, 2022-11-11:
Matter console works fine over /dev/ttyUSB0, how do I make it work on /dev/ttyACM0?

jonsmirl, 2022-11-11:
I switched the primary console over to JTAG-USB instead of it being the secondary console. I can see the matter shell prompt, but then it gets stuck in a loop.  So this appears to be broken on /dev/ttyACM0 which is a problem because that is only console on our devices.

jonsmirl, 2022-11-11:
The esp-idf example in examples/system/console/basic works fine on our ttyACM0 devices, so this appears to be matter specific.

jonsmirl, 2022-12-11:
Bump, this is annoying on C3 with only ttyACM0. I have to keep switching back to an ESP32.

shubhamdp, 2022-12-16:
Hi @jonsmirl This is fixed in https://github.com/project-chip/connectedhomeip/commit/8055d866c025ce5a64ba156c983038730824f135. It would take some time to update the connectedhomeip submodule, If you really need this right away please cherry-pick this change.

jonsmirl, 2022-12-17:
In https://github.com/espressif/esp-matter/blob/main/components/esp_matter_console/esp_matter_console.cpp
the include path is:
~~~
/home/jonsmirl/aosp/esp-matter/components/esp_matter_console/esp_matter_console.cpp:22:10: fatal error: src/platform/ESP32/ESP32Utils.h: No such file or directory
 #include <src/platform/ESP32/ESP32Utils.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~~

But in the BLE Mesh example the path is "platform/ESP32_custom/ESP32Utils.h"

How to sort this out? For a temp fix I just edited the path.
#include <platform/ESP32_custom/ESP32Utils.h>

dhrishi, 2023-01-06:
>But in the BLE Mesh example the path is "platform/ESP32_custom/ESP32Utils.h"

Fixed with https://github.com/espressif/esp-matter/commit/62ef2d9c6aec0f9352aad003b7f9ff29e2005033

jonsmirl, 2023-01-06:
That fixes the ble example, but doesn't it break the other examples?

shubhamdp, 2023-01-18:
I guess the reason it broke was `connectedhomeip/connectedhomeip/src` is an include path and not ``connectedhomeip/connectedhomeip`.

closed 2023-01-18 by dhrishi