espressif/esp-matter #88: Windows 10 install: git-lfs: command not found (can't install esp-matter SDK using provided tools/commands)

fkkjohn, 2022-11-14:
Using the OFFLINE installation files for ESP-IDF (https://dl.espressif.com/dl/esp-idf/?idf=4.4), and installing the bundled 4.4.2 with IDE as well as the updated 4.4.3 standalone, I have a functional ESP-IDF/IDE installation. Now I'm attempting to install the ESP-MATTER SDK, and after success enabling long path names, I have encountered what looks to be my last problem: The git LFS plugin won't install, and it was never included in the embedded git that came with ESP-IDF 4.4.3. I downloaded git-lfs-windows-v3.2.0.exe (current as of today) and attempted to install it, only for it to tell me git is not installed on my computer. Of course it is, but it's embedded in ESP-IDF. On my computer it is at: C:\Espressif\tools\idf-git\2.34.2\cmd\

There does not seem to be any way to install lfs from the Windows command line and the installer doesn't find git as installed by esp-idf.

I considered reporting this in the esp-idf repository, but it doesn't require LFS, so here I am.

Here's what my console looks like:
``
// starts off fine
C:\Espressif\frameworks\esp-idf-v4.4.3>git clone --recursive https://github.com/espressif/esp-matter.git
Cloning into 'esp-matter'...
remote: Enumerating objects: 3926, done.
remote: Counting objects: 100% (577/577), done.
remote: Compressing objects: 100% (163/163), done.
remote: Total 3926 (delta 478), reused 487 (delta 412), pack-reused 3349
Receiving objects: 100% (3926/3926), 3.13 MiB | 9.09 MiB/s, done.
Resolving deltas: 100% (2740/2740), done.
Submodule 'connectedhomeip/connectedhomeip' (https://github.com/espressif/connectedhomeip.git) registered for path 'connectedhomeip/connectedhomeip'
// [...]

// then this small error
Cloning into 'C:/Espressif/frameworks/esp-idf-v4.4.3/esp-matter/connectedhomeip/connectedhomeip/third_party/mbed-mcu-boot/repo/boot/espressif/hal/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc'...
fatal: '$GIT_DIR' too big
fatal: clone of 'https://github.com/kmackay/micro-ecc.git' into submodule path 'C:/Espressif/frameworks/esp-idf-v4.4.3/esp-matter/connectedhomeip/connectedhomeip/third_party/mbed-mcu-boot/repo/boot/espressif/hal/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc' failed
Failed to clone 'components/bootloader/subproject/components/micro-ecc/micro-ecc'. Retry scheduled
// [...]

// then LFS identified as a problem
Cloning into 'C:/Espressif/frameworks/esp-idf-v4.4.3/esp-matter/connectedhomeip/connectedhomeip/third_party/ot-br-posix/repo/third_party/openthread/repo'...
remote: Enumerating objects: 95005, done.
remote: Counting objects: 100% (246/246), done.
remote: Compressing objects: 100% (149/149), done.
remote: Total 95005 (delta 119), reused 182 (delta 97), pack-reused 94759
Receiving objects: 100% (95005/95005), 95.14 MiB | 9.98 MiB/s, done.
Resolving deltas: 100% (74658/74658), done.
Submodule path 'connectedhomeip/connectedhomeip/third_party/ot-br-posix/repo/third_party/cJSON/repo': checked out 'cf97c6f066d81fdbba4ef722cfd327bbbba2365c'
Submodule path 'connectedhomeip/connectedhomeip/third_party/ot-br-posix/repo/third_party/http-parser/repo': checked out '4f15b7d510dc7c6361a26a7c6d2f7c3a17f8d878'
Submodule path 'connectedhomeip/connectedhomeip/third_party/ot-br-posix/repo/third_party/openthread/repo': checked out '2ce3d3bf0218566484be2e9943b95c755cefebe3'
Submodule path 'connectedhomeip/connectedhomeip/third_party/pigweed/repo': checked out '9235aeb653e684a6f0b7b563965d85c747281a0f'
Submodule path 'connectedhomeip/connectedhomeip/third_party/pybind11/repo': checked out '8de7772cc72daca8e947b79b83fea46214931604'
git-lfs filter-process: git-lfs: command not found
fatal: the remote end hung up unexpectedly
git-lfs filter-process: git-lfs: command not found
fatal: the remote end hung up unexpectedly
Submodule path 'connectedhomeip/connectedhomeip/third_party/silabs/matter_support': checked out '2425fb72cacd540df505f11e7de81e6270ed1c40'
// [...]
Submodule path 'connectedhomeip/connectedhomeip/third_party/zap/repo': checked out 'd4ce2c13916f1b1e066c6112dbee3e0dc5f5064f'
fatal: Failed to recurse into submodule path 'connectedhomeip/connectedhomeip/third_party/mbed-mcu-boot/repo'
fatal: Failed to recurse into submodule path 'connectedhomeip/connectedhomeip/third_party/openthread/ot-efr32'
fatal: Failed to recurse into submodule path 'connectedhomeip/connectedhomeip/third_party/openthread/ot-ifx'
fatal: Unable to checkout '8aadeb120922720aba2bf0be8d5230d979856a4e' in submodule path 'connectedhomeip/connectedhomeip/third_party/qpg_sdk/repo'
fatal: Unable to checkout '799200bb6788939aa0fef97d7fc4ee9aca57148d' in submodule path 'connectedhomeip/connectedhomeip/third_party/silabs/gecko_sdk'
fatal: Failed to recurse into submodule path 'connectedhomeip/connectedhomeip'
``

fkkjohn, 2022-11-14:
I found a hacky workaround for the git-lfs errors, but the "fatal: '$GIT_DIR' too big" errors persist.

For the git-lfs problem, find your git.exe in your embedded install folder, and paste the contents of the binary download (https://github.com/git-lfs/git-lfs/releases/download/v3.2.0/git-lfs-windows-amd64-v3.2.0.zip) in there and then run "git lfs install" on the command line. I found multiple git.exe files, and instead of trying to figure out which one is the right one, I just pasted those LFS files into each folder. An unintended side effect of this hacky solution is that the esp-matter folder showed up in the ../.git/ subfolder. Since it still doesn't work, I guess that's not an issue at the moment.

shubhamdp, 2022-11-14:
At the moment supported development hosts are:
- macOS 10.15
- Debian 11 (64 bit required)
- Ubuntu 22.04 LTS

Please find more details [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#building-matter)

Although esp-matter can be setup on Windows using WSL, this is not documented anywhere (will add this to docs)
- Enable the Windows Subsystem for Linux 2 (WSL2) following instructions here: https://docs.microsoft.com/en-us/windows/wsl/install-win10
- Install Ubuntu from the Windows App Store here: https://www.microsoft.com/store/productId/9MTTCL66CPXJ
- Start the Ubuntu (search into start menu)
- Running uname -a from within WSL(Ubuntu installed from Microsoft store) should report a kernel version of 5.10.60.1 or later. You’ll need to be running a WSL 2 distro. To upgrade the kernel, run "wsl --upgrade" from Windows Power Shell (ref: https://askubuntu.com/a/1384458)
- Install usbip-win : https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/WSL.md. Required to map the COM ports on windows into WSL. (another ref: https://github.com/dorssel/usbipd-win/wiki/WSL-support)
- Setup esp-matter: https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html

fkkjohn, 2022-11-17:
Thank you very much for your reply. I'm making another attempt at the VM since I can now get past my work proxy, but if I still have trouble under native linux, I'll take your advice and try the linux subsystem.

dhrishi, 2022-12-04:
@fkkjohn Any updates you would want to share? Please close this issue if it is working as expected

fkkjohn, 2022-12-07:
Thank you. No related problems at the moment. Closing.

closed 2022-12-07 by fkkjohn