espressif/esp-matter #89: no proper docker file and image on docker hub does not follow instructions

HomeACcessoryKid, 2022-11-14:
The docker file given in tools is not a propor file because it is not complete and does not start with `FROM ubuntu` or similar.  
The file should follow the instructions in https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#development-setup

And then the docker hub file should reflect this all.

Please update to idf 4.4.3 as well.

shubhamdp, 2022-11-14:
@HomeACcessoryKid docker file is in correct format and it does contain [`FROM` statement](https://github.com/espressif/esp-matter/blob/main/tools/docker/chip-builds#L2).
As it is using some [another base image](https://github.com/project-chip/connectedhomeip/blob/master/integrations/docker/images/chip-build/Dockerfile) that's why it does not contain `FROM ubuntu` or similar statement at the start.

Just to clear one more thing, it is not the development setup for `esp-matter`, the purpose of this docker image is to run the `ci` using this one.

Regarding `v4.4.3`, `esp-matter` is using branch: `v1.0-branch` of `connectedhomeip` repository and it is still on `v4.4.2`.
Will update to `v4.4.3` after some testing is done and things looks stable.

HomeACcessoryKid, 2022-11-14:
Yes, the actual docker file starts with a FROM. But the thing it is starting from itself does not have such FROM.
And it is OK it has a different purpose.

I suggest you make a second docker image that provides the environment that you suggest people use in the developing page. And it must be something that I can track back to something official. Else I must build it myself.
I cannot afford a backdoor or so because the source is not traceable.
So please provide a clear docker file and as a convenience, the image that comes from it in docker hub.

HomeACcessoryKid, 2022-11-14:
sorry, Only now did I see you created a reference to the docker file that is the source and it does contain a FROM ubuntu.
I doubt however that I would be able to link the two from the information I have available.
But fundamentally, I still hope to get a image that focuses on the 'developing' page.

shubhamdp, 2022-12-01:
@HomeACcessoryKid Thanks for the suggestion, will generate and push the `esp-matter` development image to docker hub soon.

HomeACcessoryKid, 2022-12-02:
That is very kind, thank you.

Now that esp-idf 5.0 is official (since two hours ago ;-) could there also be an image based on that?
To give support for the c2 and h2 and to embrace the future today...

TIA,
HacK

dhrishi, 2022-12-16:
@HomeACcessoryKid We haven't officially moved Matter for ESP32, ESP32-C3, ESP32-S3 to ESP-IDF v5.0 yet. You can refer [this](https://github.com/espressif/esp-matter/issues/103#issuecomment-1347827708) if you want to give it a try

dhrishi, 2022-12-16:
>To give support for the c2 and h2 and to embrace the future today...

Yes, C2 support will be added soon and will be based on IDF v5.0. H2 already supports v5.0.

HomeACcessoryKid, 2022-12-16:
Maybe I am missing something, but https://hub.docker.com/r/espressif/esp-matter/tags only shows the two month old image.  
Did something go wrong after you would "generate and push the esp-matter development image to docker hub soon"?

shubhamdp, 2023-01-12:
@HomeACcessoryKid We have added the esp-matter development docker image. You can pull it as follows
```
docker pull espressif/esp-matter:latest
```

We have not yet consider the IDF v5.0 in this, but will soon do that.
We are closing this issue and feel free to open one for IDF v5.0

PR #138 (2023-08-28): Docker image improvements (CON-282)
I created this workflow because I needed to create an image to perform an automated building for my project, so now I'm apporting the workflow to this repository.
The dockerfile is based on esp-idf docker images (v4.4.2 by default as it is the supported IDF version , but it can be changed with build args).

Also I created a readme file that contains info about the images and build args that can be used with the dockerfile. I wish I had more info about the docker image which tag is `chip` and its purpose in order to write it into the docs too, but I left some [... cut]
closed 2023-01-12 by dhrishi