espressif/esp-matter #90: 'light' example is not fully working with Google Nest Hub

AlexeyYukhin, 2022-11-14:
Hi all. I'm trying to launch esp-matter examples for tests and esp-matter SDK study. Then my further goal is to implement Google home integration for smart Door Lock.

I've tried 'light' project. I was able to bind matter device to my Google Nest Hub gen.2 without any problem, using Google Home mobile App. Basic device information (such as device type, firmware version etc.) is readable through Google home app. But I'm not able to control the light state (on/off commands, set brightness and colour temperature). For some reason, device is unreachable in this case.

My steps:
I left everything as-is and flashed my esp32 devboard. The only thing I did change is these SDK config parameters:
CONFIG_DEFAULT_WIFI_SSID
CONFIG_DEFAULT_WIFI_PASSWORD

I set them to my home wifi credentials, so that device was able to be bound with Nest Hub.

Log file from esp32 is attached, as well as sdkconfig file and screenshots ftom Google Home app.
[esp32_logs.txt](https://github.com/espressif/esp-matter/files/10003657/esp32_logs.txt)
[sdkconfig.txt](https://github.com/espressif/esp-matter/files/10003660/sdkconfig.txt)
[image]
[image]
[image]
[image]
[image]
[image]
[image]

esp-idf - 4.4.2 (v4.4.2 tag)
esp-matter - last commit of main branch (2f37420bd5)
esp32 board - ESP32-WROOVER devboard
Master device - Google Nest Hub Gen2

Will be happy to receive any suggestions or advices arround this issue.

jonsmirl, 2022-11-15:
The google support is based on CHIP commit 8638802e06fce1501c5c73b40a1d4eae98d076cb which is from July 15. The ESP-matter support is using current CHIP. You can probably get this working by moving esp-matter back to CHIP 8638802e06fce1501c5c73b40a1d4eae98d076cb.  If not you just have to wait for Google to get around to updating their release.

dhrishi, 2022-11-15:
That's right. The Google Matter SDK for the public release is based on test event 9 (TE9) codebase.

AlexeyYukhin, 2022-11-15:
Thank you guys. I really appreciate this. Please do not close the issue, I will try to find a temporary workaround for this to test it with current Google SDK. If I mange with this I will post here with results.

dhrishi, 2022-11-18:
@AlexeyYukhin I would suggest you try with [sve branch](https://github.com/espressif/esp-matter/tree/sve). This might work with Google Home. Just make sure that you update the submodules correctly on this branch, clean and recompile followed by erase and re-flash.

AlexeyYukhin, 2022-11-20:
> @AlexeyYukhin I would suggest you try with [sve branch](https://github.com/espressif/esp-matter/tree/sve). This might work with Google Home. Just make sure that you update the submodules correctly on this branch, clean and recompile followed by erase and re-flash.

Thanks for your advice. Unfortunately it doesn't work. I am able to bind the device in Google Home App using Google Nest Hub Gen2, but after this the device is shown as 'offline'. I still able to request basic device information (fw version, vendor id etc.), it works. But not device control, as it offline for some reason.

My steps:
1) I swithced to sve branch as you suggested. Then run git submodule update --init --recursive
2) I erased flash of ESP32 and reflashed it. I use 'light' example.
3) Then everything is the same - in Google Home app I added my device, and it's always offline. and can't be reachable to control.

Do you have any other suggestions of how I can kick it start to test esp matter with Google Home?

Also, does it worth to try examples withour esp-matted SDK? Like this one?
https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/esp32

jonsmirl, 2022-11-20:
Did you try this Codelab?
https://developers.home.google.com/codelabs/matter-device-virtual?hl=en#0

AlexeyYukhin, 2022-11-25:
> Did you try this Codelab? https://developers.home.google.com/codelabs/matter-device-virtual?hl=en#0

no, but it's not anyhow related to esp-matter SDK, as it's about original connectedhomeip repository. And it's onlye about virtual matter device example, while I'm trying to get real hw device (ESP32) working.

jonsmirl, 2022-11-25:
Google has updated their Matter support again.
https://developers.home.google.com/matter/release-notes

The CHIP on your device has to match the CHIP on their side for this to work.

jonsmirl, 2022-11-25:
Their esp32 instructions...
https://developers.home.google.com/matter/vendors/espressif

Get it working first from their documentation, then you can look and see what commit they are using and move esp-matter onto that commit.

jonsmirl, 2022-11-25:
Also, note that while you should be able to get this working, when they update again it is going to stop working. Then you'll have to figure out what release they have moved onto and adjust your devices to match.  Apple, Google and Amazon are all on different commit IDs so this is a major pain to track.  Whenever Matter is actually released everyone should converge onto the same commit ID.

AlexeyYukhin, 2022-11-26:
> Their esp32 instructions... https://developers.home.google.com/matter/vendors/espressif
> 
> Get it working first from their documentation, then you can look and see what commit they are using and move esp-matter onto that commit.

thanks for the reply. So far I've tried to follow this inctructions:
https://developers.home.google.com/matter/vendors/espressif?hl=en

- esp-idf checked out on v4.4.1
- connectedhomeip checked out on v1.0-branch
- submodules updated
- I've created a project in Googe Development console and added Matter integration 
[image]

Still the same result. Bindind is working, base device info works, but control doesn't, device is 'offline'. Seems like Google Home App or Google Nest Hub are not on the same versions of Matter SDK.

As for the release notes. I've enabled "Developer Preview" for my Google Nest Hub. But my Google Nest Hub still is on 1.56.309385 version, and it's mentioned in release notes that after enabling this feature, it should be updated to 1.56.324896 version.

So I'm a bit confused about Matter SDK versions sued currently on Google Home App and Google Next Hun Gen2.

jonsmirl, 2022-12-07:
My Hub would not update either. I finally figured out that if I use the Home app to remotely reboot it, it would update to 1.56.324896

dhrishi, 2022-12-13:
@AlexeyYukhin Did you receive the update? Please close the issue if it is working fine

dhrishi, 2022-12-19:
As per Google, all the devices have been updated around 15th Dec. Closing this issue.

closed 2022-12-19 by dhrishi