espressif/esp-matter #92: light example commissioning failed

tonywu-uec, 2022-11-15:
I am trying the light example. However it seemed that the commissioning process through chip-tool failed.
Hope someone can give me a hand.

Environment:
Acer Swift 3 running Ubuntu 22.10
ESP-IDF: v4.4.3
git clone from esp-matter on Nov. 15, 2022.
Board: ESP32-C3-DevKitM-1.

In the log file, I can see the WiFi connected successfully. 
Later the fail-safe timer expired and commissioning failed.

I also tried to use google home app to connect. 
It can recognize the device as bulb. However I got
error when I click the button of the bulb.

run.log is the output of the development board.
chip-tool.log is the output of the command window running chip-tool.
build.log is the output compiling the code.

[run.log.gz](https://github.com/espressif/esp-matter/files/10013506/run.log.gz)
[chip-tool.log.gz](https://github.com/espressif/esp-matter/files/10013507/chip-tool.log.gz)
[build.log.gz](https://github.com/espressif/esp-matter/files/10013508/build.log.gz)

dhrishi, 2022-11-16:
@tonywu-uec Can you please share the `run.log` in English language.

Your chip-tool log shows that the first part is successful and later on it is unable to find the device on the Wi-Fi (IP) network and hence cannot proceed with the CASE session. Can you confirm if the host running the chip-tool and the ESP32 device are connected to the same Wi-Fi network? This will be required.

tonywu-uec, 2022-11-16:
Thanks! Today the example works! I did check they are on the same network where I can ping the device.

However I still could not be able to use Google Home to add the "Bulb". The device was commissioned correctly but the Google Home will simply show "OFF-LINE".

tonywu-uec, 2022-11-18:
I was curious to know if I can't use Google Home app to test, what else platform shall I use? Apple iPhone? or Alexa?
The chip-tool utility is not acceptable for demonstration purpose.

dhrishi, 2022-11-18:
@tonywu-uec Please see the steps here https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md for using Apple Home

tonywu-uec, 2022-11-23:
I updated the Google Home App to 2.60.60.5 released on Nov. 21 2022. It can recognize the test device and show on the screen. However the light device shows "off-line" and can not toggle the light on/off. Didn't Google try to change their code to accommodate the latest code in CHIP github repository?

dhrishi, 2022-11-24:
> Didn't Google try to change their code to accommodate the latest code in CHIP github repository?

It is work in progress AFAIK

dhrishi, 2022-12-13:
@tonywu-uec Did you get it functional with the Google Nest Hub? In any case, I will close this issue and you can track the same through #90. Please post an update there if you face any problems even after the Nest Hub and GHA update.

closed 2022-12-13 by dhrishi