espressif/esp-matter #94: Occupancy sensing not working

slompf18, 2022-11-20:
Using the default occupancy sensing configuration, the device is shown as 'Not supported' in the HomeKit app.

This code I use:

```
node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    occupancy_sensor::config_t occu_config;
    endpoint_t *endpoint = occupancy_sensor::create(node, &occu_config, ENDPOINT_FLAG_NONE, NULL);
    endpoint_id = endpoint::get_id(endpoint);
```

dhrishi, 2022-11-21:
@slompf18 Thanks for reporting the issue. It is a problem on our side and we will fix it. Your changes in the app are correct. The issue is that descriptor cluster is missing in the sensor device type and hence the iOS is not able to recognize it. I have tried by adding it and see that it works as expected.

[image]

 I also set the values of occupancy from the ESP32 CLI to detected (1) and see the same being reflected in the Apple Home app. Command below -
```
> matter esp attribute set 0x1 0x406 0x0 1 
```
[image]

If you want to try, please add
```
descriptor::create(endpoint, CLUSTER_FLAG_SERVER);
```

at this line https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.cpp#L696

closed 2022-11-22 by espressif-bot

slompf18, 2022-11-30:
Thanks, its working now.