espressif/esp-matter #95: esp32c3 run fail

shun-dgut, 2022-11-23:
environment
macos

run command
./chip-tool interactive start 
pairing ble-wifi 0x7283 <ssid> <passphrase> 20202021 3840

firmware log：

[0;32mI (5771) NimBLE: GAP procedure initiated: advertise; [0m
[0;32mI (5781) NimBLE: disc_mode=2[0m
[0;32mI (5781) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40[0m
[0;32mI (5791) NimBLE: 
[0m
[0;32mI (5801) chip[DL]: CHIPoBLE advertising started[0m

[2022-11-23 21:39:43.943]# RECV ASCII>
[0;32mI (31451) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)[0m
[0;32mI (31451) chip[DL]: Device already advertising, stop active advertisement and restart[0m
[0;32mI (31461) NimBLE: GAP procedure initiated: stop advertising.
[0m
[0;32mI (31461) NimBLE: GAP procedure initiated: advertise; [0m
[0;32mI (31481) NimBLE: disc_mode=2[0m
[0;32mI (31481) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800[0m
[0;32mI (31491

[2022-11-23 21:39:43.991]# RECV ASCII>
) NimBLE: 
[0m

chiptool log
[log.txt](https://github.com/espressif/esp-matter/files/10076005/log.txt)

shun-dgut, 2022-11-24:
The firmware should be fine
There is a problem with chiptool on Mac OS
[image]

dhrishi, 2022-11-24:
@shun-dgut Bug with the MacOS. Please see this: https://github.com/project-chip/connectedhomeip/issues/23435

dhrishi, 2022-11-24:
@shun-dgut Will you please close this and track the above issue in the upstream connectedhomeip repo instead?

closed 2022-11-25 by shun-dgut