espressif/esp-matter #96: matter's install.sh install have some problem.

swz1314520ll, 2022-11-24:
[image]
`swz@swz-ubuntu:~/esp/esp-matter$ . ./install.sh
dirname: 不适用的选项 -- b
请尝试执行 "dirname --help" 来获取更多信息。

Connection closed.

Disconnected from remote host(swz_ubuntu 22) at 08:39:51.

Type `help' to learn how to use Xshell prompt.
`

```
swz@swz-ubuntu:~/esp/esp-matter$ sudo ./install.sh

Running Matter Setup

  WELCOME TO...

   

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...[/]
b'[P2490 08:54:15.862 client.go:306 W] RPC failed transiently. Will retry in 1s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 173.236.212.42:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"1s"}\n[P2490 08:54:37.872 client.go:306 W] RPC failed transiently. Will retry in 2s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 173.236.212.42:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"2s"}\n[P2490 08:55:00.880 client.go:306 W] RPC failed transiently. Will retry in 4s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 173.236.212.42:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"4s"}\n[P2490 08:55:26.014 client.go:306 W] RPC failed transiently. Will retry in 8s    {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 173.236.212.42:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"8s"}\n[P2490 08:55:55.023 client.go:306 W] RPC failed transiently. Will retry in 16s   {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 173.236.212.42:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"16s"}\n[P2490 08:56:32.049 client.go:306 W] RPC failed transiently. Will retry in 32s   {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 31.13.96.208:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"32s"}\n[P2490 08:57:25.097 client.go:306 W] RPC failed transiently. Will retry in 1m4s  {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 31.13.96.208:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"1m4s"}\n[P2490 08:58:50.171 client.go:306 W] RPC failed transiently. Will retry in 2m8s  {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 31.13.96.208:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"2m8s"}\n[P2490 09:01:19.241 client.go:306 W] RPC failed transiently. Will retry in 4m16s {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 31.13.96.208:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"4m16s"}\n[P2490 09:05:56.349 client.go:306 W] RPC failed transiently. Will retry in 8m32s {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 31.13.96.208:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository", "sleepTime":"8m32s"}\n[P2490 09:14:50.030 client.go:350 W] RPC failed permanently: rpc error: code = Internal desc = prpc: when sending request: Post "https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix": dial tcp 179.60.193.9:443: connect: connection refused {"error":"rpc error: code = Internal desc = prpc: when sending request: Post \\"https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix\\": dial tcp 179.60.193.9:443: connect: connection refused", "host":"chrome-infra-packages.appspot.com", "method":"ListPrefix", "service":"cipd.Repository"}\nError: prpc: when sending request: Post "https://chrome-infra-packages.appspot.com/prpc/cipd.Repository/ListPrefix": dial tcp 179.60.193.9:443: connect: connection refused.\n'
Traceback (most recent call last):
  File "/home/swz/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 795, in <module>
    sys.exit(main())
  File "/home/swz/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 787, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/swz/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 457, in setup
    result = step(spin)
  File "/home/swz/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 564, in cipd
    if not cipd_update.update(cipd=cipd_client,
  File "/home/swz/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 311, in update
    if not check_auth(cipd, package_files, spin):
  File "/home/swz/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 85, in check_auth
    inaccessible_paths = _check_all_paths()
  File "/home/swz/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 68, in _check_all_paths
    output = subprocess.check_output(
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/swz/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/cipd', 'ls', 'pigweed/third_party/gcc-arm-none-eabi']' returned non-zero exit status 1.

```

alibukharai, 2022-11-24:
Hi, swz1314520ll. This is the instalation problem. Follow this [method](https://github.com/alibukharai/Blogs/blob/main/ESP-Matter(Demo)/Getting_Started/getting_started_with_matter_esp.md)

shubhamdp, 2022-12-01:
@swz1314520ll are you still facing this issue?

Can you try removing `rm -rf connectedhomeip/connectedhomeip/.environment` from your top directory and re-running `./install.sh`

swz1314520ll, 2022-12-01:
yes,I try as you said,but there is the same problem.

alibukharai, 2022-12-01:
@swz1314520ll I also got the same error while installing I guess you are using Ubuntu 22. [Follow this](https://github.com/alibukharai/Blogs/blob/main/ESP-Matter(Demo)/Getting_Started/getting_started_with_matter_esp.md)

swz1314520ll, 2022-12-01:
ok,thanks!If I have any problem,i will tell you.

dhrishi, 2022-12-06:
@swz1314520ll Is your issue solved?

swz1314520ll, 2022-12-06:
The link of baidu pan is here.链接: https://pan.baidu.com/s/1nfVhxe-Uy8LXzhwB16UZaA?pwd=6666 提取码: 6666 复制这段内容后打开百度网盘手机App，操作更方便哦

chshu, 2022-12-06:
@swz1314520ll From your previous log, it's caused by some network connectivity issue. You may need a VPN to setup the Matter build environment.

dhrishi, 2022-12-19:
@swz1314520ll Please close this if your issue is resolved

dhrishi, 2022-12-22:
Closing this now. Please re-open if you face any issue using the VPN as suggested above

closed 2022-12-22 by dhrishi