espressif/esp-matter #97: Non-matter Device How can I Notify the bridging device of a status change?

shun-dgut, 2022-11-24:
examples/blemesh_bridge 
In this example, how will the Matter bridging device be notified if the child device changes its state by a button

shun-dgut, 2022-11-25:
such as the example of a light, the key is pressed to inform the data. But in the blemesh_bridge example, if the state of the child device changes, then how do I notify the APP of the data.

[image]

InfiniteYuan, 2022-11-28:
Hi, in the blemesh_bridge example, there is no implement to get the state changes of the bridged device(Non-Matter).

In a Matter bridge device, a similar API(As you mentioned) is still used to update the state of the bridged device.

shun-dgut, 2023-03-02:
> Hi, in the blemesh_bridge example, there is no implement to get the state changes of the bridged device(Non-Matter).
> 
> In a Matter bridge device, a similar API(As you mentioned) is still used to update the state of the bridged device.

like temperature_measurement ，I need to update the value of the matter system, what api do I need to call

I tried to update like this, but the data read out using chiptool is still null

```
attribute::update(ep_temp,TemperatureMeasurement::Id,TemperatureMeasurement::Attributes::MeasuredValue::Id,&val);
```
[image]

shun-dgut, 2023-03-02:
@InfiniteYuan

shun-dgut, 2023-03-02:
@mahavirj

shun-dgut, 2023-03-02:
Sorry, I got the endpoint number wrong

closed 2023-03-02 by shun-dgut