espressif/esp-matter #98: light_switch example not working with HomeKit

scottgarner, 2022-11-27:
Not sure if this is related to #94, but I get an error message in the Home app when using the [light_switch](https://github.com/espressif/esp-matter/tree/main/examples/light_switch) example:

`This accessory is not currently supported by the Home app.`

dhrishi, 2022-11-27:
@scottgarner Seems like an actual error message as the light_switch example has OnOff client cluster. It is not supported yet. See the supported device types [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#ensuring-your-matter-accessory-works-with-iosipadostvos)

scottgarner, 2022-11-27:
Thanks for the response! I do see `Switch` in that list, which seems to refer to an on/off switch. So the issue is the added group cluster?

Can you suggest how I might get this example working? Just removing the cluster-related lines doesn't appear to change anything, so I assume some additional setup is required.

dhrishi, 2022-11-28:
@scottgarner Oops. I totally missed seeing the entry. We will verify it and get back. 

>Can you suggest how I might get this example working?

Do you want to build a light_switch that can subscribe to and/or control (on, off, toggle) an OnOff server, like a light?

scottgarner, 2022-11-28:
Basically I want some buttons connected to an ESP 32 that I can attach behaviors to in HomeKit (toggle lights, change scenes, etc.).

I think that means I want what Matter calls `OnOffSwitch`, which should show up as what Apple calls `Switch`?

jadhavrohit924, 2022-11-29:
@scottgarner  Yes, `This accessory is not currently supported by the Home app`. Looks like some issue at Apple's end. We will check with them and update

jonsmirl, 2022-11-30:
The Light Switch is only for binding to a light bulb, it is not what you are looking for. What you want is a Generic Switch. esp-matter does not have a sample for generic switch but it is not hard to convert the light switch sample over to become a generic switch.   Light Switch is likely never going to be supported in IOS -- because iOS is not a Light Bulb -- so the error is to be expected.

I have not checked in about a week, but as of a week ago Generic Switch was still not supported in IOS. I talk with the Apple developers and they have assured me it is being worked on.

mriksman, 2022-12-04:
@scottgarner I've only _just_ started reading up on Matter. From what I can tell, the `light_switch` example is an Endpoint _Client_, much like an Apple iPhone is an Endpoint Client. It is used to control other devices (Endpoint Servers). It is meant to be directly _bound_ to the `light` example code using ACLs and the bind command; this will allow a physical switch to control a smart lightbulb without any of the automation trickery that used to be required in HomeKit. You can use the Espressif Matter app on the App Store to do the binding, or the CHIP-Tool (I think?)

@jonsmirl So the [list](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#ensuring-your-matter-accessory-works-with-iosipadostvos) has Switch listed. Is that the same as the HomeKit Service `Switch`; where you could set the icon to a Fan, Lightbulb or Outlet, and have basic On/Off commands? That's not supported? So the list is not accurate?

jonsmirl, 2022-12-04:
I believe the list you linked is inaccurate, the Switch type should be Generic Switch type 0xF, not On/Off Switch type 0x103.  Matter Generic Switch is what maps to the Homekit Switch.

mriksman, 2022-12-04:
> I believe the list you linked is inaccurate, the Switch type should be Generic Switch type 0xF, not On/Off Switch type 0x103. Matter Generic Switch is what maps to the Homekit Switch.

Ahh, OK. And you say this isn't yet implemented in HomeKit? So you can have Lightbulb, Outlet, but you can't yet have the Generic Switch (which allows you to select between Light, Fan, Outlet icons in HomeKit)?

dhrishi, 2022-12-04:
> Light Switch is likely never going to be supported in IOS -- because iOS is not a Light Bulb -- so the error is to be expected.

0x0103 is On/Off Light Switch and not Generic Switch which is 0x000F. The documentation [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#ensuring-your-matter-accessory-works-with-iosipadostvos) mentions that it supports the former one.

It can be supported, wherein the option is provided only for binding for device-to-device automations (being a client cluster it does not have attributes or events). Although, currently, binding is not supported too. So, we are checking with them about what is actually supported.

mriksman, 2022-12-04:
> Look at the type of events the two switches make. On/Off sends things like Light On, Light Off. Generic Switch sends out events like Press Down, Press Up, Single Click, Double Click.
I just saw your edit.

On/Off is all you need for `Switch` in HomeKit. In fact, for a `Lightbulb` that doesn't have color/temperature/dimming, `Lightbulb` and `Switch` are identical (just the On/Off Characteristic). It's just that in the Home App, a device listed as a `Switch` Service can have its icon/type changed to show a Fan, Outlet, _or_ Lightbulb.
 
Generic Switch like what you have described sounds more like HomeKit's `StatefulProgrammableSwitch` Service, which has the `ProgrammableSwitchEvent` Characteristic, which has the states `SINGLE PRESS`, `DOUBLE PRESS` and `LONG PRESS`.
Does that sound accurate?

mriksman, 2022-12-04:
> It can be supported, wherein the option is provided only for binding for device-to-device automations (being a client cluster it does not have attributes or events). Although, currently, binding is not supported too. So, we are checking with them about what is actually supported.

Is binding performed from the Home App? Or do you need to use CHIP Tool or the Espressif-Matter iOS app to bind a Light Switch (client) to a Light Bulb (server)?

jonsmirl, 2022-12-04:
AFAIK none of Apple, Amazon or Google have implemented support to create bindings. Unless this changes, you would need to do this in your own app.

dhrishi, 2022-12-04:
@mriksman Binding is not supported from the Home app. In case of multi-admin, you can use the other options you mentioned (Note: These are test apps). Although, if the device is commissioned only through the Home App then you won't be able to do that.

mriksman, 2022-12-04:
Sorry, we've gone off topic. Last query. 
I should be able to commission in Home App, and then start the commissioning again and add it to... Espressif-Matter iOS app? Or CHIP Tool in Ubuntu? And then use those tools to bind the Switch (client) to the Lightbulb (server)?

jonsmirl, 2022-12-04:
The commissioning process is different on IOS and Android. On IOS you go into Home and share the device. It will give you a key, you then copy that key into your app to do the commissioning. On Android you go into the Home app and and share the device. It will present you with a list of apps supporting sharing. Pick one and the key will be passed on as an Intent.

jonsmirl, 2022-12-07:
I just checked Apple and Google, still no support for Generic Switch. Google will add the Generic Switch, but the events are not visible in the Routines trigger dialog. Apple says Generic Switch is still unsupported. I don't have access to the Amazon beta.

mriksman, 2022-12-07:
Hmm. Looking at the [Matter specifications](https://images.tuyaus.com/rms-static/20749630-5a92-11ed-9d99-a5aaf2a10702-1667381917459.pdf?tyName=Cluster_Matter-1.0-Device-Library-Specification.pdf), section 6 says that these endpoints are for "Controllers" (clients). So, `0x0103` is an On/Off _client_. And is what would be used to bind to a `0x0100` On/Off _server_ lightbulb. 

I wonder what `0x010a` 'On/Off Plug-in Unit' looks like in Home App. It says it is meant to be used for wall switches that are connected to a non-smart load, like a lightbulb or an appliance. Actually, seems almost identical to `0x0100` 'On/Off Light' except it is missing the 'Occupancy sensor' cluster. I would imagine the 'On/Off Plug-in Unit' will have the option to select 'Fan/Outlet/Lightbulb', and is what I consider the 'generic switch' in HomeKit.

I can't quite work out what Matter's `0x000F` 'Generic Switch' is for. I _think_ it is used for creating a physical switch device that you can assign scenes too. Instead of a light switch client binding to a lightbulb (which _must_ use the on/off cluster), you would have 'things' (lightbulbs, appliances, devices) as clients instead _subscribing_ to the 'Generic Switch' `Switch` cluster (like a client occupancy sensor). When an event happens, those clients that are subscribed to the 'Generic Switch' would act on it (turn TV on, dim lights, turn music on). I guess you could also have Apple/Google subscribing to it, and create automations when a button is pressed. I can't work out if Matter supports different events like, short press, long press, double press, etc.

AchimPieters, 2022-12-07:
I got the same error yesterday. `This accessory is not currently supported by the Home app.`

jonsmirl, 2022-12-07:
Yes, Generic Switch is the old Zigbee Scene button renamed. It has the ID -- 0xF.  Generic switch just send events like ButtonDown, ButtonUp, Clicked, Double Clicked, Long Press, etc.  Only a controller can understand these event. These event are used to trigger scripts in the controller.

Apple/Google/Amazon need to implement Generic Switch and attached routines because the Matter group has punted native scene support in to the future.

dhrishi, 2022-12-16:
> I would imagine the 'On/Off Plug-in Unit' will have the option to select 'Fan/Outlet/Lightbulb', and is what I consider the 'generic switch' in HomeKit.

@mriksman Yes, that's right. The Home app already shows this option for an outlet, during its commissioning.

dhrishi, 2023-02-02:
@scottgarner Can we close this issue?

closed 2023-02-03 by scottgarner