espressif/esp-matter #101: Matter blemesh_bridge，There is a ble broadcast that cannot be updated

shun-dgut, 2022-12-01:
## Platform
1.  ESP32C3

## Log
```
[2022-12-01 16:28:04.202]# RECV ASCII>
) Sending msg 214734041 to IP address 'UDP:[FE80::43A:819:BEB2:74D3%st1]:58400'[0m
[0;32mI (93093) chip[IN]: CASE Server enabling CASE session setups[0m

[2022-12-01 16:28:12.131]# RECV ASCII>
[0;31mE (101033) BLE_MESH: Advertising set failed: err 30[0m
[0;31mE (101033) BLE_MESH: Start advertising failed: err 30[0m
[0;33mW (101033) BLE_MESH: Failed to send adv packet[0m

[2022-12-01 16:28:22.155]# RECV ASCII>
[0;31mE (111043) BLE_MESH: Advertising set failed: err 30[0m
[0;31mE (111043) BLE_MESH: Start advertising failed: err 30[0m
[0;33mW (111043) BLE_MESH: Failed to send adv packet[0m

[2022-12-01 16:28:32.160]# RECV ASCII>
[0;31mE (121053) BLE_MESH: Advertising set failed: err 30[0m
[0;31mE (121053) BLE_MESH: Start advertising failed: err 30[0m
[0;33mW (121053) BLE_MESH: Failed to send adv packet[0m
```

dhrishi, 2022-12-04:
@shun-dgut Can you please share the steps to recreate the issue? Or does it happen with the default blemesh_bridge example without any change?

shun-dgut, 2022-12-05:
> @shun-dgut Can you please share the steps to recreate the issue? Or does it happen with the default blemesh_bridge example without any change?

I am directly using two esp32c3 to test, according to blemesh_bridge readme for environment deployment. First I powered up the ble mesh device, then I powered up the bridge device, then I started to configure the bridge device, and after the device was successfully configured, this error would appear. But when I restart the bridge device directly, the error disappears again, but it appears again after a while.

shun-dgut, 2022-12-05:
> @dhrishi Can you please share the steps to recreate the issue? Or does it happen with the default blemesh_bridge example without any change?
I am directly using two esp32c3 to test, according to blemesh_bridge readme for environment deployment. First I powered up the ble mesh device, then I powered up the bridge device, then I started to configure the bridge device, and after the device was successfully configured, this error would appear. But when I restart the bridge device directly, the error disappears again, but it appears again after a while.

dhrishi, 2022-12-05:
Thanks for the details @shun-dgut. We will check this

closed 2022-12-19 by shun-dgut