espressif/esp-matter #102: Bridge Lock enpoid cannot write data

shun-dgut, 2022-12-03:
## Platfrom
hardware：ESP32C3
SDK version:  commit id  449160b1eda
## Code 
app_bridge_create_bridged_device(node, aggregator_endpoint_id, ESP_MATTER_DOOR_LOCK_DEVICE_TYPE_ID,
                                                 ESP_MATTER_BRIDGED_DEVICE_TYPE_BLEMESH,
                                                 app_bridge_blemesh_address(0x1235));

## chiptool Log

doorlock lock-door 0x7283 3
[1670062579.716068][806181:806183] CHIP:TOO: Sending command to node 0x7283
[1670062579.716264][806181:806183] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000007283]
[1670062579.716279][806181:806183] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1670062579.716299][806181:806183] CHIP:CTL: Found an existing secure session to [1:0000000000007283]!
[1670062579.716309][806181:806183] CHIP:CTL: OperationalSessionSetup[1:0000000000007283]: State change 1 --> 5
[1670062579.716322][806181:806183] CHIP:TOO: Sending cluster (0x00000101) command (0x00000000) on endpoint 3
[1670062579.716349][806181:806183] CHIP:DMG: ICR moving to [AddingComm]
[1670062579.716362][806181:806183] CHIP:DMG: ICR moving to [AddedComma]
[1670062579.716452][806181:806183] CHIP:EM: <<< [E:29530i M:235707570] (S) Msg TX to 1:0000000000007283 [41BE] --- Type 0001:08 (IM:InvokeCommandRequest)
[1670062579.716471][806181:806183] CHIP:IN: (S) Sending msg 235707570 on secure session with LSID: 2357
[1670062579.716560][806181:806183] CHIP:DMG: ICR moving to [CommandSen]
[1670062579.920047][806181:806183] CHIP:EM: >>> [E:29530i M:145476795 (Ack:235707570)] (S) Msg RX from 1:0000000000007283 [41BE] --- Type 0001:09 (IM:InvokeCommandResponse)
[1670062579.920089][806181:806183] CHIP:EM: Found matching exchange: 29530i, Delegate: 0x7f3a9c00a848
[1670062579.920109][806181:806183] CHIP:EM: Rxd Ack; Removing MessageCounter:235707570 from Retrans Table on exchange 29530i
[1670062579.920128][806181:806183] CHIP:DMG: ICR moving to [ResponseRe]
[1670062579.920156][806181:806183] CHIP:DMG: InvokeResponseMessage =
[1670062579.920162][806181:806183] CHIP:DMG: {
[1670062579.920170][806181:806183] CHIP:DMG:    suppressResponse = false,
[1670062579.920179][806181:806183] CHIP:DMG:    InvokeResponseIBs =
[1670062579.920199][806181:806183] CHIP:DMG:    [
[1670062579.920210][806181:806183] CHIP:DMG:            InvokeResponseIB =
[1670062579.920225][806181:806183] CHIP:DMG:            {
[1670062579.920233][806181:806183] CHIP:DMG:                    CommandStatusIB =
[1670062579.920242][806181:806183] CHIP:DMG:                    {
[1670062579.920250][806181:806183] CHIP:DMG:                            CommandPathIB =
[1670062579.920260][806181:806183] CHIP:DMG:                            {
[1670062579.920270][806181:806183] CHIP:DMG:                                    EndpointId = 0x3,
[1670062579.920280][806181:806183] CHIP:DMG:                                    ClusterId = 0x101,
[1670062579.920289][806181:806183] CHIP:DMG:                                    CommandId = 0x0,
[1670062579.920298][806181:806183] CHIP:DMG:                            },
[1670062579.920310][806181:806183] CHIP:DMG:
[1670062579.920318][806181:806183] CHIP:DMG:                            StatusIB =
[1670062579.920328][806181:806183] CHIP:DMG:                            {
[1670062579.920339][806181:806183] CHIP:DMG:                                    status = 0xc6 (NEEDS_TIMED_INTERACTION),
[1670062579.920348][806181:806183] CHIP:DMG:                            },
[1670062579.920358][806181:806183] CHIP:DMG:
[1670062579.920365][806181:806183] CHIP:DMG:                    },
[1670062579.920377][806181:806183] CHIP:DMG:
[1670062579.920385][806181:806183] CHIP:DMG:            },
[1670062579.920396][806181:806183] CHIP:DMG:
[1670062579.920402][806181:806183] CHIP:DMG:    ],
[1670062579.920416][806181:806183] CHIP:DMG:
[1670062579.920423][806181:806183] CHIP:DMG:    InteractionModelRevision = 1
[1670062579.920430][806181:806183] CHIP:DMG: },
[1670062579.920465][806181:806183] CHIP:DMG: Received Command Response Status for Endpoint=3 Cluster=0x0000_0101 Command=0x0000_0000 Status=0xc6
[1670062579.920480][806181:806183] CHIP:TOO: Error: IM Error 0x000005C6: General error: 0xc6 (NEEDS_TIMED_INTERACTION)
[1670062579.920500][806181:806183] CHIP:DMG: ICR moving to [AwaitingDe]
[1670062579.920555][806181:806183] CHIP:EM: <<< [E:29530i M:235707571 (Ack:145476795)] (S) Msg TX to 1:0000000000007283 [41BE] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1670062579.920571][806181:806183] CHIP:IN: (S) Sending msg 235707571 on secure session with LSID: 2357
[1670062579.920580][806181:806181] CHIP:TOO: Run command failure: IM Error 0x000005C6: General error: 0xc6 (NEEDS_TIMED_INTERACTION)
[1670062579.920630][806181:806183] CHIP:EM: Flushed pending ack for MessageCounter:145476795 on exchange 29530i

jadhavrohit924, 2022-12-30:
@shun-dgut Please try command `./out/debug/chip-tool doorlock lock-door <nodeId> <endpointId> --timedInteractionTimeoutMs 100`

dhrishi, 2023-01-03:
@shun-dgut Please try and close the issue if this works fine

dhrishi, 2023-01-06:
Closing. Feel free to re-open if this does not work as expected

closed 2023-01-06 by dhrishi