espressif/esp-matter #103: Not ESP-IDF V5.0 ready

AchimPieters, 2022-12-03:
Is the ESP-MATTER repro not ready for IDF V5.0? I got this error:

```
connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/DiagnosticDataProviderImpl.cpp -o obj/third_party/connectedhomeip/src/platform/ESP32/ESP32.DiagnosticDataProviderImpl.cpp.o
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/DiagnosticDataProviderImpl.cpp:37:10: fatal error: spi_flash_mmap.h: No such file or directory
   37 | #include "spi_flash_mmap.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
[473/716] c++ obj/third_party/connectedhomeip/src/platform/ESP32/ESP32.ConfigurationManagerImpl.cpp.o
[474/716] c++ obj/third_party/connectedhomeip/src/platform/ESP32/ESP32.ConnectivityManagerImpl_Ethernet.cpp.o
[475/716] c++ obj/third_party/connectedhomeip/src/platform/ESP32/ESP32.ConnectivityManagerImpl.cpp.o
[476/716] c++ obj/third_party/connectedhomeip/src/platform/ESP32/ESP32.DeviceNetworkProvisioningDelegateImpl.cpp.o
[477/716] c++ obj/third_party/connectedhomeip/src/platform/ESP32/ESP32.ESP32Config.cpp.o
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
HINT: Please make sure that the header name is correct. Also please check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'. If the component is not present then it should be added by the IDF Component Manager. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the header file has been removed, renamed or relocated - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the /Users/achimpieters/esp/esp-matter/examples/light_switch/build/log/idf_py_stderr_output_19985 and /Users/achimpieters/esp/esp-matter/examples/light_switch/build/log/idf_py_stdout_output_19985
achimpieters@Achims-iMac light_switch % 
```

dhrishi, 2022-12-04:
Currently, the connectedhomeip supports v4.4.x for ESP32 and ESP32-C3. We will update to ESP-IDF v5.0 after some testing is done.

AchimPieters, 2022-12-05:
@dhrishi Thank you for your reply! Will the ESP32-C2/ESP8684 be supported too?

dhrishi, 2022-12-05:
@AchimPieters Yes, ESP32-C2 will be supported too.

benjaminnestler, 2022-12-12:
Hi @dhrishi 
can you please give us a hint (maybe in this issue) if esp-matter repo is ready to use with esp-idf V5.0? That was great! Thanks a lot.
I'm running in the same issue ...

shubhamdp, 2022-12-13:
@benjaminnestler @AchimPieters Below patches can unblock you for building an example with ESP-IDF v5.0.
NOTE: Tested the builds on ESP32 and ESP32-C3 and commissioning on ESP32 so a lot more tests has to be ran before releasing v5.0 support.

```
git apply 0001-idf-v5.0-support-patch.txt
cd connectedhomeip/connectedhomeip
git apply 0002-add-spi_flash-dependency-for-idf-v5.0.txt
cd ../../examples/light
idf.py build
```

#### Patches:
[0001-idf-v5.0-support-patch.txt](https://github.com/espressif/esp-matter/files/10215221/0001-idf-v5.0-support-patch.txt)
[0002-add-spi_flash-dependency-for-idf-v5.0.txt](https://github.com/espressif/esp-matter/files/10215174/0002-add-spi_flash-dependency-for-idf-v5.0.txt)

dhrishi, 2022-12-13:
@AchimPieters @benjaminnestler I hope the above patches unblock you. Note that, we won't be immediately moving to ESP-IDF v5.0 officially. Please close the issue if your issue is resolved

AshUK, 2023-03-02:
Any update on if esp-idf v5 is coming anytime soon, v4.3/4 are currently are now in 18th month maintenance period making it hard to use esp-matter for new designs.