espressif/esp-matter #104: Developing on Windows

mriksman, 2022-12-04:
I've yet to try (I'm away for business), but I have been reading up extensively. I program on VSCode, in Windows (and tend to fumble my way through development), so setting up for `connectedhomeip`/`esp-matter` is going to be... interesting... given that it is quite new.

I see instructions [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/VSCODE_DEVELOPMENT.md) which will set up WSL 2 and link it to VSCode. I should be able to download the specific [commit required](https://github.com/espressif/connectedhomeip/tree/87bee4de) and use that for the initial build in VSCode. 

I have ESP-IDF working great in VSCode already, and have developed several programs.

I see in the [CMakeLists.txt file](https://github.com/espressif/esp-matter/blob/main/examples/light/CMakeLists.txt#L26-L27), that it would want the `esp-matter` SDK in the same location as the `connectedhomeip` Matter SDK?
```
set(ESP_MATTER_PATH $ENV{ESP_MATTER_PATH})
set(MATTER_SDK_PATH ${ESP_MATTER_PATH}/connectedhomeip/connectedhomeip)
```
So does that mean I'll have `esp-matter` and `connectedhomeip` (and `esp-idf`) in regular folders on Windows, _and_ the `connectedhomeip` also in a docker container with all the binaries connected to VSCode?

Anyone get `esp-matter` working in Windows with VSCode and have any instructions?

shubhamdp, 2022-12-05:
@mriksman What we have tried is setting development environment in WSL (ubuntu) and using the VSCode and mounting that with some VSCode extensions. Let me compile the doc for this.

closed 2022-12-15 by espressif-bot

shubhamdp, 2022-12-15:
@mriksman Please check here: https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#development-setup

Added the docs for setting up esp-matter on Windows Host using WSL

mriksman, 2022-12-29:
Hi @shubhamdp. Thanks for that. Unfortunately, I didn't find it very clear.

In the end, I did the following

1. Install WSL
2. Install Docker Desktop
3. Install VSCode
4. Install ESP-IDF Extension
5. Use ESP-IDF Extension to create a New Project
6. Edit the Dockerfile to include stuff for Matter

However, things seem a little odd. See #142. The Build button on the Activity bar doesn't work, because the Matter `pip requirements` aren't loaded. Every time I open a Terminal manually, I need to run `source ./activate.sh`.

So I am close, but not quite there. Are you able to work out what I am missing? I used the devcontainer.json that is created when you use the ESP-IDF to create 'New Project', and then I modified the Dockerfile to add in the stuff for Matter.

```
ARG ESP_IDF_TAG
FROM espressif/idf:$ESP_IDF_TAG

ARG DEBIAN_FRONTEND=nointeractive

RUN apt-get update \
  && apt install -y -q \
  cmake \
  git \
  hwdata \
  libglib2.0-0 \
  libnuma1 \
  libpixman-1-0 \
  linux-tools-virtual \
  && rm -rf /var/lib/apt/lists/*

RUN update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20

# QEMU
ENV QEMU_REL=esp-develop-20220919
ENV QEMU_SHA256=f6565d3f0d1e463a63a7f81aec94cce62df662bd42fc7606de4b4418ed55f870
ENV QEMU_DIST=qemu-${QEMU_REL}.tar.bz2
ENV QEMU_URL=https://github.com/espressif/qemu/releases/download/${QEMU_REL}/${QEMU_DIST}

ENV LC_ALL=C.UTF-8
ENV LANG=C.UTF-8

RUN wget --no-verbose ${QEMU_URL} \
  && echo "${QEMU_SHA256} *${QEMU_DIST}" | sha256sum --check --strict - \
  && tar -xf $QEMU_DIST -C /opt \
  && rm ${QEMU_DIST}

ENV PATH=/opt/qemu/bin:${PATH}

#############################
# Additional lines for Matter installation
# https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html

# Prerequisites for Matter
RUN : \
  && apt-get update \
  && apt-get install -y \
  git \
  gcc \
  g++ \
  pkg-config \
  libssl-dev \
  libdbus-1-dev \
  libglib2.0-dev \
  libavahi-client-dev \
  ninja-build \
  python3-venv \
  python3-dev \
  python3-pip \
  unzip \
  libgirepository1.0-dev \
  libcairo2-dev \
  libreadline-dev \
  && apt-get autoremove -y \
  && rm -rf /var/lib/apt/lists/* \
  && : # last line

ARG ESP_MATTER_CLONE_URL=https://github.com/espressif/esp-matter.git

# perform a shallow clone
RUN git clone --depth 1 $ESP_MATTER_CLONE_URL $IDF_TOOLS_PATH/esp-matter && \
	cd $IDF_TOOLS_PATH/esp-matter && \
    git submodule update --init --depth=1 && \
    ./connectedhomeip/connectedhomeip/scripts/checkout_submodules.py --platform esp32 --shallow

# run Matter setup and build host tools and python dependencies for mfg_tool
RUN cd $IDF_TOOLS_PATH/esp-matter && \
    ./install.sh

# ** note: we may need to modify the entrypoint.sh file below to include the esp-matter export.sh
RUN echo "source /opt/esp/esp-matter/export.sh > /dev/null 2>&1" >> ~/.bashrc
RUN echo "source /opt/esp/esp-matter/connectedhomeip/connectedhomeip/scripts/activate.sh  > /dev/null 2>&1" >> ~/.bashrc

#############################
# Back to orginal Dockerfile

RUN echo "source /opt/esp/idf/export.sh > /dev/null 2>&1" >> ~/.bashrc

ENTRYPOINT [ "/opt/esp/entrypoint.sh" ]

CMD ["/bin/bash"]
```

mriksman, 2022-12-29:
Further to this; does `chip-tool` work in VSCode/Dev Container? I am seeing errors; I think related to the network adapter.
```
[1672322082.520893][18955:18957] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on eth0:  ../../src/inet/UDPEndPointImplSockets.cpp:411: OS Error 0x02000063: Cannot assign requested address
[1672322082.520932][18955:18957] CHIP:DIS: mDNS broadcast had only partial success: 1 successes and 1 failures.
[1672322090.524295][18955:18957] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on eth0:  ../../src/inet/UDPEndPointImplSockets.cpp:411: OS Error 0x02000063: Cannot assign requested address
[1672322090.524335][18955:18957] CHIP:DIS: mDNS broadcast had only partial success: 1 successes and 1 failures.
[1672322105.529298][18955:18957] CHIP:CTL: Discovery timed out
```

Perhaps I need some of these (taken from `connectedhomeip` devcontainer.json file)
```
    "runArgs": [
        "--cap-add=SYS_PTRACE",
        "--security-opt",
        "seccomp=unconfined",
        "--network=host",
        "--privileged",
        "-v",
        "/dev/bus/usb:/dev/bus/usb:ro",
        "--device-cgroup-rule=a 189:* rmw",
        "--add-host=host.docker.internal:host-gateway"
    ],
```

brianignacio5, 2023-01-04:
Have you configure the ESP-IDF extension to use esp-matter ? Set the `idf.espMatterPath` in your settings.json to the ESP-matter path. It might be the reason the build button doesn't work. Also check the `ESP-IDF: Doctor command` to see your configuration is correct.

Diegorro98, 2023-01-04:
> I did not try `chip-tool` from docker image. If you have a network interface to that docker image then it should work. (Considering no-ble scenarios). For BLE, yes, you will need to expose the Windows BLE adapter to running docker image. (No idea how to do that)

It is possible to expose Windows BLE, I did it and I was able to run `chip-tool` and perform a device commission through BLE using WSL.
You need to have a Bluetooth adapter that can be attached to WSL using usbipd, you can check it using `usbipd list` (I think that the bast majority of the bluetooth adapters are capable). Then, and this is the hardest part, you will need to recompile the WSL2 kernel in order to enable your required Bluetooth drivers. For more info, check this discussion at usbipd repo: [How to make a USB bluetooth adapter work with WSL](https://github.com/dorssel/usbipd-win/discussions/310)

Diegorro98, 2023-01-04:
> Further to this; does `chip-tool` work in VSCode/Dev Container? I am seeing errors; I think related to the network adapter.
>
> ```
> [1672322082.520893][18955:18957] CHIP:DIS: Warning: Attempt to mDNS [...]

@mriksman In order to make `chip-tool` work in WSL on the Network part (Bluetooth and commissioning part explained in comment above) you will need to setup an Hyper-V virtual switch for WSL and enable IPv6. Despite what people think, is possible right now, but it is undocumented and not officially released.

Diegorro98, 2023-01-04:
Some time ago I made the following guide in order to help Windows devs to prepare their environment to for Matter and I was thinking to commit it to the Matter repo, but I never found the right place to store it. Maybe this issue is a right place: 

# Windows Subsystem for Linux environment setup
As right now, Windows is not supported by the tools and examples from matter, this guide will allow you to use these on Windows through WSL.

Tested on:
- [Ubuntu 20.04 LTS](https://www.microsoft.com/store/productId/9MTTCL66CPXJ "Ubuntu 20.04.4 LTS on Microsoft Store")

## Flashing examples through USB

1. Install udev rules that include your device in `/etc/udev/rules.d` to allow non-sudo users to access the device.

2. To make effective the rules from step 1, start udev service:

		$ sudo service udev restart

	Alternatively, you can start udev service every time the distribution is booted by setting on `/etc/wsl.conf` the following content: 

		[boot]
		command = "service udev restart"

	If you had another command set before, append it with `&&`.

3. [Usbipd](https://github.com/dorssel/usbipd-win) will be used to share the device connected through USB with WSL. By executing the following line on powershell it will be installed:

		PS > winget install usbipd

4. Install requisites on the target distribution

		$ sudo apt install linux-tools-virtual hwdata
		$ sudo update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20

5. Connect the desired device to the target distribution:

		PS > usbipd wsl --distribution <NAME> --busid <BUSID>

	If you don't know the distribution name you can check the names of the distributions installed using this command:

		PS > wsl --list

	Get the list of connected USBs:

		PS > usbipd wsl list

## Chip-tool requirements in WSL
### Commissioning into a network/wifi network
Commissioning into a network or WiFi network on WSL2 with default configurations results in problems, but can they can be solve by using the bridege network mode in WSL2:

**Disclaimer**: WSL bridge mode has not been released yet as an official feature by Microsoft.

1. Enable Hyper-V features on by executing as administrator the following line in powershell:

		PS > Enable-WindowsOptionalFeature -Online -FeatureName Microsoft-Hyper-V -All

2. Restart the host machine.

3. Shutdown wsl if running:

		PS > wsl --shutdown

3. Create an extern virtual switch on the Virtual Switch Manager that is on the Hyper-V Manager.

	Alternatively, run this command on powershell:

		PS > New-VMSwitch -name <VIRTUAL SWITCH NAME>  -NetAdapterName <ADAPTER NAME (i.e., Ethernet, Wi-Fi)> -AllowManagementOS $true

	You can get a list of the available adapter names and other info running `Get-NetAdapter`.

4. At your user folder (`C:\Users\yourUser`, can be accessed running `%USERPROFILE%`), create a txt file named `.wslconfig` if it does not exists yet and set the following parameters:

		[wsl2]
		networkingMode = bridged
		vmSwitch = <VIRTUAL SWITCH NAME>
		ipv6=true

### Commissioning over Bluetooth LE
Commissioning over Bluetooth LE its possible in WSL2 if the Bluetooth adapter from the host machine is available on the USB list on usbipd.

1. Follow these [instructions to build a custom kernel for wsl2](https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel), but in **menuconfig** you will need to enable the following features and those on which they depend (They can change depending on the bluetooth adapter):

	Networking support -> Bluetooth subsystem support (CONFIG_BT)

	Networking support -> Bluetooth subsystem support -> Bluetooth device drivers -> Bluetooth HCI USB driver (CONFIG_BT_HCIBTUSB)

2. Install bluez:

		$ sudo apt-get install bluez

3. Enable Bluetooth:

		$ echo 'export BLUETOOTH_ENABLED=1' | sudo tee /etc/default/bluetooth

4. Start dbus and Bluetooth services:

		$ sudo service dbus start
		$ sudo service bluetooth start

	If you want to start the services every time the distribution is booted, you can setting on `/etc/wsl.conf` the following content: 

		[boot]
		command = "service dbus start && service bluetooth start"

	If you had another command set before, append it with `&&`

5. Attach the Bluetooth device to the WSL2 distribution using [usbipd](https://github.com/dorssel/usbipd-win).
        You can test that the Bluetooth adapter has been correctly connected and is usable by running `bluetoothctl` and its commands (for example `scan on`, which will print the scanned devices by the bluetooth adapter).

Now you have WSL ready to develop Matter devices and test them with chip-tool

mriksman, 2023-01-09:
> Have you configure the ESP-IDF extension to use esp-matter ? Set the `idf.espMatterPath` in your settings.json to the ESP-matter path. It might be the reason the build button doesn't work. Also check the `ESP-IDF: Doctor command` to see your configuration is correct.

If I click the Build button, I get the following
```
-- Found Git: /usr/bin/git (found version "2.25.1") 
CMake Error at /opt/esp/esp-matter/device_hal/device/m5stack/esp_matter_device.cmake:3 (message):
  please set esp32 as the IDF_TARGET using 'idf.py set-target esp32'
Call Stack (most recent call first):
  CMakeLists.txt:32 (include)
```

If I open a Terminal, and type `echo $IDF_TARGET`, I get `esp32c3`. The environment variable set by running `idf.py set-target esp32c3` does not seem to be set when running via the Build command.

The Build command is trying to use `/opt/esp/esp-matter/device_hal/device/m5stack/`. Why? In Terminal, I have run ` export ESP_MATTER_DEVICE_PATH=/opt/esp/esp-matter/device_hal/device/esp32c3_12f_devkit` to set a custom device HAL. Again, this environment variable does not appear to be set when the Build button is pressed in the Activity bar.

Where should I be setting `IDF_TARGET` and `ESP_MATTER_DEVICE_PATH`? In `settings.json`? Or perhaps `devcontainer.json`?

mriksman, 2023-01-09:
@Diegorro98 Wow. Thanks! I actually had already followed your Bluetooth instructions from a post on `usbipd-win`. 

Regarding the Hyper-V virtual switch. My laptop is connected to my Wi-Fi AP/router. Internet works fine. As soon as I run this command `New-VMSwitch -name hyperv -NetAdapterName Wi-Fi -AllowManagementOS $true`, Internet is lost. Network connections show the following
[image]
Is that normal? How can I have my PC still access the Internet over the Wi-Fi, **and** have my ESP32 Matter Accessory connect to the Wi-Fi AP/router and be seen/discoverable by CHIP-Tool running in a Docker Container?

mriksman, 2023-01-09:
@Diegorro98 I can get `bluetoothctl list` working in a `wsl` prompt, but not in a Docker Container (which has my ESP Matter SDK) accessed from VSCode.
[image]
It just hangs when I run `bluetoothctl list`. Anything you can think of off the top of your head?

mriksman, 2023-01-09:
@Diegorro98 Ugh. Bluetooth works fine in WSL
[image]

But not having any luck from within the Docker Container (Dockerfile is the one created by ESP-IDF New Project button in VSCode)
[image]
That `bluetooth` service won't start.

The ESP-IDF Dockerfile is
```
ARG ESP_IDF_TAG
FROM espressif/idf:$ESP_IDF_TAG

ARG DEBIAN_FRONTEND=nointeractive

RUN apt-get update \
  && apt install -y -q \
  cmake \
  git \
  hwdata \
  libglib2.0-0 \
  libnuma1 \
  libpixman-1-0 \
  linux-tools-virtual \
  && rm -rf /var/lib/apt/lists/*

RUN update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20

<snip>

#############################
# Add bluetooth support

RUN apt-get update \
  && apt install -y -q \
  bluez

#############################
# Back to orginal Dockerfile

RUN echo "source /opt/esp/idf/export.sh > /dev/null 2>&1" >> ~/.bashrc
ENTRYPOINT [ "/opt/esp/entrypoint.sh" ]
CMD ["/bin/bash"]
```
Maybe some prerequisite is missing in the Dockerfile?

Diegorro98, 2023-01-09:
> @Diegorro98 Wow. Thanks! I actually had already followed your Bluetooth instructions from a post on `usbipd-win`.
>
> Regarding the Hyper-V virtual switch. My laptop is connected to my Wi-Fi AP/router. [...]

@mriksman Creating a virtual switch over a WiFi adapter can lead to some problems, I will always recommend to do this kind of things using ethernet cable.
I remember that last summer, I got this problem also and I realized that the problem was because I was using fixed IP settings in my WiFi adapter. Just change it to DHCP and the problem might disappear.
If you really need to have a fixed IP, I just have done some testing to solve it. You can try to solve it by verifying that the IP settings at the third adapter from your image (vEtherner (hyperv)) are the ones you need, because maybe something is missing (in my case the gateway and the primary DNS server was missing).

About Bluetooth and docker concern, I have never used docker and bluetooth because WSL and Bluetooth worked great for me; Maybe you can ask about this problem at the usbipd bluetooth discussion.

brianignacio5, 2023-01-10:
> If I click the Build button, I get the following
>
> ```
> -- Found Git: /usr/bin/git (found version "2.25.1")
> CMake Error at /opt/esp/esp-matter/device_hal/device/m5stack/esp_matter_device.cmake:3 [...]

You could set IDF_TARGET and ESP_MATTER_DEVICE_PATH in either. The extension has a command and shows the current IDF_TARGET in the activity bar (blue bar at the bottom).

Apologies for the ESP_MATTER_DEVICE_PATH, we set it to m5stack but should allow the user to configure it. Will post a bug fix asap

mriksman, 2023-01-10:
@brianignacio5 The Activity bar is set to `esp32c3`. So perhaps the Build button failure is because of the `ESP_MATTER_DEVICE_PATH` set to `m5stack` which expects `esp32`.

mriksman, 2023-01-10:
@Diegorro98 Nice! I actually found an article referencing issues with bridging Wi-Fi adapters, with one comment saying DHCP often doesn't work behind it. I switched to Ethernet, used WSL (with Bluetooth attached using `usbipd attach`) and was able to commission a device using CHIP-Tool!

But.

I was only able to issue `toggle` commands for about 30 seconds. Then it stopped working. 
```
>>> onoff toggle 0x7283 0x1
[1673375655.204076][310:312] CHIP:TOO: Sending command to node 0x7283
[1673375655.204158][310:312] CHIP:TOO: Sending cluster (0x00000006) command (0x00000002) on endpoint 1
[1673375655.204211][310:312] CHIP:DMG: ICR moving to [AddingComm]
[1673375655.204218][310:312] CHIP:DMG: ICR moving to [AddedComma]
[1673375655.204270][310:312] CHIP:EM: <<< [E:571i M:7337177] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[1673375655.204300][310:312] CHIP:IN: (S) Sending msg 7337177 on secure session with LSID: 29256
[1673375655.204353][310:312] CHIP:DMG: ICR moving to [CommandSen]
[1673375655.338115][310:312] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1673375655.437287][310:311] CHIP:DL: Indication received, conn = 0x7f7e8004a190
[1673375655.437363][310:312] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1673375655.437452][310:312] CHIP:EM: >>> [E:571i M:248873470] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
[1673375655.437473][310:312] CHIP:EM: Found matching exchange: 571i, Delegate: 0x7f7e8004bb18
[1673375655.437480][310:312] CHIP:DMG: ICR moving to [ResponseRe]
[1673375655.437502][310:312] CHIP:DMG: InvokeResponseMessage =
[1673375655.437513][310:312] CHIP:DMG: {
[1673375655.437516][310:312] CHIP:DMG:  suppressResponse = false,
[1673375655.437518][310:312] CHIP:DMG:  InvokeResponseIBs =
[1673375655.437523][310:312] CHIP:DMG:  [
[1673375655.437541][310:312] CHIP:DMG:          InvokeResponseIB =
[1673375655.437562][310:312] CHIP:DMG:          {
[1673375655.437574][310:312] CHIP:DMG:                  CommandStatusIB =
[1673375655.437577][310:312] CHIP:DMG:                  {
[1673375655.437581][310:312] CHIP:DMG:                          CommandPathIB =
[1673375655.437584][310:312] CHIP:DMG:                          {
[1673375655.437597][310:312] CHIP:DMG:                                  EndpointId = 0x1,
[1673375655.437632][310:312] CHIP:DMG:                                  ClusterId = 0x6,
[1673375655.437636][310:312] CHIP:DMG:                                  CommandId = 0x2,
[1673375655.437637][310:312] CHIP:DMG:                          },
[1673375655.437641][310:312] CHIP:DMG:
[1673375655.437642][310:312] CHIP:DMG:                          StatusIB =
[1673375655.437645][310:312] CHIP:DMG:                          {
[1673375655.437657][310:312] CHIP:DMG:                                  status = 0x00 (SUCCESS),
[1673375655.437659][310:312] CHIP:DMG:                          },
[1673375655.437661][310:312] CHIP:DMG:
[1673375655.437672][310:312] CHIP:DMG:                  },
[1673375655.437675][310:312] CHIP:DMG:
[1673375655.437679][310:312] CHIP:DMG:          },
[1673375655.437682][310:312] CHIP:DMG:
[1673375655.437692][310:312] CHIP:DMG:  ],
[1673375655.437696][310:312] CHIP:DMG:
[1673375655.437707][310:312] CHIP:DMG:  InteractionModelRevision = 1
[1673375655.437717][310:312] CHIP:DMG: },
[1673375655.437730][310:312] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0006 Command=0x0000_0002 Status=0x0
[1673375655.437747][310:312] CHIP:DMG: ICR moving to [AwaitingDe]
[1673375658.019696][310:312] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1673375672.939197][310:312] CHIP:BLE: ack recv timeout, closing ep 0x562a7a269ac0
[1673375672.939236][310:312] CHIP:IN: Clearing BLE pending packets.
[1673375672.940240][310:312] CHIP:DL: HandlePlatformSpecificBLEEvent 16388
[1673375672.940289][310:312] CHIP:BLE: Auto-closing end point's BLE connection.
[1673375672.940293][310:312] CHIP:DL: Closing BLE GATT connection (con 0x7f7e8004a190)
[1673375672.940368][310:311] CHIP:DL: BluezDisconnect peer=CB:B3:8A:AF:8D:FE
[1673375675.570930][310:311] CHIP:DL: Bluez disconnected
[1673375675.570960][310:311] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
>>> onoff toggle 0x7283 0x1
[1673375680.116757][310:312] CHIP:TOO: Sending command to node 0x7283
[1673375680.116825][310:312] CHIP:TOO: Sending cluster (0x00000006) command (0x00000002) on endpoint 1
[1673375680.116859][310:312] CHIP:DMG: ICR moving to [AddingComm]
[1673375680.116879][310:312] CHIP:DMG: ICR moving to [AddedComma]
[1673375680.116920][310:312] CHIP:EM: <<< [E:572i M:7337178] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[1673375680.116955][310:312] CHIP:IN: (S) Sending msg 7337178 on secure session with LSID: 29256
[1673375680.116985][310:312] CHIP:IN: Message appended to BLE send queue
[1673375680.116990][310:312] CHIP:DMG: ICR moving to [CommandSen]
[1673375697.120804][310:312] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 572i
[... 1 lines cut ...]
[1673375697.120874][310:312] CHIP:DMG: ICR moving to [AwaitingDe]
[1673375697.121181][310:310] CHIP:TOO: Run command failure: ../../src/app/CommandSender.cpp:248: CHIP Error 0x00000032: Timeout
```

I can see it in a mDNS browser 
[image]

And I can see it in the Home app; ready to be added. So, it is connected to Wi-Fi, and it isn't running Bluetooth commissioning.
```
I (7490) chip[DL]: IP_EVENT_GOT_IP6
[... cut ...]
I (7530) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 1D783DEF192F2D54.
I (7560) chip[DIS]: mDNS service published: _matterc._udp
```

In CHIP-Tool, I should be able to run this command... right? `>>> pairing onnetwork 0x11 20202021`. But it just times out.
```
[1673378226.130578][374:376] CHIP:DIS: Timeout waiting for mDNS resolution.
[1673378315.125571][374:374] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:512: CHIP Error 0x00000032: Timeout
```
So - is there some issue with my WSL and mDNS....?