espressif/esp-matter #105: How to change setup passcode and discriminator?

Firesuiry, 2022-12-05:
I have 2 esp32c3-devkit-m and I want to have different setup passcode and discriminator.
How can I change it?

Firesuiry, 2022-12-05:
@dhrishi Thank you!
I see
```
esptool.py -p <serial_port> write_flash <address> path/to/<uuid>-partition.bin
```
what is this address?
I want to use the example "light" in esp-matter sdk

dhrishi, 2022-12-05:
@Firesuiry Factory partition, default address 0x3E0000 https://github.com/espressif/esp-matter/blob/main/examples/light/partitions.csv

Firesuiry, 2022-12-05:
@dhrishi I erase-flash and then flash 'light' project and then flash the factory partition
```
esptool.py v3.3.2-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 58:cf:79:e9:96:f8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x003e0000 to 0x003e5fff...
Compressed 24576 bytes to 1701...
Wrote 24576 bytes (1701 compressed) at 0x003e0000 in 0.5 seconds (effective 376.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

and then 
```
matter device factoryreset
```
but the discriminator has not changed.

Is my operation correct?

dhrishi, 2022-12-05:
>matter device factoryreset

Why is this done? Ideally, you should:
1. Erase the flash
2. Flash the firmware using idf.py flash
3. Flash the factory partition at address 0x3E0000 using the esptool.py command

Firesuiry, 2022-12-05:
```
esptool.py -p /dev/ttyUSB0 write_flash 0x3E0000 /home/yinsy/esp/esp-matter/tools/mfg_tool/out/fff2_8001/4f8609a4-a575-4d76-b43c-742387ed56d6/4f8609a4-a575-4d76-b43c-742387ed56d6-partition.bin
```
I use this to flash the factory partition.

shubhamdp, 2022-12-05:
@Firesuiry Can you please check the value of  `CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL`.
If its NVS you are supposed to flash the generated binary at `nvs` address. (Default is 0x10000, please verify your partition table).

If the above is set to `fctry` then it should have been `0x3e0000`.

Firesuiry, 2022-12-05:
@shubhamdp 
```
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="nvs"
```
thank you for your help~  I will try to flash 0x10000

Firesuiry, 2022-12-05:
@shubhamdp 
thank you! It works~
Is there any document that is needed about "CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL"?

shubhamdp, 2022-12-05:
@Firesuiry Is it present but it is in connectedhomeip repository https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/factory_data.md#configuration-options

Will add link to this into esp-matter documentation

dhrishi, 2022-12-05:
@Firesuiry Please close the issue if your original question is answered

Firesuiry, 2022-12-05:
@shubhamdp thank you~
@dhrishi ok~

anuarrozman, 2023-08-31:
hi @Firesuiry did you manage to change setup passcode and discriminator as per guide in [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/factory_data.md#configuration-options) ? Need some insight on how you do it. Thank you !