espressif/esp-matter #106: Window covering device not responding

bernch1nv, 2022-12-05:
Hello,

I am trying to implement a window covering device, based on the light example, which I already got to work on my setup (iPhone with Home-App + HomePod mini as controller). 
It works as far as I can add the device to my Home App, it gets detected as blind, but then it says "device not responding".

The serial log also shows some errors of this kind:
[...]
E (110799) chip[DMG]: Error retrieving data from clusterId: 0x0000_001F, err = b
[...]
E (110929) chip[DMG]: Error retrieving data from clusterId: 0x0000_0028, err = b
[...]

Main looks like this:
```
extern "C" void app_main()
{
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */
    nvs_flash_init();

    /* Initialize driver */
    app_driver_handle_t window_handle = app_driver_window_init(); 

    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, NULL);
    //node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
 
    //window endpoint config
    window_covering_device::config_t window_config;
    endpoint_t *endpoint = window_covering_device::create(node, &window_config, ENDPOINT_FLAG_NONE, window_handle);

    /* These node and endpoint handles can be used to create/add other endpoints and clusters. */
    if (!node || !endpoint) {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

    window_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "Window created with endpoint_id %d", window_endpoint_id);

    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

    /* Starting driver with default values */
    app_driver_window_set_defaults(window_endpoint_id);

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::init();
#endif
}
```
I am still using the QR-code from development guide, and set up all the tools & repos as described there.

Is there maybe anything broken in the implementation on either ESP-Matter side or Apple Home side regarding window_covering_device ?

For testing I just replaced the created window_covering_device enpoint with on_off_plugin_unit endpoint.
As this works, I guess the problem is within the window_covering_device.

dhrishi, 2022-12-05:
@bernch1nv I have just pushed a change that involves window covering changes. Can you please pull, recompile and check. With this, we have verified window opening/closing percentage wise using the Apple Home. Please let us know if it still doesn't work.

bernch1nv, 2022-12-06:
Hello @dhrishi , I did as you said, but unfortunately it still does not work for me, while the behaviour seems to have remained the same

jadhavrohit924, 2022-12-06:
Hi @bernch1nv, We need to add features in main like this:-
```
extern "C" void app_main()
{
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */
    nvs_flash_init();

    /* Initialize driver */
    app_driver_handle_t window_handle = app_driver_window_init(); 

    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, NULL);
    //node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
 
    //window endpoint config
    window_covering_device::config_t window_config;
    endpoint_t *endpoint = window_covering_device::create(node, &window_config, ENDPOINT_FLAG_NONE, window_handle);

    /* These node and endpoint handles can be used to create/add other endpoints and clusters. */
    if (!node || !endpoint) {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

    window_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "Window created with endpoint_id %d", window_endpoint_id);

     /* Add additional features to the node */
     cluster_t *cluster = cluster::get(endpoint, WindowCovering::Id);
     cluster::window_covering::feature::lift::config_t lift;
     cluster::window_covering::feature::tilt::config_t tilt;
     cluster::window_covering::feature::position_aware_lift::config_t position_aware_lift;
     cluster::window_covering::feature::position_aware_tilt::config_t position_aware_tilt;
     cluster::window_covering::feature::absolute_position::config_t absolute_position;
     cluster::window_covering::feature::lift::add(cluster, &lift);
     cluster::window_covering::feature::tilt::add(cluster, &tilt);
     cluster::window_covering::feature::position_aware_lift::add(cluster, &position_aware_lift);
     cluster::window_covering::feature::position_aware_tilt::add(cluster, &position_aware_tilt);
     cluster::window_covering::feature::absolute_position::add(cluster, &absolute_position);

    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

    /* Starting driver with default values */
    app_driver_window_set_defaults(window_endpoint_id);

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::init();
#endif
}
```

Then we can operate though apple home app.
[image]

dhrishi, 2022-12-08:
@bernch1nv Please try it out and close the issue if it works as expected

bernch1nv, 2022-12-08:
@jadhavrohit924 thank you, this works now.
I'm just wondering: how should you know that you have to add those features for it to work?

closed 2022-12-08 by bernch1nv

jadhavrohit924, 2022-12-08:
@bernch1nv Matter spec says, window coverings cluster should have either Lift or Tilt feature enabled.