espressif/esp-matter #108: Generic Switch EP error: IncreaseClusterDataVersion!

jonsmirl, 2022-12-07:
What is this error?
~~~
E (1642) chip[DMG]: Endpoint 2, Cluster 0x0000_001D not found in IncreaseClusterDataVersion!
I (1652) esp_matter_core: Dynamic endpoint 1 added
I (1662) esp_matter_attribute: ********** Endpoint 0x0002's Cluster 0x0003's Attribute 0x0001 is 0 **********
I (1672) esp_matter_core: Dynamic endpoint 2 added
~~~
code ...
~~~
        generic_switch::config_t generic_config;
        endpoint_t *endpoint1 = generic_switch::create(node, &generic_config, ENDPOINT_FLAG_NONE, button_handle);
        cluster::fixed_label::create(endpoint1, NULL, CLUSTER_FLAG_SERVER);
        cluster::user_label::create(endpoint1, NULL, CLUSTER_FLAG_SERVER);
        set_parent_endpoint(endpoint, endpoint1);
~~~

jadhavrohit924, 2022-12-09:
@jonsmirl Please try by moving set_parent_endpoint after esp_matter::start().

wqx6, 2022-12-12:
Can you please show me the code that you create the `endpoint`? In `esp-matter::start()`, the endpoints will be enabled in the order that they are created. If you create the `endpoint` before the `endpoint1`, the parent endpoint (which is `endpoint1` in your code) of the `endpoint` is not enabled when enabling the `endpoint`. That's the reason of your error. Please create the parent endpoints before the child endpoints.

dhrishi, 2022-12-13:
@jonsmirl Can you please confirm the above. Please close the issue if it is working as expected

jonsmirl, 2022-12-13:
Moving this after esp_matter start fixes the issue.