espressif/esp-matter #110: The callback function cannot get the target value

shun-dgut, 2022-12-08:
## 使用 WindowCovering 例子 做了如下修改
components/esp_matter/esp_matter_cluster.cpp
```
    feature::lift::add(cluster, &(config->lift));
    feature::tilt::add(cluster, &(config->tilt));
    feature::position_aware_lift::add(cluster, &(config->position_aware_lift));
    feature::position_aware_tilt::add(cluster, &(config->position_aware_tilt));
    feature::absolute_position::add(cluster, &(config->absolute_position));
```
components/esp_matter/esp_matter_cluster.h
```
    feature::lift::config_t  lift;
    feature::tilt::config_t  tilt;
    feature::position_aware_lift::config_t  position_aware_lift;
    feature::position_aware_tilt::config_t  position_aware_tilt;
    feature::absolute_position::config_t    absolute_position;
```

## Log
[image]

When I modify the TargetPositionLiftPercent100ths value of the curtain, I receive an illegal value

As shown in the figure below, I understand that the value will be passed to the application normally
[image]

jadhavrohit924, 2022-12-09:
Can you please point me to code file?? Or you are trying to set TargetPositionLiftPercent100ths ?

Please note : TargetPositionLiftPercent100ths is read only attribute.

shun-dgut, 2022-12-10:
> I didn’t get the above log. What are you trying to do? Can you please share more detailed steps?
> 
> Please note : TargetPositionLiftPercent100ths is read only attribute.
I want to get the data in the application layer, but I can't get the value in the callback function.

[image]

closed 2022-12-11 by shun-dgut