espressif/esp-matter #112: Fixed and User labels -- how to set?

jonsmirl, 2022-12-10:
~~~
    esp_matter_attr_val_t val = esp_matter_array((uint8_t*)"MyFixedLabel", strlen("MyFixedLabel"), 1);
    err = attribute::update(1, FixedLabel::Id, FixedLabel::Attributes::LabelList::Id, &val);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Fixed label failed: %d", err);
    }
~~~
gives error:
~~~
I (2189) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0040's Attribute 0x0000 is <invalid type: 4> **********
~~~

wqx6, 2022-12-12:
Fixed Labels are stored in factory nvs partition, and the factory partition binary files can be generated by [mfg_tool](https://github.com/espressif/esp-matter/blob/main/tools/mfg_tool/mfg_tool.py#L486). The user labels are always empty when initializing and added by the controller.

closed 2022-12-13 by jonsmirl

FHauschild, 2023-12-08:
Hi,

thanks for the fast answer.

so I should follow the generic switch example?

```
if (nvs_get_i32(handle, "fl-sz/1", &out_value) == ESP_ERR_NVS_NOT_FOUND)
    {
       nvs_set_i32(handle, "fl-sz/1", 2);
       nvs_set_str(handle, "fl-k/1/0", "myEP1LBL1");
       nvs_set_str(handle, "fl-v/1/0", "valEP1LBL1");
       nvs_set_str(handle, "fl-k/1/1", "myEP1LBL2");
       nvs_set_str(handle, "fl-v/1/1", "valEP1LBL2");
    }
```

Is there any documantation about the nvs lables?

I think "fl-sz" is the number of Lables in the List
"fl-k" is the FixedLabel key
"fl-v" is the FixedLabel value
The first number is the endpoint Id
And the second Number is the INdex for the List

Am I correct?

I've read in Espressif's documentation that there is a way to provie a custom DeviceInfo Provider.
Are there any examples for writing a custom DeviceInfo Provider?