espressif/esp-matter #113: How to send events from switch cluster?

jonsmirl, 2022-12-10:
I used chip-tool to commission and create a subscription...
~~~
 switch subscribe-event initial-press 0 1000 12345 1
~~~
then in the device I sent an event...
~~~
        switch_cluster::event::send_initial_press(1, new_position);
~~~
Chip-tool doesn't receive anything.

jonsmirl, 2022-12-10:
~~~
 switch_cluster::event::send_initial_press(1, new_position);
~~~
Is returning 0x5000102, not sure how to interpret that.

jonsmirl, 2022-12-11:
I (49018) chip[DMG]: Refresh Subscribe Sync Timer with min 0 seconds and max 30 seconds

After further testing,  it is not honoring a minimum sync interval. It is only implementing the max. So there is a bug here in the CHIP code.

bzbarsky-apple, 2022-12-12:
If you want events at min-interval, given the current state of the spec and SDK, you need to set the "is urgent" flag on the subscription paths for those events.  For `chip-tool`, use `--is-urgent true`.

jonsmirl, 2022-12-12:
No one is going to figure out  "is urgent"  without being told about it.  It's like min-interval is just ignored unless you use the  "is urgent"  flag.

I had not noticed wildcard subscriptions. I will need to read and see if I can get events from all eight of my keypad buttons in a single subscription. If that works, my worries about subscriptions will be much smaller.

dhrishi, 2022-12-13:
Closing this as the original issue is resolved.

closed 2022-12-13 by dhrishi