espressif/esp-matter #114: About Flash's partition table division

shun-dgut, 2022-12-12:
What are the functions of sec_cert and fctry of the partition table? Isn’t it enough to store the DAC certificate in the fctry partition?

dhrishi, 2022-12-13:
The PAI, DAC and DAC private key (Matter attestation PKI) can either be in the factory or in the encrypted sec_cert partition. The partition name is now updated to esp_secure_cert.

esp_secure_cert partition uses the [ESP Secure Certificate Manager](https://github.com/espressif/esp_secure_cert_mgr) for the Matter attestation PKI. The ESP pre-provisioning service makes use of that partition.

shun-dgut, 2022-12-13:
> The PAI, DAC and DAC private key (Matter attestation PKI) can either be in the factory or in the encrypted sec_cert partition. The partition name is now updated to esp_secure_cert.
>
> esp_secure_cert [...]

Does that mean that my product only needs the fctry partition, remove the  esp_secure_cert partition?

dhrishi, 2022-12-13:
If your product is going to get the DAC from Espressif using our [pre-provisioning service](https://www.espressif.com/en/news/Pre-provisioning_Matter) then that will be required. If not, then you can remove the partition. Just note that DAC private key should always be safeguarded, whichever CSA approved DAC provider you choose.

closed 2022-12-14 by dhrishi

shun-dgut, 2022-12-14:
> If your product is going to get the DAC from Espressif using our [pre-provisioning service](https://www.espressif.com/en/news/Pre-provisioning_Matter) then that will be required. If not, then you [...]

But the fctry partition in the example is not encrypted, does it mean that I need to realize the encryption of this partition.