espressif/esp-matter #117: Bluetooth GATT connection with BLE device best practices

Diegorro98, 2022-12-13:
I'm trying to make a Matter device that communicates with another Bluetooth Low Energy device. I would like to know what are the best practices to achieve this because there could be some conflicts with CHIPoBLE.
Right now the first thing that the firmware does is connecting to the Bluetooth device, but because of this, sometimes when commissioning ends, the bluetooth device to which I connect is disconnected (reason 8, I tried to find the meaning of this error code but I didn't find any info). But it works right again after a restart. I also receive `esp_matter_core: BLE deinit failed` on the console logs when commissioning is completed.

Although it usually works, maybe it would be better to perform the connection between the esp32 and the BLE device in another moment, but which moment is the right one and how I can get to that moment?

I was thinking that maybe there is a `ChipDeviceEvent` where I can safely perform this connection when it gets raised.
`kCHIPoBLEConnectionClosed` is a good candidate but this doesn't raise after restarting once the esp32 is commissioned, maybe there is any variable to know if the device is already commissioned.

jonsmirl, 2022-12-14:
Support for BLE Mesh is already in esp-matter. It would likely be easier to switch your device over to BLE Mesh. 
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_sensor_model/sensor_client/README.md

Note that battery powered devices in BLE Mesh don't have to stay turned on and route mesh packets, they are leaf nodes. All of the chip vendors support BLE Mesh in addition to normal BLE.

If instead you want to do a lot more work, convert the BLE Mesh example over to normal BLE.
https://github.com/espressif/esp-matter/tree/main/examples/blemesh_bridge

Why BLE Mesh? Consider BLE trackers that move around. They will bind to the nearest mesh node. In normal BLE they are stuck to a single node.

dhrishi, 2022-12-15:
@Diegorro98 

>I'm trying to make a Matter device that communicates with another Bluetooth Low Energy device.

When is this communication expected to happen? Anytime or something specific like post commissioning (in which case, Matter won't require BLE)

> I would like to know what are the best practices to achieve this because there could be some conflicts with CHIPoBLE.

You can create a custom/external platform and include your BLE advertisement data and characteristics. You can check the BLE Mesh to Matter bridge example for reference

>but because of this, sometimes when commissioning ends, the bluetooth device to which I connect is disconnected (reason 8, I tried to find the meaning of this error code but I didn't find any info). 

If the disconnect is called by the device, then may be, you can handle this in the custom platform.

> I also receive esp_matter_core: BLE deinit failed

Currentlly, by default, we disable and deinit BLE after commissioning. If you need BLE in your application, you can unset this option `idf.py menuconfig -> Component config -> CHIP Device Layer -> BLE Options`, re-build and re-flash

Diegorro98, 2022-12-17:
My use case is a esp32 that receives commands from matter and then sends commands to an actuator (not a sensor, I can't even get info from the device). So its a bridge, but I don't know if its like a Matter bridge because I don't understand yet what a Matter bridge means, and this makes me ask my self if endpoints should use the bridge flag in that case. 

The bluetooth device can't be modified because it's not something I built and as far as I now, it is not modifiable.

I just have used disabled the option that disables BLE after commissioning and the mentions to ble deinit have been dissapeared, so that's cool, but I still facing some conflict problems.

dhrishi, 2023-04-30:
@Diegorro98 Totally missed the response here. Sorry about that. Please close the issue if it is now resolved or let us know the current state. Thanks!

closed 2023-04-30 by Diegorro98