espressif/esp-matter #118: mfg_tool examples

jonsmirl, 2022-12-13:
If you change the examples to use the environment variable you will be able to copy and use them....
~~~
./mfg_tool.py -cn "My bulb" -v 0xFFF2 -p 0x8001 --pai     -k $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem     -c $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem     -cd $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der     --passcode 20202021 --discriminator 3840  --enable-rotating-device-id --rd-id-uid d2f351f57bb9387445a5f92a601d1c14
~~~
instead of 
~~~
./mfg_tool.py -cn "My bulb" -v 0xFFF2 -p 0x8001 --pai \
    -k path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
    -c path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
    -cd path/to/esp-matter/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
    --passcode 20202021 --discriminator 3840  --enable-rotating-device-id --rd-id-uid d2f351f57bb9387445a5f92a601d1c14
~~~

jonsmirl, 2022-12-13:
I just discovered mfg_tool and see now that I can use it to set things like Product and Vendor name. I think mfg_tool should be combined into the examples as part of the cmake process. First come up with a set of workable defaults and explicitly set those in the cmake file. Then as I need to make changes I can edit the cmake file to implement them.

For certificates it can initially use the sample ones. Or can you generate new ones?

Don't esp chip have hardware serial numbers? Option to use that?
~~~
            # If serial number is not passed, then generate one
            if (args.serial_num is None):
                chip_factory_update('serial-num', binascii.b2a_hex(os.urandom(SERIAL_NUMBER_LEN)).decode('utf-8'))
Don't save this tool until the end of the process, include it from the begining.
~~~

edWin-m, 2023-02-24:
@jonsmirl did you figure out where and how to include the generated `<uuid>-partition.bin`?

dhrishi, 2023-04-30:
@edWin-m The factory partition binary is to be flashed at the factory partition address in the partition table.

Closing this issue as the examples are added: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/production.html#the-mfg-tool-example

closed 2023-04-30 by dhrishi

edWin-m, 2023-05-04:
@dhrishi Got it. Thanks for the update.