espressif/esp-matter #119: Fixed label

jonsmirl, 2022-12-13:
I have a keypad with eight generic switches and each has a Fixed Label. Say I want to name them "SW1"..."SW8". mfg_tool has this:
~~~
sage: mfg_tool.py [-h] [-n COUNT] [-s SIZE] [--passcode PASSCODE] [--discriminator DISCRIMINATOR] [-cf {0,1,2}] [-dm {0,1,2}]
                   [-cn CN_PREFIX] [-lt LIFETIME] [-vf VALID_FROM] [-c CERT] [-k KEY] -cd CERT_DCLRN [--dac-cert DAC_CERT]
                   [--dac-key DAC_KEY] [--paa | --pai] [-v VENDOR_ID] [--vendor-name VENDOR_NAME] [-p PRODUCT_ID]
                   [--product-name PRODUCT_NAME] [--hw-ver HW_VER] [--hw-ver-str HW_VER_STR] [--mfg-date MFG_DATE]
                   [--serial-num SERIAL_NUM] [--enable-rotating-device-id] [--rd-id-uid RD_ID_UID]
                   [--calendar-types CALENDAR_TYPES [CALENDAR_TYPES ...]] [--locales LOCALES [LOCALES ...]]
                   [--fixed-labels FIXED_LABELS [FIXED_LABELS ...]] [--product-label PRODUCT_LABEL] [--product-url PRODUCT_URL]
                   [--csv CSV] [--mcsv MCSV]
~~~

So how does "--fixed-labels FIXED_LABELS [FIXED_LABELS ...]] " map to the generic switches? Is this mapped by the order I create the Fixed Labels? Or are these multiple labels for translations and you don't handle multiple clusters with fixed labels?

jonsmirl, 2022-12-14:
There is a problem here with locales and fixed-labels. If you set a different locale, you need a different set of fixed labels.

shubhamdp, 2022-12-14:
How are those dependent? Both are in different clusters.

shubhamdp, 2022-12-14:
As per the specification, fixed labels are read-only attributes, so I am not sure how you want to change it.

But, you can have a list of fixed labels on you device and based on locale the commissioner/controller should figure out the one it wants to use.

Also, the options in the mfg_tool lets to add supported_locales and not the active locale.

jonsmirl, 2022-12-14:
When the controller sets the active locale into the device, shouldn't the fixed labels change to match that locale?

shubhamdp, 2022-12-14:
Then this is not just about fixed-labels or extents to every possible string the controller can read.
I am not sure how to maintain that in the SDK.

jonsmirl, 2022-12-14:
Boris wrote back to me -- the labels are truly fixed, they don't change to match locale.    But... there are proposals to change this.

shubhamdp, 2022-12-14:
Probably once we have some clarity on that we can incorporate that into the script.

closed 2022-12-14 by jonsmirl