espressif/esp-matter #120: Lift & tilt are required for absolute positioning at window covering?

Diegorro98, 2022-12-14:
I've been testing with window covering device types and have found that I can't add the absolute positioning feature if I only support lift (and position aware lift) or tilt (and position aware tilt).

I can't find at the Matter specification any conformance that force you to use both features if you want to use absolute positioning.

jadhavrohit924, 2022-12-15:
@Diegorro98 If you check the attributes of window covering cluster, attributes having conferences absolute positioning also have features conferences lift (and position aware lift) or tilt (and position aware tilt).

Thats why we decided to keep lift/tilt features as a base.

Also matter spec says `At least one of the Lift and Tilt features SHALL be supported` So we cannot use only absolute positioning. Lift or Tilt feature is required for window covering.

jadhavrohit924, 2022-12-15:
We can use absolute positioning with `lift and position aware lift` and/or `tilt and position aware tilt`.

dhrishi, 2022-12-15:
> We can use absolute positioning with `lift and position aware lift` and/or `tilt and position aware tilt`.

@jadhavrohit924 How and using which commissioner has this been tested?

jadhavrohit924, 2022-12-15:
This has been tested with google and amazon commissioner

dhrishi, 2022-12-15:
@Diegorro98 Can you share the code you've written for this? We can try out with the same

Diegorro98, 2022-12-15:
Sure, this is my main code:
```
extern "C" void app_main() {
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */
    nvs_flash_init();

    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    window_covering_device::config_t device_config;
    device_config.window_covering.type = (uint8_t)chip::app::Clusters::WindowCovering::Type::kAwning;
    device_config.window_covering.config_status =
        (uint8_t)chip::app::Clusters::WindowCovering::ConfigStatus::kOperational
        | (uint8_t)chip::app::Clusters::WindowCovering::ConfigStatus::kTiltPositionAware;
    device_config.window_covering.end_product_type = 
        (uint8_t)chip::app::Clusters::WindowCovering::EndProductType::kUnknown;
    endpoint_t *device_endpoint = window_covering_device::create(node, &device_config, ENDPOINT_FLAG_NONE, NULL);

    /* These node and ~endpoint~ handles can be used to create/add other endpoints and clusters. */
    if (!node || !device_endpoint) {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

    device_endpoint_id = endpoint::get_id(device_endpoint);
    ESP_LOGI(TAG, "Back rest created with endpoint_id %d", device_endpoint_id);

    /* Add additional features to the node */
    cluster_t *device_cluster = cluster::get(device_endpoint, WindowCovering::Id);
    cluster::window_covering::feature::tilt::config_t device_tilt;
    cluster::window_covering::feature::tilt::add(device_cluster, &device_tilt);
    cluster::window_covering::feature::position_aware_tilt::config_t device_position_aware_tilt;
    cluster::window_covering::feature::position_aware_tilt::add(device_cluster, &device_position_aware_tilt);
    cluster::window_covering::feature::absolute_position::config_t device_absolute_position;
    cluster::window_covering::feature::absolute_position::add(device_cluster, &device_absolute_position);
    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::init();
#endif
}
```

With the above code, I get an error log that says `esp_matter_feature: Cluster shall support Position_Aware_Lift feature` (if I write the same code for lift only, I get the same log error but refearing to `Position_Aware_Tilt`).
Because @jadhavrohit924 said

> Also matter spec says `At least one of the Lift and Tilt features SHALL be supported`

I have dived into esp_matter_feature.cpp to find the error, as I found where the logic was failing, I fixed it and created a pull request to solve this error and other errors I've found related to `window_covering`.

dhrishi, 2022-12-15:
>I have dived into esp_matter_feature.cpp to find the error, as I found where the logic was failing, I fixed it and created a pull request to solve this error and other errors I've found related to `window_covering`.

Thank you very much :-)

closed 2022-12-22 by espressif-bot