espressif/esp-matter #121: Matter /esp32

hadheemi, 2022-12-14:
hello everyone ; 
I want to know if I have a smart plug and I want to detect it through my esp32 and connect it to the cloud , meaning that my esp32 acts like a gateway between the plug and the cloud . what is the process to do that I cant find alot of information in the documentation

alibukharai, 2022-12-15:
@hadheemi check the related examples in [esp-idf](https://github.com/espressif/esp-idf/tree/master/examples)

hadheemi, 2022-12-15:
thank you for your answers . actually i've tested the light example and it works correctl i controlled the light via chiptool android app . my problem is how is the process of the binding between the matter end device (plug) and the gateway (esp32)?

jonsmirl, 2022-12-15:
Where is the repository for the S3 Box controller example?
https://www.youtube.com/watch?v=Jr4Lut_NgqA

dhrishi, 2022-12-16:
>my problem is how is the process of the binding between the matter end device (plug) and the gateway (esp32)?

@hadheemi The steps [here](https://github.com/espressif/esp-matter/tree/main/examples/light_switch#21-bind-light-to-switch) will help you set up ACL and Binding between the end device and the controller so that the device automations are set up

dhrishi, 2022-12-16:
>https://github.com/espressif/esp-matter/tree/main/examples/light_switch#21-bind-light-to-switch

@jonsmirl This is not released yet. It was the light_switch code + [S3 box example code](https://github.com/espressif/esp-box/tree/master/examples/factory_demo) integrated together

dhrishi, 2022-12-22:
@hadheemi Please close this if your questions are answered

dhrishi, 2023-01-06:
Closing. Feel free to re-open in case of anything

closed 2023-01-06 by dhrishi