espressif/esp-matter #122: ESP32S3 Commissioning and Connection failed.

aidnani, 2022-12-14:


aidnani, 2022-12-14:
I am using ESP32S3. I was able to flash the firmware on it. When I use chip-tool I was not able to commission the device. I used 'matter wifi' command to provide wifi credentials. Now when I use 'chip-tool pairing onnetwork 0x7283 20202021' I keeps timing out. See logs.

I am using idf.py version 4.4.2
My esp-matter shows a ORIG_HEAD of 4cc86d11ee7b210458762d6c803ebf416e354c8f
[Esp32S3-log.txt](https://github.com/espressif/esp-matter/files/10232344/Esp32S3-log.txt)
[chip-tool-log.txt](https://github.com/espressif/esp-matter/files/10232345/chip-tool-log.txt)

dhrishi, 2022-12-14:
@aidnani Can you confirm if the host machine running the chip-tool is connected to the same Access Point to which the device is connected? If not, then the chip-tool won't be able to discover the device over IP and there will be a timeout

aidnani, 2022-12-15:
Good point. I was not connected to the same AP, but it was the same subnet. I will anyway try again by connecting to the same AP.

dhrishi, 2022-12-16:
Thanks for the confirmation @aidnani

closed 2022-12-16 by dhrishi