espressif/esp-matter #125: Unable to connect to ESP32S3 using iOS app.

aidnani, 2022-12-16:
I am using iOS app 'ESP Matter' to commission my device. I can see that when I scan the barcode for my device the app briefly connects, but then the connection is dropped immediately with an error. Here are the log lines from ESP32S3. What could be wrong?

 I (421798) chip[DL]: BLE GAP connection established (con 1)
I (421798) chip[DL]: CHIPoBLE advertising stopped
I (422728) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (422728) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (422738) NimBLE: GAP procedure initiated: advertise; 
I (422738) NimBLE: disc_mode=2
I (422738) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (422758) NimBLE: 

I (422758) chip[DL]: CHIPoBLE advertising started

dhrishi, 2022-12-16:
The ESP Matter iOS app is based on older codebase and won't work with the later esp-matter codebase. We are in the process of upgrading the app on the iOS store.

For now, you can use Google or Apple as the Matter commissioner and try it out. Note that, you will need a hub in both the cases.
Google: https://developers.home.google.com/matter
Apple: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md

aidnani, 2022-12-16:
@dhrishi Thank you.

closed 2022-12-17 by dhrishi

mriksman, 2023-01-09:
> @dhrishi Thank you.
> 
> @dhrishi I know you closed. this. Just wondering why is a hub needed, if the ESP device has both radios BLE and WiFi (and is not really a Thread) device?

I have the same question! I can't find any reference as to why a Hub is required. I created a new 'Home' in Apple, and tried to add a Matter device. It wouldn't let me. I added a HomePod Mini, and then powered it off. It added, and appeared in the Home app, but had 'No Response'. It wasn't until the HomePod Mini was powered up that it started working. Curious as to why...

mriksman, 2023-01-09:
> The ESP Matter iOS app is based on older codebase and won't work with the later esp-matter codebase. We are in the process of upgrading the app on the iOS store.
>
> For now, you can use Google or [...]

@dhrishi Any ETA on the Espressif Matter iOS app being updated? I see it supports 'binding' which I really want to try. Given that I develop on Windows, I haven't worked out how to use CHIP-Tool (docker containers don't support IPv6 nor bridged networks - so how could it ever see an external ESP dev board??)