espressif/esp-matter #127: Adding three child devices does not respond

shun-dgut, 2022-12-19:
iphone  ios 16.2 + homepod mini 16.2
use app_bridge_create_bridged_device function to create ble mesh sub-devices, creating three different sub-devices.
Adding a device using home shows no response.

If I reduce to two devices, then it is possible to display the response,and send and receive data normally

wqx6, 2022-12-19:
Hi, Can you provide your device log of adding three child devices on blemesh bridge? I did not get the error of no-responding when I add three child-endpoints to to aggregator endpoint. My test environment is iphone-xr with IOS16.1 and homepod mini 16.2. Also I have tested the bridge app with the chip-tool and got no error.

shun-dgut, 2022-12-19:
> Hi, Can you provide your device log of adding three child devices on blemesh bridge? I did not get the error of no-responding when I add three child-endpoints to to aggregator endpoint. My test [...]

I used the newly added door_lock device

```
esp_matter_bridge.cpp 

case ESP_MATTER_DOOR_LOCK_DEVICE_TYPE_ID:{
        door_lock::config_t door_lock_conf;
        bridged_device->endpoint = door_lock::add(bridged_device->endpoint, &door_lock_conf);
        break;
    }
```

jonsmirl, 2022-12-19:
this could be related to: https://github.com/espressif/esp-matter/issues/77

shun-dgut, 2022-12-20:
> Hi, Can you provide your device log of adding three child devices on blemesh bridge? I did not get the error of no-responding when I add three child-endpoints to to aggregator endpoint. My test [...]

This is the log.
[3 lock dev.txt](https://github.com/espressif/esp-matter/files/10264222/3.lock.dev.txt)

shun-dgut, 2022-12-20:
[3 lock dev chiptool.txt](https://github.com/espressif/esp-matter/files/10264436/3.lock.dev.chiptool.txt)
This is chip tool log

wqx6, 2022-12-20:
Have you defined override functions for `emberAfPluginDoorLockOnDoorLockCommand` and `emberAfPluginDoorLockOnDoorUnlockCommand` which are weakly defined [here](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/door-lock-server/door-lock-server-callback.cpp#L49-L61)? You need to define your callbacks otherwise the `lockdoor` and `unlockdoor` commands will always return failure.

wqx6, 2022-12-20:
I suggest you to add these functions in main component instead of the `door-lock-server.cpp`. And you can also create class [`LockEndpoint`](https://github.com/project-chip/connectedhomeip/blob/master/examples/lock-app/linux/src/LockEndpoint.cpp) and class [`LockManager`](https://github.com/project-chip/connectedhomeip/blob/master/examples/lock-app/linux/src/LockManager.cpp) like the upstream linux door-lock app.

shun-dgut, 2022-12-20:
> I suggest you to add these functions in main component instead of the `door-lock-server.cpp`. And you can also create class [...]

Ok, I'll try this, but I can add a door_lock device and control it.

shun-dgut, 2022-12-20:
> I suggest you to add these functions in main component instead of the `door-lock-server.cpp`. And you can also create class [...]

It is not possible for me to use the following code

```
  app_bridge_create_bridged_device(node, aggregator_endpoint_id, ESP_MATTER_WINDOW_COVERING_DEVICE_TYPE_ID,
                                                 ESP_MATTER_BRIDGED_DEVICE_TYPE_BLEMESH,
                                                 app_bridge_blemesh_address(0x0012));
    app_bridge_create_bridged_device(node, aggregator_endpoint_id, ESP_MATTER_DOOR_LOCK_DEVICE_TYPE_ID,
                                                 ESP_MATTER_BRIDGED_DEVICE_TYPE_BLEMESH,
                                                 app_bridge_blemesh_address(0x0013));        
    app_bridge_create_bridged_device(node, aggregator_endpoint_id, ESP_MATTER_ON_OFF_LIGHT_DEVICE_TYPE_ID,
                                                 ESP_MATTER_BRIDGED_DEVICE_TYPE_BLEMESH,
                                                 app_bridge_blemesh_address(0x0014));
```

The following code is ok

```
app_bridge_create_bridged_device(node, aggregator_endpoint_id, ESP_MATTER_DOOR_LOCK_DEVICE_TYPE_ID,
                                                 ESP_MATTER_BRIDGED_DEVICE_TYPE_BLEMESH,
                                                 app_bridge_blemesh_address(0x0012));
    app_bridge_create_bridged_device(node, aggregator_endpoint_id, ESP_MATTER_DOOR_LOCK_DEVICE_TYPE_ID,
                                                 ESP_MATTER_BRIDGED_DEVICE_TYPE_BLEMESH,
                                                 app_bridge_blemesh_address(0x0013));        
    app_bridge_create_bridged_device(node, aggregator_endpoint_id, ESP_MATTER_DOOR_LOCK_DEVICE_TYPE_ID,
                                                 ESP_MATTER_BRIDGED_DEVICE_TYPE_BLEMESH,
                                                 app_bridge_blemesh_address(0x0014));  
```

shun-dgut, 2022-12-20:
Three light sub-settings are acceptable, three lock sub-settings are also normal, and three windowcovering sub-settings are also normal.

wqx6, 2022-12-21:
@shun-dgut So problem will happen if you add three endpoints with different device type on one node? Then it could be the issue of the homepod mini or Apple Home app.

shun-dgut, 2022-12-23:
> @shun-dgut So problem will happen if you add three endpoints with different device type on one node? Then it could be the issue of the homepod mini or Apple Home app.

will not appear. Can you provide an example of creating lock and curtain?

wqx6, 2023-01-31:
> Can you provide an example of creating lock and curtain?

@shun-dgut We provided an example for doorlock manage [here](https://github.com/project-chip/connectedhomeip/tree/master/examples/platform/esp32/lock) in upstream repo.

> The problem only occurs with Apple home with Google home working fine

@stefano055415 Then it could be the issue of Apple software component. We will test that later.

dhrishi, 2023-02-28:
@shun-dgut  @wqx6 Is anything pending here? Or it can be closed if it's not an issue on the device side?

dhrishi, 2023-04-30:
Closing this. Please re-open in case of anything

closed 2023-04-30 by dhrishi