espressif/esp-matter #129: Devices are showing Offline

SYESHASWINI, 2022-12-20:
I have flashed Light example to Esp32c3 board and added the device to google home. It is successfully added, but the device is showing offline. I have factory reset the nest hub,  updated google home and tried, but nothing worked. Requesting for the suggestions.

Board: esp32c3
Nest Hub version: 1.56.324896

[image]

Thank You.

dhrishi, 2023-01-06:
@SYESHASWINI Do you still see any issue with the latest codebase? We can get it to work as expected

closed 2023-01-09 by SYESHASWINI

sidhoo2, 2023-05-28:
@dhrishi Is there any commits for this issue? i found the device hv not get the ipv6 addr when the Google Home show Offline