espressif/esp-matter #130: Matter device showing offline in apple homekit

SYESHASWINI, 2022-12-20:
I have flashed matter Light example to Esp32c3 board and added the device to apple homekit. It is successfully added, but the device is showing offline. I'm using home pod to control the device. I have factory reset and updated home pod, but nothing worked. Requesting for the suggestions.

Board: esp32c3
Home pod version:16.2
[image]

shubhamdp, 2022-12-20:
@SYESHASWINI I tried three scenarios and device is reachable in all three.
1. Homepod, Phone, and ESP32C3 in same network and connected to internet
2. Homepod, phone, and ESP32C3 in same network and no internet connectivity
3. Homepod and ESP32 in different wifi network and connected to internet. Phone in different wifi network and connected to internet.

Can you try flashing the matter light example from here: https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml

SYESHASWINI, 2022-12-21:
@shubhamdp With launchpad its working fine. When I flash the code using terminal/cmd the devices are showing offline. Is there any other approach for it to make device reachable.

Thank You

dhrishi, 2022-12-21:
@SYESHASWINI The binaries on the Launchpad are based on the local codebase. So, those are the the default ones.

Can you please follow the steps below:

```
$ cd /path/to/esp-matter/examples/light
$ rm -rf sdkconfig sdkconfig.old build/
$ idf.py set-target esp32c3
$ idf.py erase-flash
$ idf.py build flash monitor
```

SYESHASWINI, 2022-12-21:
@dhrishi I tried this, but after may trails the device is reachable. The device is not reachable everytime

SYESHASWINI, 2022-12-21:
@dhrishi I have tried as you said, but after many trials the device is reachable.  Everytime I flash the device, very often the device is unreachable only few times its reachable. 

I have made few changes in the sdk so that it have 4 endpoints. Sometimes the device is added with only 3 endpoints.
Any solutions for this.

dhrishi, 2022-12-21:
@SYESHASWINI Please close this issue and create a new one with the details of the changes you've done (patch)

closed 2022-12-21 by SYESHASWINI