espressif/esp-matter #131: Rapid updates to persistent attributes

jonsmirl, 2022-12-21:
Per Boris, Espressif needs to implement an equivalent to this:
https://github.com/project-chip/connectedhomeip/pull/23366

My comment would also be that writing an attribute should not sleep in the thread doing the writing. That's because CHIP writes attributes from timer ticks and if you sleep in a timer tick it will go recursive. 

In
     app/clusters/level-control/level-control.cpp
there is this routine
     void emberAfLevelControlClusterServerTickCallback(EndpointId endpoint)
That routine assumes it is going to run in a millisecond or so. But on
my devices it is taking over a second to run. The problem is this
line:
       status = Attributes::CurrentLevel::Set(endpoint, currentLevel);
Flow goes into that Set() call and doesn't come back out for about a
second. That doesn't work very well in a tick handler routine. So why
is it blocking in that call?  If I comment that call out it works as
expected.

I think you should be able to demonstrate this problem using chip-tool
but I haven't tried it yet. Send a Move command to the light-bulb demo
and vary the rate parameter on the command. Note that it always moves
the brightness much slowly than the rate tells it to. For example a
rate of 100 should count from 0 to 254 in 2.5 seconds.

My guess what is happening? currentLevel is marked as being persistent
so it is being written to flash on every call. That flash write has a
commit_sync after it which could be blocking for a long time.

jonsmirl, 2022-12-22:
This problem is much more involved, it is problem in CHIP core.
https://github.com/project-chip/connectedhomeip/issues/24193

jonsmirl, 2022-12-23:
Updating to 1.0.0.1 will pick up the deferred persistence support.
Once you have the framework in your tree, you still need Espressif specific driver.

dhrishi, 2022-12-23:
>Once you have the framework in your tree, you still need Espressif specific driver.

Ack

shubhamdp, 2024-01-11:
@jonsmirl we've added the support for deferred attributes in https://github.com/espressif/esp-matter/commit/80a0868271965e1959bf9ffc7743a2380f0f86aa. This issue can be closed now.

jonsmirl, 2024-01-11:
You don't want to set these obvious lighting attributes to defer by default in the esp core code?

jonsmirl, 2024-01-11:
It might be better to not cancel the timer and restart it. Just leave the first timer running. For example consider a light show, a light show would continuously change the level. In the current code continuously changing the level results in the attribute never getting written. Is that what we want? Or do you want the attribute to get written at least once per ESP_MATTER_DEFERRED_ATTR_PERSISTENCE_TIME_MS.

I don't care either way, need to ask the lighting people.

wqx6, 2024-01-15:
> You don't want to set these obvious lighting attributes to defer by default in the esp core code?

No, I think it is just enough to provide an interface to set the attribute deferred persistence since the Matter SPEC has no related description about this feature. The user can decide whether to make an attribute deferred persistence by call (or not call) the interface.

wqx6, 2024-01-15:
> It might be better to not cancel the timer and restart it. Just leave the first timer running. For example consider a light show, a light show would continuously change the level. In the current [...]

Thanks for the suggestion, will make the changes ASAP

shubhamdp, 2024-04-03:
This was fixed in 9c105d551c683d9577cdbb02309fca3e47e9670a  and this issue can be closed now.

closed 2024-04-03 by jonsmirl