espressif/esp-matter #132: Device showing offline in apple home kit

SYESHASWINI, 2022-12-21:
As said in #130 issue I have tried.

$ cd /path/to/esp-matter/examples/light
$ rm -rf sdkconfig sdkconfig.old build/
$ idf.py set-target esp32c3
$ idf.py erase-flash
$ idf.py build flash monitor

@dhrishi flashed the device with the above changes, but after many trials the device is reachable. Every time I flash the device, very often the device is unreachable only few times its reachable.

I have made few changes in the SDK so that it have 4 endpoints. Sometimes the device is added with only 3 endpoints and remaining time the devices are showing offline.
Any solution for this.

dhrishi, 2022-12-21:
>I have made few changes in the SDK so that it have 4 endpoints. Sometimes the device is added with only 3 endpoints

Can you please provide these changes as a patch or diff? So that we can try it out at our end

SYESHASWINI, 2022-12-21:
Changes I have made:

on_off_light::config_t light_config;
light_config.on_off.on_off = DEFAULT_POWER;
light_config.on_off.lighting.start_up_on_off = nullptr;
endpoint_t *endpoint = on_off_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
endpoint_t *endpoint1 = on_off_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
endpoint_t *endpoint2 = on_off_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
endpoint_t *endpoint3 = on_off_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);

There are the changes made to get 4 endpoints at the device side. Even through the sample Light examples I'm facing issue as the devices as going offline.

dhrishi, 2022-12-21:
>Even through the sample Light examples I'm facing issue as the devices as going offline.

Is there a set of steps that you follow when you encounter this issue? Or it happens randomly? Is there any observation around this which you can share.

On a different note, I thought the endpoint thing to be a separate issue altogether and hence requested you to create a new issue. But, it looks like your issue is the same :-)

SYESHASWINI, 2022-12-22:
$ cd /path/to/esp-matter/examples/light
$ idf.py set-target esp32c3
$ idf.py erase-flash
$ idf.py flash monitor

These are the steps I'm following while flashing the device. I have tried adding devices to different networks, but there is no change, the devices are showing offline.

SYESHASWINI, 2022-12-23:
are the devices sending any beacons to the home app to recognise and make device reachable in the apple home.

shubhamdp, 2022-12-23:
@SYESHASWINI can you provide the commit id that you are using for this.

SYESHASWINI, 2022-12-24:
@shubhamdp commit id: 0f8362ad84fe4a446b807ffbfc9e65d078ecf732

dhrishi, 2023-01-06:
@SYESHASWINI Is the issue with Apple Home fixed? If yes, please close this

closed 2023-01-09 by SYESHASWINI