espressif/esp-matter #133: BT on S3 with PSRAM

jonsmirl, 2022-12-22:
I have new S3 board with PSRAM and I get this error at boot:
~~~
W (1430) BT_INIT: esp_bt_controller_mem_release not implemented, return OK
I (1440) BT_INIT: BT controller compile version [421c279]
I (1440) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
E (1560) BT_INIT: Malloc failed
E (1560) chip[DL]: Disabling CHIPoBLE service due to error: Error CHIP:0x0000000B
~~~
This was working fine on my S3 board without PSRAM, so I need to config something different for new board?

I tried all three of the BT allocation options: internal, spiram, default. All same error.

dhrishi, 2022-12-22:
@jonsmirl Can you please share your sdkconfig file

jonsmirl, 2022-12-22:
sdkconfig

[sdkconfig.s3.txt](https://github.com/espressif/esp-matter/files/10289317/sdkconfig.s3.txt)

dhrishi, 2022-12-23:
Just to confirm, is this with the default esp-matter example or your custom example?

alibukharai, 2022-12-23:
@jonsmirl I faced the same problem some days ago, try to use different configuration. I solve this problem by changing serial flash configuration. There are many S3 dev kits with different PSRAM manifacturer so one configuration cannot be used on other, you can check the module top for more detail information about it.

jonsmirl, 2022-12-23:
That was on my code, but the error is before my code has a chance to run. I just checked the ble_mesh example on my new S3 board. The ble_mesh example works. So this problem must be fixed in the ble_mesh code. 

Tomorrow I will copy the current ble_mesh code into my project. Hopefully the problem will go away.

jonsmirl, 2022-12-23:
Copying over ble-mesh code made no difference. Here's the full log up to the error...
I am likely running out of memory somehow.
~~~
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375be0
0x40375be0: esp_restart_noos at /home/jonsmirl/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1664
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2e8c
entry 0x403c9954
I (24) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (24) boot: compile time 23:54:11
I (24) boot: chip revision: 0
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (64) boot:  1 nvs              WiFi data        01 02 00010000 00008000
I (72) boot:  2 otadata          OTA data         01 00 00018000 00002000
I (79) boot:  3 phy_init         RF data          01 01 0001a000 00001000
I (87) boot:  4 ota_0            OTA app          00 10 00020000 001d0000
I (94) boot:  5 ota_1            OTA app          00 11 001f0000 001d0000
I (101) boot:  6 fctry            WiFi data        01 02 003c0000 00006000
I (109) boot: End of partition table
I (113) esp_image: segment 0: paddr=00020020 vaddr=3c130020 size=4377ch (276348) map
I (171) esp_image: segment 1: paddr=000637a4 vaddr=3fc98230 size=05744h ( 22340) load
I (176) esp_image: segment 2: paddr=00068ef0 vaddr=40374000 size=07128h ( 28968) load
I (184) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=12b67ch (1226364) map
I (406) esp_image: segment 4: paddr=0019b6a4 vaddr=4037b128 size=0d0fch ( 53500) load
I (418) esp_image: segment 5: paddr=001a87a8 vaddr=50000000 size=00010h (    16) load
I (427) boot: Loaded app from partition at offset 0x20000
I (427) boot: Disabling RNG early entropy source...
I (439) spiram: Found 16MBit SPI RAM device
I (439) spiram: SPI RAM mode: sram 40m
I (439) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (444) cpu_start: Pro cpu up.
I (447) cpu_start: Starting app cpu, entry point is 0x40375504
0x40375504: call_start_cpu1 at /home/jonsmirl/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (436) cpu_start: App cpu up.
I (873) spiram: SPI SRAM memory test OK
I (882) cpu_start: Pro cpu start user code
I (882) cpu_start: cpu freq: 160000000
I (882) cpu_start: Application information:
I (885) cpu_start: Project name:     lowpan
I (889) cpu_start: App version:      v1.0
I (894) cpu_start: Compile time:     Dec 22 2022 23:54:05
I (900) cpu_start: ELF file SHA256:  ec337ee0d28378bd...
I (906) cpu_start: ESP-IDF:          v4.4.3-dirty
I (912) heap_init: Initializing. RAM available for dynamic allocation:
I (919) heap_init: At 3FCB8748 len 00030FC8 (195 KiB): D/IRAM
I (925) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (932) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (938) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (944) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (953) spi_flash: detected chip: generic
I (957) spi_flash: flash io: dio
I (962) sleep: Configure to isolate all GPIO pins in sleep state
I (968) sleep: Enable automatic switching of GPIO sleep configuration
I (975) coexist: coexist rom version e7ae62f
I (980) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[... 61 lines cut ...]
I (1400) chip[DL]: NVS set: chip-counters/reboot-count = 30 (0x1E)
I (1410) chip[DL]: Real time clock set to 946684800 (0100/00/01 00:00:00 UTC)
W (1410) BT_INIT: esp_bt_controller_mem_release not implemented, return OK
I (1420) BT_INIT: BT controller compile version [421c279]
I (1420) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
E (1560) BT_INIT: Malloc failed
E (1560) chip[DL]: Disabling CHIPoBLE service due to error: Error CHIP:0x0000000B
E (1560) chip[DL]: Long dispatch time: 149 ms, for event type 2
I (1570) chip[DL]: Starting ESP WiFi layer
I (1570) wifi:mode : sta (70:04:1d:cd:77:d0)
I (1580) wifi:enable tsf
I (1570) app_main: Free 2114783
W (1580) wifi:Haven't to connect to a suitable AP now!
I (1590) chip[DL]: Done driving station state, nothing else to do...
W (1590) wifi:Haven't to connect to a suitable AP now!
I (1600) chip[DL]: Done driving station state, nothing else to do...
I (1610) chip[SVR]: Server initializing...
I (1610) chip[TS]: Last Known Good Time: 2022-12-22T23:55:56
I (1620) chip[DMG]: AccessControl: initializing
I (1620) chip[DMG]: Examples::AccessControlDelegate::Init
I (1630) chip[DMG]: AccessControl: setting
I (1630) chip[DMG]: DefaultAclStorage: initializing
I (1640) chip[DMG]: DefaultAclStorage: 0 entries loaded
I (1690) chip[ZCL]: Using ZAP configuration...
~~~

jonsmirl, 2022-12-23:
This appears to be a problem with allocating DMA capable memory.  This is my first time dealing with PSRAM issues. Is there a way to push CHIP memory use into PSRAM and out of the DMA capable memory? Did those thirty end points I allocated come out of DMA memory?

The Linux allocator always gives non-DMA memory first. It will only give you DMA memory if you explicitly ask for it or there is nothing else.

jonsmirl, 2023-03-24:
This is much better shape now that I can push esp-matter into PSRAM