espressif/esp-matter #134: Limit on number of end points

jonsmirl, 2022-12-22:
I need more than 16 endpoints.
~~~
I (3114) esp_matter_core: Dynamic endpoint 13 added
I (3114) esp_matter_attribute: ********** Endpoint 0x000E's Cluster 0x0003's Attribute 0x0001 is 0 **********
I (3114) esp_matter_attribute: ********** Endpoint 0x000E's Cluster 0x0004's Attribute 0x0000 is 128 **********
I (3124) esp_matter_attribute: ********** Endpoint 0x000E's Cluster 0x0004's Attribute 0xFFFC is <invalid type: 0> **********
I (3144) esp_matter_core: Dynamic endpoint 14 added
I (3144) esp_matter_attribute: ********** Endpoint 0x000F's Cluster 0x0003's Attribute 0x0001 is 0 **********
I (3154) esp_matter_core: Dynamic endpoint 15 added
E (3164) esp_matter_core: Error adding dynamic endpoint 16: 0x89
E (3174) esp_matter_core: Error adding dynamic endpoint 17: 0x89
E (3174) esp_matter_core: Error adding dynamic endpoint 18: 0x89
E (3184) esp_matter_core: Error adding dynamic endpoint 19: 0x89
E (3194) esp_matter_core: Error adding dynamic endpoint 20: 0x89
E (3194) esp_matter_core: Error adding dynamic endpoint 21: 0x89
E (3204) esp_matter_core: Error adding dynamic endpoint 22: 0x89
E (3214) esp_matter_core: Error adding dynamic endpoint 23: 0x89
E (3214) esp_matter_core: Error adding dynamic endpoint 24: 0x89
E (3224) esp_matter_core: Error adding dynamic endpoint 25: 0x89
~~~
32 at least, that gives room for seven birdged devices, 40 would be better.

dhairyashah1, 2022-12-22:
Hello @jonsmirl 

1. I recreated your issue by trying to add `more than 16 endpoints ` (**on_off endpoints**) in the [light example](https://github.com/espressif/esp-matter/tree/main/examples/light).
Seems like there is a **limit of 16 endpoints by default**.  Maximum dynamic endpoint count is governed by a macro `CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT`

2. You can find and change this limit, `<your_app>/main/zap_generated/endpoint_config.h` - [light/main/zap_generated/endpoint_config.h](https://github.com/espressif/esp-matter/blob/5ff7d0d7fa9e93d248b557f0c4eee2a7df5410d8/examples/light/main/zap-generated/endpoint_config.h#L49).

2. Logs for my setup - ( I created 20 endpoints in `app_main`)
After updating the number of maximum dynamic endpoints ( `CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT` = 20)
[image]

jonsmirl, 2022-12-22:
this works.

closed 2022-12-22 by jonsmirl