espressif/esp-matter #135: error 13 in install.sh immediately closes wsl2 window

fkkjohn, 2022-12-22:
I've finally gotten everything setup successfully until esp-matter, and as soon as I type in ". ./install.sh", my wsl2 window is closed by an error that can't be seen by the naked eye unless you run the file with "sh install.sh", then you can see there's error 13, source cannot be found. At least that's the error I got after going back in and trying without all of the exports. This is extremely frustrating because it closes all of my virtual environments with it. Now I have to go back in and re-export.sh matter and esp-idf just to try something else, at which point it will undoubtedly close my window again. I've searched the .sh files for an exit command and I can't find one. Why is it closing my window on a small error like this? What can we do to prevent such anti-user behavior?

shubhamdp, 2022-12-22:
Install.sh script is supposed to be executed and not sourced. I see you have source it -> `. ./install.sh`.

dhrishi, 2022-12-26:
@fkkjohn Please close the issue if this helps

dhrishi, 2023-01-03:
Closing. Feel free to reopen in case of any issue

closed 2023-01-03 by dhrishi