espressif/esp-matter #137: generic switch issues

jonsmirl, 2022-12-26:
In general everything seems to be working. A few minor issues.....

What is the chip-tool command to subscribe to all of the events with the is-urgent flag?
This is as close as I could come
any subscribe-all 0x3b,0x3b 0xFFFFFFFF 0xFFFFFFFF 0 1 12345 3,3

The multipress count is not reset on button_multipress_complete

Turn on the features:
~~~
        generic_switch::config_t generic_config;
        generic_config.switch_cluster.number_of_positions = (NUM_BUTTONS == 1 ? 2 : 3);
        device.endpoint[i].hGeneric = generic_switch::create(node, &generic_config, ENDPOINT_FLAG_NONE, (void *)i);
        /* Add additional features to the node */
        cluster_t *cluster = cluster::get(device.endpoint[i].hGeneric, Switch::Id);
        cluster::switch_cluster::feature::momentary_switch::add(cluster);
        cluster::switch_cluster::feature::momentary_switch_release::add(cluster);
        cluster::switch_cluster::feature::momentary_switch_long_press::add(cluster);
        cluster::switch_cluster::feature::momentary_switch_multi_press::config_t multi_config;
        multi_config.multi_press_max = 3;
        cluster::switch_cluster::feature::momentary_switch_multi_press::add(cluster, &multi_config);
~~~

Add the optional labels to the example?
~~~
        fixed_label::config_t fixed_config;
        fixed_config.cluster_revision = 1;
        cluster::fixed_label::create(device.endpoint[i].hGeneric, &fixed_config, CLUSTER_FLAG_SERVER);
        user_label::config_t user_config;
        user_config.cluster_revision = 1;
        cluster::user_label::create(device.endpoint[i].hGeneric, &user_config, CLUSTER_FLAG_SERVER);
~~~
And then combine mfg_tool into the example and show how to set the fixed label.

jadhavrohit924, 2022-12-30:
Chip-tool don't have is-urgent flag for `any` command.

`The multipress count is not reset` is fixed. Please try pulling latest main.

jonsmirl, 2022-12-30:
Don't the features need to be turned on? Otherwise the controller won't know the switch supports the events.

jadhavrohit924, 2023-01-02:
All features are not mandatory so not enabled by default.

dhrishi, 2023-01-27:
Closing this as the mfg_tool part here is kept open in #118

closed 2023-01-27 by dhrishi