espressif/esp-matter #140: [API consult] where can i find the matter ble adv api?

sidhoo2, 2022-12-27:
cause i need to pair both matter and my own app，so can i open two ble adv ?

dhrishi, 2022-12-27:
@sidhoo2 APIs are:
`ble_gap_adv_set_data()` usage [here](https://github.com/espressif/connectedhomeip/blob/87bee4de25886ba00f1a177cd0f8791a48528a1f/src/platform/ESP32/nimble/BLEManagerImpl.cpp#L738) followed by `ble_gap_adv_start` usage [here](https://github.com/espressif/connectedhomeip/blob/87bee4de25886ba00f1a177cd0f8791a48528a1f/src/platform/ESP32/nimble/BLEManagerImpl.cpp#L1222)

jonsmirl, 2022-12-27:
You should not be pairing Matter and then your own app with a separate BT protocol.  There is a standard pairing procedure on both IOS and Android which will pair both your app and Apple/Google Home all in a single step.  After that your app should implement a vendor specific, Matter endpoint/cluster to do custom setup.

sidhoo2, 2022-12-28:
@dhrishi thanks, Is there any api to add anothor ble slave ?

sidhoo2, 2022-12-28:
@jonsmirl yes, using the matter pairing procedure is ok, but my project needs to transfer different data within the ble protocol, so,  is there any way to establish two ble connections ?

dhrishi, 2022-12-28:
@sidhoo2 Why do you need 2 BLE connections? When do you want to exchange the data over BLE? Will post-commissioning be okay? As for Matter, you would not need BLE anytime after the first commissioning is done

sidhoo2, 2022-12-29:
@dhrishi post-commissioning ?  such as the ble mesh example?

dhrishi, 2022-12-30:
@sidhoo2 Yes. Depending on your usecase. That's why I want to know:

>When do you want to exchange the data over BLE?  

Also, if you can share the purpose of this BLE data transfer, it will help

dhrishi, 2023-01-23:
Closing as the original question is answered. Please share details for any further help.

closed 2023-01-23 by dhrishi