espressif/esp-matter #142: FAILED: gen/third_party/connectedhomeip/src/controller/data_model/app/PluginApplicationCallbacks.h

mriksman, 2022-12-29:
I have set up an environment in VSCode using the following steps
1. Install WSL
2. Install Docker Desktop
3. Install VSCode
4. Install ESP-IDF Extension
5. Use ESP-IDF Extension to create a New Project
6. Edit the Dockerfile to include stuff for Matter

```
ENV PATH=/opt/qemu/bin:${PATH}

#############################
# Additional lines for Matter installation
# https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html

# Prerequisites for Matter
RUN : \
  && apt-get update \
  && apt-get install -y \
  git \
  gcc \
  g++ \
  pkg-config \
  libssl-dev \
  libdbus-1-dev \
  libglib2.0-dev \
  libavahi-client-dev \
  ninja-build \
  python3-venv \
  python3-dev \
  python3-pip \
  unzip \
  libgirepository1.0-dev \
  libcairo2-dev \
  libreadline-dev \
  && apt-get autoremove -y \
  && rm -rf /var/lib/apt/lists/* \
  && : # last line

ARG ESP_MATTER_CLONE_URL=https://github.com/espressif/esp-matter.git

# perform a shallow clone
RUN git clone --depth 1 $ESP_MATTER_CLONE_URL $IDF_TOOLS_PATH/esp-matter && \
	cd $IDF_TOOLS_PATH/esp-matter && \
    git submodule update --init --depth=1 && \
    ./connectedhomeip/connectedhomeip/scripts/checkout_submodules.py --platform esp32 --shallow

# run Matter setup and build host tools and python dependencies for mfg_tool
RUN cd $IDF_TOOLS_PATH/esp-matter && \
    ./install.sh

# ** note: we may need to modify the entrypoint.sh file below to include the esp-matter export.sh
RUN echo "source /opt/esp/esp-matter/export.sh > /dev/null 2>&1" >> ~/.bashrc

#############################
# Back to orginal Dockerfile
```

Everything seems to build OK.

I then copy the `light` example in to my `/workspaces` folder (the project workspace in VSCode) and perform a Build.

I get the following error

```
[96/97] Generating binary image from built executable
esptool.py v3.3.2-dev
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
[... 67 lines cut ...]
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
[47/729] stamp obj/third_party/connectedhomeip/src/app/app_config.stamp
[48/729] stamp obj/third_party/connectedhomeip/src/app/common/cluster-objects.inputdeps.stamp
[49/729] stamp obj/third_party/connectedhomeip/src/app/server/server.inputdeps.stamp
[50/729] stamp obj/third_party/connectedhomeip/src/app/tests/binding-test-srcs.inputdeps.stamp
[51/729] stamp obj/third_party/connectedhomeip/src/app/tests/helpers.inputdeps.stamp
[52/729] stamp obj/third_party/connectedhomeip/src/app/tests/ota-requestor-test-srcs.inputdeps.stamp
[53/729] stamp obj/third_party/connectedhomeip/src/app/tests/tests_lib.inputdeps.stamp
[54/729] stamp obj/third_party/connectedhomeip/src/ble/ble.inputdeps.stamp
[55/729] c++ obj/third_party/connectedhomeip/third_party/nlfaultinjection/repo/src/libnlfaultinjection.nlfaultinjection.cpp.o
[56/729] c++ obj/third_party/connectedhomeip/src/access/libaccess.AccessControl.cpp.o
[57/729] c++ obj/third_party/connectedhomeip/src/access/examples/libaccess.ExampleAccessControlDelegate.cpp.o
[58/729] c++ obj/third_party/connectedhomeip/third_party/jsoncpp/repo/src/lib_json/jsoncpp.json_writer.cpp.o
[59/729] c++ obj/third_party/connectedhomeip/src/app/tests/suites/credentials/dac_provider.TestHarnessDACProvider.cpp.o
[60/729] c++ obj/third_party/connectedhomeip/src/app/util/mock/mock_ember.attribute-storage.cpp.o
[61/729] c++ obj/third_party/connectedhomeip/third_party/jsoncpp/repo/src/lib_json/jsoncpp.json_reader.cpp.o
[62/729] c++ obj/third_party/connectedhomeip/third_party/jsoncpp/repo/src/lib_json/jsoncpp.json_value.cpp.o
[63/729] c++ obj/third_party/connectedhomeip/src/access/tests/libaccesstest.TestAccessControl.cpp.o
ninja: build stopped: subcommand failed.
[668/1312] Linking C static library esp-idf/app_update/libapp_update.a
ninja: build stopped: subcommand failed.
```

Any ideas?

mriksman, 2022-12-29:
Possibly the Matter `requirements.txt` wasn't run?
https://github.com/project-chip/connectedhomeip/blob/19c8ea42b4f4af2669643b758caa4cf412cfee69/scripts/requirements.txt#L69-L71
I ran `pip install lark` and the next error I got was for `stringcase`

So something wrong with my Dockerfile, or with the `./install.sh` in `esp-matter`.

mriksman, 2022-12-29:
There are other 'issues' as well with VSCode and Dev Containers.

When I run the Build command from the Activity bar (the little icon that looks like a cylinder), it errors with 
```
-- Found Git: /usr/bin/git (found version "2.25.1") 
CMake Error at /opt/esp/esp-matter/device_hal/device/m5stack/esp_matter_device.cmake:3 (message):
  please set esp32 as the IDF_TARGET using 'idf.py set-target esp32'
Call Stack (most recent call first):
  CMakeLists.txt:32 (include)
```

But in a new Bash terminal, I have already run `idf.py set-target esp32c3`, and `idf.py build` runs fine.

Someone needs to explain how VSCode, linux Containers and the environment variables are behaving. I bet it is a simple fix once I understand how each of these windows are loading their environments.

shubhamdp, 2023-01-19:
@mriksman sorry for the late turn around!

> ModuleNotFoundError: No module named 'lark'

This has been fixed with 350dc6ae8b011b1e29070797daa55100840b3fbf

> \# run Matter setup and build host tools and python dependencies for mfg_tool
> RUN cd $IDF_TOOLS_PATH/esp-matter && \
>     ./install.sh
>
Before running `install.sh` please add `source $IDF_PATH/export.sh`, this would install the `lark` in IDF python environment preventing the `lark` error.

> My `~/.bashrc` has the following at the end
> 
> ```
> source /opt/esp/esp-matter/export.sh > /dev/null 2>&1
> source /opt/esp/idf/export.sh > /dev/null 2>&1
> ```

Sourcing sequence may be problem here, you are expected to source `esp-idf/export.sh` first and then `esp-matter/export.sh`

dhrishi, 2023-01-31:
@mriksman Can you please try this out and close the issue if it works as expected

mriksman, 2023-01-31:
> @mriksman Can you please try this out and close the issue if it works as expected

I'm working abroad at the moment, and I don't have much download data. I have it on my 'to-do' list for when I return home; about 2 weeks. Can we leave it open for 2 weeks, and I'll test when I am home and report back?

Thanks.

dhrishi, 2023-01-31:
>Can we leave it open for 2 weeks, and I'll test when I am home and report back?

Yes, definitely. Not a problem

mriksman, 2023-02-18:
Hi,

- I started a new project from ESP IDF Extension, and selected 'Sample Project'
- I assume `FROM espressif/idf` will get the `latest`, so I modify the line to be `FROM espressif/idf:v4.4.3`
- I add the following _after_ `RUN echo "source /opt/esp/idf/export.sh > /dev/null 2>&1" >> ~/.bashrc` (**as per your comment**)
```
#############################
# Additional lines for Matter installation
# https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html

# Prerequisites for Matter
RUN : \
  && apt-get update \
  && apt-get install -y \
  git \
  gcc \
  g++ \
  pkg-config \
  libssl-dev \
  libdbus-1-dev \
  libglib2.0-dev \
  libavahi-client-dev \
  ninja-build \
  python3-venv \
  python3-dev \
  python3-pip \
  unzip \
  libgirepository1.0-dev \
  libcairo2-dev \
  libreadline-dev \
  && apt-get autoremove -y \
  && rm -rf /var/lib/apt/lists/* \
  && : # last line

ARG ESP_MATTER_CLONE_URL=https://github.com/espressif/esp-matter.git

# perform a shallow clone
RUN git clone --depth 1 $ESP_MATTER_CLONE_URL $IDF_TOOLS_PATH/esp-matter && \
	cd $IDF_TOOLS_PATH/esp-matter && \
    git submodule update --init --depth=1 && \
    ./connectedhomeip/connectedhomeip/scripts/checkout_submodules.py --platform esp32 --shallow

# run Matter setup and build host tools and python dependencies for mfg_tool
```
Modified this to include ` . $IDF_PATH/export.sh && \` **as per your comment**.
```
RUN cd $IDF_TOOLS_PATH/esp-matter && \
    . $IDF_PATH/export.sh && \
    ./install.sh

# ** note: we may need to modify the entrypoint.sh file below to include the esp-matter export.sh
RUN echo "source /opt/esp/esp-matter/export.sh > /dev/null 2>&1" >> ~/.bashrc

#############################
# Back to orginal Dockerfile
```

**??** When it completed (and each time I open the VSCode Container), it showed the following in the Output tab. Not sure if its relevant?
```
[2/18/2023, 7:08:27 AM] Unable to resolve configuration with compilerPath "C:\Espressif\esp32\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.exe".  Using "/opt/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/arm/bin/arm-none-eabi-gcc" instead.
[2/18/2023, 7:08:27 AM] Unable to resolve configuration with compilerPath "C:\Espressif\esp32\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.exe".  Using "/opt/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/arm/bin/arm-none-eabi-gcc" instead.
```

When I tried to build the `light` sample, I get
```
-- Building ESP-IDF components for target esp32c3
CMake Error at /opt/esp/idf/tools/cmake/build.cmake:452 (message):
Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:8 (project)
```

I modified the `devcontainer.json` file to point to the correct python path.
```
//		"idf.pythonBinPath": "/opt/esp/python_env/idf5.1_py3.8_env/bin/python",
		"idf.pythonBinPath": "/opt/esp/python_env/idf4.4_py3.8_env/bin/python",
```

🎉🥳
This time (after having to delete the `managed_components` folder in `/workspaces` - this is new..?) I was able to compile.

Do I need to modify the default `entrypoint.sh`, even though they were added via `source` in the `Dockerfile`? It currently contains. It compiled OK without any modifications...
```
#!/usr/bin/env bash
set -e

. $IDF_PATH/export.sh

exec "$@"
```

Is this the correct/best way of creating a new ESP Matter project in VSCode Docker Container? Or is there a better way?

mriksman, 2023-02-20:
Given that it built OK, what would modifying the `entrypoint.sh` do? Fix the errors that appear in the Output log when opening the VSCode Container?

I'm away again for 18 days, and will try when I am back on my PC.

shubhamdp, 2023-03-01:
This just sets up the required stuff to run esp-matter. As sourcing $IDF_PATH/export.sh is necessary before building any IDF application, similarly sourcing esp-matter/export.sh is necessary for building any esp-matter examples. That's the reason I suggested that you add one more line there.

dhrishi, 2023-04-30:
I am closing this now. Please re-open in case of any issue

closed 2023-04-30 by dhrishi