espressif/esp-matter #143: Home App - Fails to Pair Accessory -  Received error (protocol code 2) during pairing process

mriksman, 2022-12-29:
Hi,
### 1. I am an Admin, and my fiancee is the Owner of the Apple HomeKit Home.
I used my phone to add the Matter device, but as I am Admin of the house and not the Owner (my fiancee is), I get the following display.
[image]

- If I click OK and try and add a new accessory, the Matter accessory is visible on my phone. 
- If I try and add the device on my fiancee's phone, it isn't visible. 
- On my phone, I click the Matter device, and the above warning appears, and I notice it sends a command to open commissioning window. It is only at that time that the device becomes visible on my finacee's phone. Very cool. I can now proceed with adding the device.

However, when I select the Matter device on her phone, it just hangs up
[image]

In the logs, I see errors
```
I (19024) ROUTE_HOOK: Received RIO
I (19034) ROUTE_HOOK: prefix FD46:BA6D:CE9D:: lifetime 1800

I (19324) ROUTE_HOOK: Received RIO
I (19324) ROUTE_HOOK: prefix FD99:7AEE:449A:: lifetime 1800

I (19844) ROUTE_HOOK: Received RIO
I (19844) ROUTE_HOOK: prefix FD99:7AEE:449A:: lifetime 1800

E (30074) chip[IN]: Data received on an unknown session (LSID=10196). Dropping it!
E (34784) chip[IN]: Data received on an unknown session (LSID=10196). Dropping it!
E (39294) chip[IN]: Data received on an unknown session (LSID=10196). Dropping it!
I (44414) ROUTE_HOOK: Received RIO
I (44414) ROUTE_HOOK: prefix FD99:7AEE:449A:: lifetime 1800

E (46664) chip[IN]: Data received on an unknown session (LSID=10196). Dropping it!
I (52604) chip[EM]: >>> [E:51149r M:178417320] (U) Msg RX from 0:33450209EADD6E45 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (52604) chip[IN]: CASE Server received Sigma1 message. Starting handshake. EC 0x3fc98cf8
I (52614) chip[IN]: CASE Server disabling CASE session setups
I (52634) chip[EM]: <<< [E:51149r M:46017097 (Ack:178417320)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (52644) chip[IN]: (U) Sending msg 46017097 to IP address 'UDP:[FE80::1037:4BB5:7FBC:4569%st1]:59734'
I (52654) chip[SC]: Received Sigma1 msg
I (52664) chip[EM]: <<< [E:51149r M:46017098 (Ack:178417320)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:33 (SecureChannel:CASE_Sigma2Resume)
I (52674) chip[IN]: (U) Sending msg 46017098 to IP address 'UDP:[FE80::1037:4BB5:7FBC:4569%st1]:59734'
I (53014) chip[EM]: >>> [E:51149r M:178417321 (Ack:46017098)] (U) Msg RX from 0:33450209EADD6E45 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (53024) chip[SC]: Success status report received. Session was established
I (53044) chip[SC]: SecureSession[0x3fc974d0]: Moving from state 'kEstablishing' --> 'kActive'
I (53044) chip[IN]: CASE Session established to peer: <61A31AE3257E956B, 2>
I (53054) chip[IN]: CASE Server enabling CASE session setups
I (53054) chip[EM]: <<< [E:51149r M:46017099 (Ack:178417321)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (53074) chip[IN]: (U) Sending msg 46017099 to IP address 'UDP:[FE80::1037:4BB5:7FBC:4569%st1]:59734'
I (53084) chip[EM]: >>> [E:51150r M:127948830] (S) Msg RX from 2:61A31AE3257E956B [4A5F] --- Type 0001:0a (IM:TimedRequest)
I (53104) chip[EM]: <<< [E:51150r M:86783068 (Ack:127948830)] (S) Msg TX to 2:61A31AE3257E956B [4A5F] --- Type 0001:01 (IM:StatusResponse)
I (53114) chip[IN]: (S) Sending msg 86783068 on secure session with LSID: 5441
I (53124) chip[EM]: >>> [E:51150r M:127948831 (Ack:86783068)] (S) Msg RX from 2:61A31AE3257E956B [4A5F] --- Type 0001:08 (IM:InvokeCommandRequest)
I (53134) esp_matter_command: Received command 0x0000 for enpoint 0x0000's cluster 0x0000003C
I (53144) chip[ZCL]: Received command to open commissioning window
I (53154) chip[DIS]: Updating services using commissioning mode 2
I (53154) chip[DIS]: CHIP minimal mDNS started advertising.
I (53184) chip[DIS]: Advertise operational node 62276A0E3AA35B0D-0000000063E40CB5
I (53184) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 62276A0E3AA35B0D-0000000063E40CB5.
I (53204) chip[DIS]: mDNS service published: _matter._tcp
I (53204) chip[DIS]: Advertise operational node 99403E686FAB4A5F-000000001D300FA0
I (53224) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 99403E686FAB4A5F-000000001D300FA0.
I (53244) chip[DIS]: mDNS service published: _matter._tcp
I (53244) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3600/14 cm=2
I (53254) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 05A551279FB9BA75.
I (53294) chip[DIS]: mDNS service published: _matterc._udp
I (53294) chip[ZCL]: Commissioning window is now open
I (53304) chip[EM]: <<< [E:51150r M:86783069 (Ack:127948831)] (S) Msg TX to 2:61A31AE3257E956B [4A5F] --- Type 0001:09 (IM:InvokeCommandResponse)
I (53314) chip[IN]: (S) Sending msg 86783069 on secure session with LSID: 5441
E (53324) chip[DL]: Long dispatch time: 203 ms, for event type 3
I (53324) app_main: Commissioning window opened
I (53344) chip[EM]: >>> [E:51150r M:127948832 (Ack:86783069)] (S) Msg RX from 2:61A31AE3257E956B [4A5F] --- Type 0000:10 (SecureChannel:StandaloneAck)
E (58444) chip[IN]: Data received on an unknown session (LSID=10196). Dropping it!
I (73184) chip[EM]: >>> [E:62910r M:190106792] (U) Msg RX from 0:5456093A6581D1E7 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
[... 688 lines cut ...]
I (98368) chip[FS]: Fail-safe cleanly disarmed
I (98368) app_main: Commissioning window opened
I (99218) ROUTE_HOOK: Received RIO
I (99218) ROUTE_HOOK: prefix FD46:BA6D:CE9D:: lifetime 1800

I (112658) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 0
I (112658) chip[DL]: CHIPoBLE unsubscribe received
E (112658) chip[BLE]: no endpoint for unsub recvd
I (112668) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (112678) NimBLE: GAP procedure initiated: advertise; 
I (112678) NimBLE: disc_mode=2
I (112688) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (112708) NimBLE: 

I (112708) chip[DL]: CHIPoBLE advertising started
I (112718) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (112718) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (112728) chip[DL]: Device already advertising, stop active advertisement and restart
I (112728) NimBLE: GAP procedure initiated: stop advertising.

I (112738) NimBLE: GAP procedure initiated: advertise; 
I (112738) NimBLE: disc_mode=2
I (112758) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (112768) NimBLE: 
```

mriksman, 2022-12-29:
I tried with the QR code and it worked.
Ugh. I tried with passcode `20202021` and it worked....
What is `34970112332` for!?

closed 2022-12-29 by mriksman

dhrishi, 2023-01-09:
>What is 34970112332 for!?

@mriksman It's the manual pairing code that can be used in certain situations next to or instead of QR code. An example could be commissioning through a voice assistant during which you can say (audio) the manual pairing code as the proof of possession.

mriksman, 2023-01-09:
@dhrishi Thanks. But. What is the difference between the 'manual code' `20202021` vs `34970112332`. Why does the iOS Home App use `20202021`, whilst the another app (Espressif Matter iOS App???) uses `34970112332`? What different information does it convey?

dhrishi, 2023-01-09:
`20202021` is not the manual pairing code. It is the setup passcode, that serves as the proof of possession. 

This setup passcode along with other parameters like discriminator, vid, pid etc. are embedded into QR code as well as the manual pairing code.