espressif/esp-matter #144: Using gpio isr, when the level is changed, the board will be reset.

Firesuiry, 2022-12-31:
the code is as follows:
```
#define GPIO_INPUT_TOUCH_SENSER     GPIO_NUM_4
#define GPIO_INPUT_PIN_SEL  (1ULL<<GPIO_INPUT_TOUCH_SENSER) 
    //zero-initialize the config structure.
    gpio_config_t io_conf = {};
    //disable interrupt
    io_conf.intr_type = GPIO_INTR_ANYEDGE;
    //set as output mode
    io_conf.mode = GPIO_MODE_INPUT;
    //bit mask of the pins that you want to set,e.g.GPIO18/19
    io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
    //disable pull-down mode
    io_conf.pull_down_en = (gpio_pulldown_t)0;
    //disable pull-up mode
    io_conf.pull_up_en = (gpio_pullup_t)1;
    gpio_config(&io_conf);

    //install gpio isr service
    gpio_install_isr_service(ESP_INTR_FLAG_LEVEL3);
    //hook isr handler for specific gpio pin
    gpio_isr_handler_add(GPIO_INPUT_TOUCH_SENSER, gpio_isr_handler, (void*) GPIO_INPUT_TOUCH_SENSER);
```

when I change the voltage of GPIO_NUM_4, esp32c3 will restart.
I try it without esp-matter and it success.
Is there any wrong?

dhairyashah1, 2023-01-05:
@Firesuiry ,
1. You have defined an `anyEdge interrupt` on `GPIO_NUM_4` which calls the `ISR and resets the device` on detecting **anyEdge** viz. **any change in voltage level** on `GPIO_NUM_4` which is also the expected behaviour. 
2. Since this corresponds to basic GPIO interrupts ( [ref_link](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/gpio/generic_gpio/main/gpio_example_main.c#L67) ) could you exactly specify where and how you used it ? 
3. What issue did you face with regards to `using gpio interrupts` on an `esp-matter device`?

Firesuiry, 2023-01-05:
2. I try to display the status of GPIO4 when it changes, the code is as follows:
```
static void IRAM_ATTR gpio_isr_handler(void* arg)
{
    uint32_t gpio_num = (uint32_t) arg;
    printf("GPIO[%d] intr, val: %d\n", gpio_num, gpio_get_level((gpio_num_t)gpio_num));
}
```
3. I want to make a contact sensor, if GPIO4 changes, then I will change the state of the contact sensor.

AshishESP, 2023-01-06:
@Firesuiry 
You have enabled PULLUP for GPIO4. So ideally you should use NegEdge to detect Interrupt. (general use case when you GND particular GPIO which is pulled up internally, interrupt will be triggered)
I am not able to understand  if there is contact Sensor which is connected to GPIO4 or you are using GPIO4 pin itself as Contact sensor? If you have connected Contact sensor at GPIO4 then what is the output from sensor? Also what is the action you want to take when interrupt is triggered? ( you have just displayed GPIO level in ISR service)
Pls explain your use case in detail.

Firesuiry, 2023-01-06:
@AshishESP Thank you for your reply~
I want to use GPIO4 pin itself as a contact sensor~
If GPIO4 changes, I will use attribute::update to update the state in matter.

Firesuiry, 2023-01-07:
new code is as follows:
```
static void IRAM_ATTR gpio_isr_handler(void* arg)
{
    uint32_t gpio_num = (uint32_t) arg;
    printf("GPIO[%d] intr, val: %d\n", gpio_num, gpio_get_level((gpio_num_t)gpio_num));
    esp_matter_attr_val_t val = esp_matter_bool(gpio_get_level((gpio_num_t)gpio_num));
    ESP_ERROR_CHECK(attribute::update(contact_senser_endpoint_id, BooleanState::Id, BooleanState::Attributes::StateValue::Id, &val));
}
```

Firesuiry, 2023-01-07:
all the code is here:https://github.com/Firesuiry/controller
the version of esp-idf is 4.4.2-dirty
the version of esp-matter is at commit 85abe2cdd457f6f1d198af0ff1ee339ccd3c9bfb

AshishESP, 2023-01-17:
@Firesuiry Sorry for the late reply, was on leave.
Have you tried using other GPIO apart from IO4?
Can you pls share logs here?

Firesuiry, 2023-01-18:
@AshishESP  thank you for your reply!
because I am on vacation, and the device is not in my home, I will try it and post logs later.

dhrishi, 2023-01-31:
@Firesuiry Did you get a chance to try this out? Please close the issue if it works

Firesuiry, 2023-01-31:
I will try it this week~

closed 2023-02-02 by Firesuiry